zigpy/zha-device-handlers #2184: [Device Support Request] Add support for Thermostat _TZE200_2ekuz3dz

dala318, 2023-02-08:
**Is your feature request related to a problem? Please describe.**
This version of the device is not compatible with ZHA yet.

**Describe the solution you'd like**
The device is added to ZHA without any entities
Seem very similar as #2182  and #1439

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0051",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_2ekuz3dz",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>
[zha-2b0fd529875b6d880667f0217196a97f-_TZE200_2ekuz3dz TS0601-9b1f91052c7097094bf38f5c0bc23261.json.txt](https://github.com/zigpy/zha-device-handlers/files/10685856/zha-2b0fd529875b6d880667f0217196a97f-_TZE200_2ekuz3dz.TS0601-9b1f91052c7097094bf38f5c0bc23261.json.txt)

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
https://www.aliexpress.com/item/1005004716914337.html

MattWestb, 2023-02-08:
Your device have getting one updated Zigbee module and is having one extra Zigbee GPP endpoint.
So need adding one new device class for it to getting it working.

javicalle, 2023-02-08:
@MattWestb I can try to create the quirk from another existing one. Any good quirk candidate for this device?

MattWestb, 2023-02-09:
Added GPP class to `ts0601_electric_heating.py`.
I have letting the device from the normal class in it then its likely they is also getting Zigbee module with GPP in the future. If not like it i can deleting them.
[ts0601_electric_heating.zip](https://github.com/zigpy/zha-device-handlers/files/10694433/ts0601_electric_heating.zip)

@dala318 Configure local quirk and adding the unzipped quirk in the new directory and restarting HA. If al is going well the system is loading it for your device and you is getting one thermostat on the device card.
Configure local quirks https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274.

Pleas report back !

MattWestb, 2023-02-09:
@javicalle Shall we trying adding the `EnchantedDevice` class to this quirk ?
Its not so large and dont do so mush damage and its easy rolling back if not working well.
The original looks like this: `class MoesBHTGPP(TuyaThermostat):`
How do i doing it right ?
Importing and adding in the class but im not sure how.

dala318, 2023-02-10:
Adding the script to custom_quirks gives me two entities for the device but both show up as unknown
[image]

Thermostat has no target temperature setting and trying to set Operation to Heat or Off jumps back to Unknown after a few seconds

This is reported after a reconfiguration
[image]

Device diagnostic
[zha-2b0fd529875b6d880667f0217196a97f-_TZE200_2ekuz3dz TS0601-9b1f91052c7097094bf38f5c0bc23261.json (1).txt](https://github.com/zigpy/zha-device-handlers/files/10704327/zha-2b0fd529875b6d880667f0217196a97f-_TZE200_2ekuz3dz.TS0601-9b1f91052c7097094bf38f5c0bc23261.json.1.txt)

<details>
<summary>Home Assistant debug log</summary>

```
2023-02-10 07:43:03.033 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=210), 92, -77, 0x246f, 255, 255, b'\ty\x02\x00\xb8\x18\x02\x00\x04\x00\x00\x00\xbf']
2023-02-10 07:43:03.034 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=210), 92, -77, 0x246f, 255, 255, b'\ty\x02\x00\xb8\x18\x02\x00\x04\x00\x00\x00\xbf']
2023-02-10 07:43:03.034 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x246F), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=210, profile_id=260, cluster_id=61184, data=Serialized[b'\ty\x02\x00\xb8\x18\x02\x00\x04\x00\x00\x00\xbf'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=92, rssi=-77)
2023-02-10 07:43:03.035 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Received ZCL frame: b'\ty\x02\x00\xb8\x18\x02\x00\x04\x00\x00\x00\xbf'
2023-02-10 07:43:03.036 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=121, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-10 07:43:03.037 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=0, tsn=184, command_id=536, function=0, data=[4, 0, 0, 0, 191]))
2023-02-10 07:43:03.037 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Received command 0x02 (TSN 121): set_data_response(param=Command(status=0, tsn=184, command_id=536, function=0, data=[4, 0, 0, 0, 191]))
2023-02-10 07:43:03.038 DEBUG (MainThread) [zhaquirks.tuya] [0x246f:1:0xef00] Received value [0, 0, 0, 191] for attribute 0x0218 (command 0x0002)
2023-02-10 07:43:03.038 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x246F:1:0x0201]: Attribute report 'MoesBHTThermostat'[local_temperature] = 1910
2023-02-10 07:43:03.047 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=121, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-10 07:43:03.047 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2023-02-10 07:43:03.048 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x246F), dst_ep=1, source_route=None, extended_timeout=False, tsn=121, profile_id=260, cluster_id=61184, data=Serialized[b'\x18y\x0b\x02\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-10 07:43:03.048 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x246f, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=121), 121, b'\x18y\x0b\x02\x00')
2023-02-10 07:43:03.048 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze200_2ekuz3dz_ts0601_thermostat: Attribute 'local_temperature' = 1910 update
2023-02-10 07:43:03.074 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendUnicast: [<EmberStatus.SUCCESS: 0>, 161]
2023-02-10 07:43:03.145 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 9327, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=161), 121, <EmberStatus.SUCCESS: 0>, b'']
[... cut ...] Y_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 0, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 5, BROADCAST_TABLE_FULL = 0, PTA_LO_PRI_REQUESTED = 0, PTA_HI_PRI_REQUESTED = 0, PTA_LO_PRI_DENIED = 0, PTA_HI_PRI_DENIED = 0, PTA_LO_PRI_TX_ABORTED = 0, PTA_HI_PRI_TX_ABORTED = 0, ADDRESS_CONFLICT_SENT = 0, EZSP_FREE_BUFFERS = 242]
```

MattWestb, 2023-02-10:
Great the quirk is being loaded but somthing is not working OK.
Can you try restarting HA one time and see if its start getting contact with the TRV ?
And if is not working well take out the battery for one minute and putting them back and look what happening.

@javicalle Its one strange things in the log but not every time:
```
2023-02-10 07:43:40.107 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:Default_Response(command_id=0, status=<Status.UNSUP_MANUF_CLUSTER_COMMAND: 131>)
```
Its coming as one response then trying sending somthing to the TRV.

javicalle, 2023-02-10:
> How do i doing it right ?
> Importing and adding in the class but im not sure how.

Yes, that's the way:
```python
from zhaquirks.tuya.mcu import EnchantedDevice
```
```python
class MoesBHTGPP(EnchantedDevice, TuyaThermostat):
```

That's the response to any command like this one:
```
2023-02-10 07:43:38.495 DEBUG (MainThread) [zhaquirks.tuya] [0x246f:1:0x0201] Mapping standard system_mode (0x001c) with value <SystemMode.Heat: 4> to custom {257: 1}
2023-02-10 07:43:38.495 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=126, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2023-02-10 07:43:38.496 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending request: set_data(param=Command(status=0, tsn=126, command_id=257, function=0, data=[1, 1]))
```

It seems to be a `NO_MANUFACTURER_ID`.
But not an easy fix because TRV is using several buses to handle commands and will be needed to fix every one:
* TuyaThermostatCluster
* TuyaUserInterfaceCluster

MattWestb, 2023-02-10:
Shall we testing the `EnchantedDevice` or waiting until we have the device working OK ?

dala318, 2023-02-10:
Did a restart of HA host (Hyper-V installation)
It did take some extra time for Zigbee to report working after start and might be some failed attempts to set the state of one of my Aqara plugs before it started to work.

[home-assistant_2023-02-10T07-46-06.590Z.log](https://github.com/zigpy/zha-device-handlers/files/10704939/home-assistant_2023-02-10T07-46-06.590Z.log)

MattWestb, 2023-02-10:
Looks like receiving command is working OK.
```
2023-02-10 08:44:26.188 DEBUG (MainThread) [zhaquirks.tuya] [0x246f:1:0xef00] Received value [0, 0, 0, 191] for attribute 0x0218 (command 0x0002)
```
But is getting the error then try sending to the device.

@dala318 did the TRV i  ZHA getting temperature populated or some other data ?

dala318, 2023-02-10:
I do get readings of current temperature but not sure I get set-temp on the device.
[image]

In the below log-snip I increased the set temperature to 25.0 deg C by the buttons on the device.

<details>
<summary>log snip</summary>

```
2023-02-10 10:09:46.022 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x246F), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=119, profile_id=260, cluster_id=61184, data=Serialized[b'\t\x00\x02\x00\xfd\x10\x02\x00\x04\x00\x00\x00\xf0'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=116, rssi=-71)
2023-02-10 10:09:46.022 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Received ZCL frame: b'\t\x00\x02\x00\xfd\x10\x02\x00\x04\x00\x00\x00\xf0'
2023-02-10 10:09:46.022 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=0, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-10 10:09:46.023 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=0, tsn=253, command_id=528, function=0, data=[4, 0, 0, 0, 240]))
2023-02-10 10:09:46.023 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Received command 0x02 (TSN 0): set_data_response(param=Command(status=0, tsn=253, command_id=528, function=0, data=[4, 0, 0, 0, 240]))
2023-02-10 10:09:46.023 DEBUG (MainThread) [zhaquirks.tuya] [0x246f:1:0xef00] Received value [0, 0, 0, 240] for attribute 0x0210 (command 0x0002)
2023-02-10 10:09:46.023 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x246F:1:0x0201]: Attribute report 'MoesBHTThermostat'[occupied_heating_setpoint] = 24000
2023-02-10 10:09:46.024 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=0, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-10 10:09:46.024 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
2023-02-10 10:09:46.024 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x246F), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\x18\x00\x0b\x02\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-10 10:09:46.024 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x246f, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=0), 178, b'\x18\x00\x0b\x02\x00')
2023-02-10 10:09:46.025 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.tze200_2ekuz3dz_ts0601_thermostat: Attribute 'occupied_heating_setpoint' = 24000 update
2023-02-10 10:09:46.025 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=climate.tze200_2ekuz3dz_ts0601_thermostat, old_state=<state climate.tze200_2ekuz3dz_ts0601_thermostat=unknown; hvac_modes=[<HVACMode.OFF: 'off'>, <HVACMode.HEAT: 'heat'>], min_temp=7.0, max_temp=30.0, current_temperature=19.2, temperature=None, occupied_heating_setpoint=23500, friendly_name=_TZE200_2ekuz3dz TS0601 Thermostat, supported_features=ClimateEntityFeature.TARGET_TEMPERATURE @ 2023-02-10T08:43:58.037590+01:00>, new_state=<state climate.tze200_2ekuz3dz_ts0601_thermostat=unknown; hvac_modes=[<HVACMode.OFF: 'off'>, <HVACMode.HEAT: 'heat'>], min_temp=7.0, max_temp=30.0, current_temperature=19.2, temperature=None, occupied_heating_setpoint=24000, friendly_name=_TZE200_2ekuz3dz TS0601 Thermostat, supported_features=ClimateEntityFeature.TARGET_TEMPERATURE @ 2023-02-10T08:43:58.037590+01:00>>
2023-02-10 10:09:46.042 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendUnicast: [<EmberStatus.SUCCESS: 0>, 96]
2023-02-10 10:09:46.076 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received messageSentHandler: [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 9327, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=96), 178, <EmberStatus.SUCCESS: 0>, b'']
2023-02-10 10:09:46.076 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 9327, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=96), 178, <Ember [... cut ...] ERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=103), 185, <EmberStatus.SUCCESS: 0>, b'']
2023-02-10 10:10:00.140 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 9327, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=103), 185, <EmberStatus.SUCCESS: 0>, b'']
```
</details>

dala318, 2023-02-10:
Likely does not matter, but the device I have installed is the "3A Boiler" variant. See the zip you provided is named "electric heating". Maybe only relevant for documentation later on...

javicalle, 2023-02-10:
> In the below log-snip I increased the set temperature to 25.0 deg C by the buttons on the device.

The device reports the change:
```
2023-02-10 10:09:46.324 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Received ZCL frame: b'\t\x04\x02\x00\x01\x10\x02\x00\x04\x00\x00\x00\xfa'
2023-02-10 10:09:46.324 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=4, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-10 10:09:46.324 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Decoded ZCL frame: MoesBHTManufCluster:set_data_response(param=Command(status=0, tsn=1, command_id=528, function=0, data=[4, 0, 0, 0, 250]))
2023-02-10 10:09:46.324 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Received command 0x02 (TSN 4): set_data_response(param=Command(status=0, tsn=1, command_id=528, function=0, data=[4, 0, 0, 0, 250]))
2023-02-10 10:09:46.325 DEBUG (MainThread) [zhaquirks.tuya] [0x246f:1:0xef00] Received value [0, 0, 0, 250] for attribute 0x0210 (command 0x0002)
2023-02-10 10:09:46.325 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0x246F:1:0x0201]: Attribute report 'MoesBHTThermostat'[occupied_heating_setpoint] = 25000
2023-02-10 10:09:46.325 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending reply header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=False, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=4, command_id=<GeneralCommand.Default_Response: 11>, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-02-10 10:09:46.325 DEBUG (MainThread) [zigpy.zcl] [0x246F:1:0xef00] Sending reply: Default_Response(command_id=2, status=<Status.SUCCESS: 0>)
```

That would confirm that is related to `is_manufacturer_specific=0`.

Maybe...
maybe we can try overwritting the `write_attributes` in the `MoesBHTManufCluster` like that:
```python
class MoesBHTManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some electric heating thermostats."""

    attributes = {
        MOESBHT_TARGET_TEMP_ATTR: ("target_temperature", t.uint32_t, True),
        MOESBHT_TEMPERATURE_ATTR: ("temperature", t.uint32_t, True),
        MOESBHT_SCHEDULE_MODE_ATTR: ("schedule_mode", t.uint8_t, True),
        MOESBHT_MANUAL_MODE_ATTR: ("manual_mode", t.uint8_t, True),
        MOESBHT_ENABLED_ATTR: ("enabled", t.uint8_t, True),
        MOESBHT_RUNNING_MODE_ATTR: ("running_mode", t.uint8_t, True),
        MOESBHT_CHILD_LOCK_ATTR: ("child_lock", t.uint8_t, True),
    }

    }

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

    def _update_attribute(self, attrid, value):
        super()._update_attribute(attrid, value)
        if attrid == MOESBHT_TARGET_TEMP_ATTR:
```

You will need to import the `foundation` package like that:
```python
from zigpy.zcl import foundation
```

MattWestb, 2023-02-10:
@javicalle This is also one Telink chip on the new Zigbee module so we is getting more problems in the near future with working MCU device that getting new modules :-((

dala318, 2023-02-20:
Thought to leave a bit of update after connecting my other 3 thermostats of the same make and model (all 4 bought in one batch). So adding all 4 of them they show some different behavior. 
One more look exactly as the first while the other two show some more information, while still not correct.
[image]
I can change enter set-temp and operation-state but after a few seconds it falls back to the values shown in image
[zha-2b0fd529875b6d880667f0217196a97f-_TZE200_2ekuz3dz TS0601-bd709337882c44a1b47cd8489acefff0.json.txt](https://github.com/zigpy/zha-device-handlers/files/10788022/zha-2b0fd529875b6d880667f0217196a97f-_TZE200_2ekuz3dz.TS0601-bd709337882c44a1b47cd8489acefff0.json.txt)

dala318, 2023-03-31:
Hi, any progress on this topic?

jacekk015, 2023-04-21:
You could try quirk ts0601_thermostat_avatto.py
from
https://github.com/jacekk015/zha_quirks

Lederhaut, 2023-04-24:
Thank you @jacekk015! The quirk works very well for me.

dala318, 2023-04-28:
Confirming it's working for me as well
[image]

closed 2023-08-16 by dala318

gbrd, 2023-10-29:
For me too. But it seems to be a problem with calibration. If I set +10°C the device understand +1°C. 
I added *10 at line :
```
            await AvattoManufClusterSelf[
                self.endpoint.device.ieee
            ].endpoint.tuya_manufacturer.write_attributes(
                {AVATTO_TEMP_CALIBRATION_ATTR: value*10},
                manufacturer=None,
            )
```
and :
```
        elif attrid == AVATTO_TEMP_CALIBRATION_ATTR:
            self.endpoint.device.AvattoTempCalibration_bus.listener_event(
                "set_value", value/10
            )
```

And it seems better

jacekk015, 2023-10-29:
@gbrd To help other you need to provide DEBUG level logs.
https://github.com/jacekk015/zha_quirks#in-case-of-errors-or-to-support-new-function
Without that I can only say that no other users complain about that.

gbrd, 2023-10-29:
Right ;) I will try to do that 

In the meantime, maybe @dala318 and @Lederhaut can check they have the same behavior?

Lederhaut, 2023-10-30:
Hello @gbrd,
I can confirm, set it to 99, local temperature went up 10 degrees, here are the (relevant) log entries:
``` log
2023-10-30 14:06:54.229 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.thermostat_2_kuche_thermostat: Attribute 'local_temperature' = 2210 update

2023-10-30 14:07:03.271 DEBUG (MainThread) [homeassistant.components.zha.websocket_api] Set attribute for: cluster_id: [13] cluster_type: [in] endpoint_id: [3] attribute: [85] value: [99] manufacturer: [None] response: [([WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)],)]

2023-10-30 14:07:31.391 DEBUG (MainThread) [homeassistant.components.zha.entity] climate.thermostat_2_kuche_thermostat: Attribute 'local_temperature' = 3290 update

2023-10-30 14:07:37.807 DEBUG (MainThread) [homeassistant.components.zha.websocket_api] Read attribute for: cluster_id: [513] cluster_type: [in] endpoint_id: [1] attribute: [16] manufacturer: [None] response: [99] failure: [{}],

2023-10-30 14:07:42.141 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xF954:1:0x0201]: Attribute report 'AvattoThermostat'[local_temperature] = 3290

```

jacekk015, 2023-10-30:
Quirk updated
ts0601_thermostat_avatto.py