zigpy/zha-device-handlers #2189: [Device Support Request] Xiaomi Aqara E1 motion sensor

jcyyu, 2023-02-12:
**Is your feature request related to a problem? Please describe.**
No quirks avaliable for lumi.motion.acn001 .

**Describe the solution you'd like**
Please add support of Aqara E1 motion sensor RTCGQ15LM

Reference:
https://www.zigbee2mqtt.io/devices/RTCGQ15LM.html
https://github.com/Koenkk/zigbee2mqtt/issues/13479

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.acn001",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.2.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.0-53-generic",
    "run_as_root": true
  },
  "custom_components": {
    "jellyfin": {
      "version": "1.1.1",
      "requirements": [
        "jellyfin-apiclient-python==1.7.2"
      ]
    },
    "hisense_tv": {
      "version": "22.05.09",
[... 293 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

luisiam, 2023-08-29:
I am looking forward for it to be supported as well

shanghaiultra, 2023-10-18:
Would be most grateful if this device could be supported. I have a few spare sensors laying around which I can't use with Home Assistant - these devices are cheap and effective and used by many.

manymuch, 2024-05-22:
Please check this PR https://github.com/zigpy/zha-device-handlers/pull/3163. It works for me.

closed 2024-05-28 by TheJulianJES