zigpy/zha-device-handlers #2190: _TZE200_3towulqd TS0601 Tuya motion+illuminance sensor, cannot set keep time and sensitivity [BUG]

jure123, 2023-02-12:
**Describe the bug**
I've been using the quark that @javicalle suggested for the _TZE200_3towulqd TS0601 motion+illuminance sensor. Sensor is working fine, but I cannot set keep_time and sensitivity settings. Tried on multiple devices. 

I am aware the device is a low power device and reading/setting values can be difficult, and were trying to write values when sersor detected motion (like written here: https://github.com/jaspernlp/zigbee2mqtt.io/blob/master/docs/devices/ZG-204ZL.md#reading-and-setting-values)

I also tried to set my devices with Zigbee2Mqtt, where setting keep_time and sensitivity values worked fine. Only in ZHA not working.

**To Reproduce**
1. Installed ZHA and add the ZHA quirk as suggest here - https://github.com/zigpy/zha-device-handlers/issues/1599
2. Tried to write **keep_time** of value **2** (60 seconds). I was repeatedly and rapidly pushing the write attribute button when sensor was changing occupancy state to ensure the data should be received by the device (pushing the write repeadely and quickly 2-times per second).
3. Pushing the read attribute button showed "**OnTimeValues._60_SEC**" as expected (however I suspect the value was read from some internal cache, because pushing the read button always shows the value immediately, no matter if sensor is changing state or is inactive). Not sure if I really receive the actual value stored on the device.
4. **The sensor still cleared the occupancy state after 15 seconds** (as a default keep time value), and not after 60 seconds as it would be expected.
5. Resetting HASS, and reading keep time attribute still shows **OnTimeValues._60_SEC**, but the device still clearing state after 15 seconds. I suspect the read attribute button show value from some internal cache? (because pushing the read button always shows the value immediately, no matter if sensor is changing state or is inactive).

<details>
<summary>Debug log</summary>

```
2023-02-12 13:15:57.621 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] write_attributes --> record: Attribute(attrid=0xEF0A, value=TypeValue(type=OnTimeValues, value=<OnTimeValues._60_SEC: 2>))
2023-02-12 13:15:57.623 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_name='tuya_manufacturer', cluster_attr='keep_time', attr_value=<OnTimeValues._60_SEC: 2>, expect_reply=False)
2023-02-12 13:15:57.623 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] get_dp_mapping --> found DP: 10
2023-02-12 13:15:57.623 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] from_cluster_data: {10: DPToAttributeMapping(ep_attribute='tuya_manufacturer', attribute_name='keep_time', dp_type=<TuyaDPType.ENUM: 4>, converter=<function PirMotionManufCluster.<lambda> at 0x7f9725f370>, dp_converter=None, endpoint_id=None)}
2023-02-12 13:15:57.624 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] value: OnTimeValues._60_SEC
2023-02-12 13:15:57.624 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] raw: b'\x02'
2023-02-12 13:15:57.625 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] tuya_commands: [TuyaCommand(status=0, tsn=125, datapoints=[TuyaDatapointData(dp=10, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x02', *payload=<enum8.undefined_0x02: 2>))])]
2023-02-12 13:15:57.628 DEBUG (MainThread) [homeassistant.components.zha.core.device] [0x218F](TS0601): set: 2 for attr: 61194 to cluster: 61184 for ept: 1 - res: [[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]
2023-02-12 13:15:57.629 DEBUG (MainThread) [homeassistant.components.zha.api] Set attribute for: cluster_id: [61184] cluster_type: [in] endpoint_id: [1] attribute: [61194] value: [2] manufacturer: [None] response: [[[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]]
2023-02-12 13:15:57.631 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] Sending request header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4417, tsn=126, command_id=0, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
2023-02-12 13:15:57.632 DEBUG (MainThread) [zigpy.zcl] [0x218F:1:0xef00] Sending request: set_data(data=TuyaCommand(status=0, tsn=125, datapoints=[TuyaDatapointData(dp=10, data=TuyaData(dp_type=<TuyaDPType.ENUM: 4>, function=0, raw=b'\x02', *payload=<enum8.undefined_0x02: 2>))]))
2023-02-12 13:15:57.634 DEBUG (MainThread) [bellows.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x218F), dst_ep=1, source_route=None, extended_timeout=False, tsn=126, profile_id=260, cluster_id=61184, data=Serialized[b'\x05A\x11~\x00\x00}\n\x04\x00\x01\x02'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-12 13:15:57.635 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f97adf440>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (a4:c1:38:5f:56:df:dc:b0, 1, 61184, 61194, '2'))
2023-02-12 13:15:57.634 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x218f, EmberApsFrame(profileId=260, clusterId=61184, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=126), 127, b'\x05A\x11~\x00\x00}\n\x04\x00\x01\x02')
2023-02-12 13:15:57.636 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'318821a9602a153d78904b25455493099d4e27d592c262caecb8638981743ba7eacfadc67e'
2023-02-12 13:15:57.637 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f97adf440>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (a4:c1:38:5f:56:df:dc:b0, 1, 61184, 61194, '2')) completed
2023-02-12 13:15:57.642 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f97adf440>)
2023-02-12 13:15:57.642 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f97adf440>) completed
2023-02-12 13:15:57.644 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f97adf440>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (a4:c1:38:5f:56:df:dc:b0, 1, 61184, 61194, <OnTimeValues._60_SEC: 2>))
2023-02-12 13:15:57.645 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7f97adf440>, 'INSERT INTO attributes_cache_v11 VALUES (?, ?, ?, ?, ?)\n                    ON CONFLICT (ieee, endpoint_id, cluster, attrid)\n                    DO UPDATE SET\n                        value=excluded.value WHERE value != excluded.value', (a4:c1:38:5f:56:df:dc:b0, 1, 61184, 61194, <OnTimeValues._60_SEC: 2>)) completed
2023-02-12 13:15:57.647 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1488a1a9602a156e74187e'
2023-02-12 13:15:57.647 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2023-02-12 13:15:57.648 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received sendUnicast: [<EmberStatus.SUCCESS: 0>, 220]
2023-02-12 13:15:57.649 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f97adf440>)
2023-02-12 13:15:57.650 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7f97adf440>) completed
2023-02-12 13:15:58.577 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2488b1a9112a15b65894a524ab5593499cdbb76b62ef9874f0cf628bfc5b33a5ebc9de6f8c7da8c17e'
[... cut ...]
</details>

BTW: in the logs I noticed below line that sends the keep time attribute to the device, and getting `Status.SUCCESS: 0` response. Should this line get `Status.SUCCESS: 1` on successful update? I tried for several times and logs only contain Status.SUCCESS: 0.

`[homeassistant.components.zha.api] Set attribute for: cluster_id: [61184] cluster_type: [in] endpoint_id: [1] attribute: [61194] value: [2] manufacturer: [None] response: [[[WriteAttributesStatusRecord(status=<Status.SUCCESS: 0>)]]]`

javicalle, 2023-02-12:
A couple of comments:
* Which HA version are you running? You have an `UNSUP_MANUF_CLUSTER_COMMAND` that must not be there.
* That `Status.SUCCESS: 0` is the Status identifier (and not a boolean value). All the posible values are defined here:
https://github.com/zigpy/zigpy/blob/819f8e7dc97e27627335a398dac22ad9a9b7688a/zigpy/zcl/foundation.py#L25-L62
* Try to remove the battery from device and put it on again. At this moment usually the device reports all the current values (and its DPs). That information would give us some clues about what can be happening.

jure123, 2023-02-12:
I'm running Home Assistant 2023.2.3 (and the same problem on older versions from 2023, but before that I didn't have these sensor devices yet).

I removed the battery from device, after that it showed keep_time OnTimeValues._10_SEC.

jure123, 2023-02-12:
This is my full diagnostic:

<details>

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Vienna",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_miot": {
      "version": "0.7.5",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.6",
        "micloud>=0.3"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.7",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.92",
      "zigpy-deconz==0.19.2",
      "zigpy==0.53.0",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
[... 254 lines cut ...]
                "value": 20935.21685162235
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

javicalle, 2023-02-12:
I'm not sure  which is the exact version for your quirk but I would say that the `UNSUP_MANUF_CLUSTER_COMMAND` is related t the Tuya `NO_MANUFACTURER_ID` issue...

I would suggest to try with this changes:
1. import the `foundation` package:
```python
from zigpy.zcl import foundation
```
2. update the `PirMotionManufCluster` class adding the `write_attributes` method like that:
```python
class PirMotionManufCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF0A: ("keep_time", OnTimeValues)})

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        .../...
```

Save changes, delete any `__pycache__` folder and restart HA.
Check if the `UNSUP_MANUF_CLUSTER_COMMAND` error is still showing up in your traces.

jure123, 2023-02-12:
Your suggestion fixed my issue!

Here is my fixed quirk:

<details>

<summary>ts0601_pirmotion.py</summary>

```py
"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Ota, Time
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl import foundation

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaDPType,
    TuyaMCUCluster,
    TuyaPowerConfigurationCluster,
)

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class SensitivityLevel(t.enum8):
    """Sensitivity level enum."""

    LOW = 0x00
    MEDIUM = 0x01
    HIGH = 0x02

class OnTimeValues(t.enum8):
    """Sensitivity level enum."""

    _10_SEC = 0x00
    _30_SEC = 0x01
    _60_SEC = 0x02
    _120_SEC = 0x03

class PirMotionManufCluster(TuyaMCUCluster):
    """Neo manufacturer cluster."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update({0xEF09: ("sensitivity_level", SensitivityLevel)})
    attributes.update({0xEF0A: ("keep_time", OnTimeValues)})

    async def write_attributes(self, attributes, manufacturer=None):
        """Overwrite to force manufacturer code."""

        return await super().write_attributes(
            attributes, manufacturer=foundation.ZCLHeader.NO_MANUFACTURER_ID
        )

    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOccupancySensing.ep_attribute,
            "occupancy",
            dp_type=TuyaDPType.BOOL,
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,
        ),
        4: DPToAttributeMapping(
            TuyaPowerConfigurationCluster.ep_attribute,
            "battery_percentage_remaining",
            dp_type=TuyaDPType.VALUE,
        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "sensitivity_level",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: SensitivityLevel(x),
        ),
        10: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "keep_time",
            dp_type=TuyaDPType.ENUM,
            converter=lambda x: OnTimeValues(x),
        ),
        12: DPToAttributeMapping(
            TuyaIlluminanceMeasurement.ep_attribute,
            "measured_value",
            dp_type=TuyaDPType.VALUE,
            converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        4: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        10: "_dp_2_attr_update",
        12: "_dp_2_attr_update",
    }

class PirMotion(CustomDevice):
    """Tuya PIR motion sensor."""

    signature = {
        MODELS_INFO: [("_TZE200_3towulqd", "TS0601")],
        ENDPOINTS: {
            # endpoints=1 profile=260 device_type=0x0402
            # in_clusters=[0x0000, 0x0001, 0x0500],
            # out_clusters=[0x000a, 0x0019]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster.cluster_id,
                    IasZone.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TuyaPowerConfigurationCluster,
                    PirMotionManufCluster,
                    TuyaOccupancySensing,
                    TuyaIlluminanceMeasurement,
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            }
        }
    }
```

</details>

TheJulianJES, 2023-02-12:
Let's keep this open until it's fixed in zha-quirks (without using a custom quirk).

jack5mikemotown, 2023-02-19:
After using this fixed quirk I can no longer see Motion, I can still see Illuminance and Occupancy

jure123, 2023-02-19:
> After using this fixed quirk I can no longer see Motion, I can still see Illuminance and Occupancy

@jack5mikemotown, can you post your previous version of your quirk that worked for you?

javicalle, 2023-02-19:
Neither the device or the quirk declares a `Motion` cluster. Maybe you were using another quirk before?

adi90x, 2023-03-01:
Hello,

Seems with last release TuyaDPType is not in mcu anymore so quirk need some ajustement...

Regards,

javicalle, 2023-03-02:
Just remove all the `dp_type=TuyaDPType.XXXXXX` from the `dp_to_attribute` part:
```python
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            MotionSensorCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,   # <-- REMOVE FROM ALL DPs
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,
        ),
        .../...
```

timm-severin, 2023-08-14:
Hi, I have the same sensor, but apparently I have quite different clusters:

```json
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0402",
          "input_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0400",
            "0x0500",
            "0xe002",
            "0xef00"
          ],
          "output_clusters": []
        }
      },
      "manufacturer": "_TZE200_3towulqd",
      "model": "TS0601"
    },
```

I cannot get the quirk to work, I removed the output clusters, but I also do not really know how to deal with `0xe002` and `0xef00`. Tried removing them and whatnot, but no chance.

When I try to add the device my zha log prints the following lines:
```
[...]
2023-08-14 18:45:56.804 INFO (MainThread) [zigpy.device] [0x124b] Read model 'TS0601' and manufacturer '_TZE200_3towulqd' from <Endpoint id=1 in=[basic:0x0000, identify:0x0003, ias_zone:0x0500, None:0xE002, None:0xEF00, power:0x0001, illuminance:0x0400] out=[] status=<Status.ZDO_INIT: 1>>
2023-08-14 18:45:56.804 INFO (MainThread) [zigpy.device] [0x124b] Discovered basic device information for <Device model='TS0601' manuf='_TZE200_3towulqd' nwk=0x124B ieee=**** is_initialized=True>
2023-08-14 18:45:56.805 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE200_3towulqd' nwk=0x124B ieee=**** is_initialized=True>
2023-08-14 18:45:56.809 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE200_3towulqd TS0601 (a4:c1:38:0b:1e:e7:64:86)
2023-08-14 18:45:56.809 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_pirmotion.PirMotion'>
2023-08-14 18:45:56.809 DEBUG (MainThread) [zigpy.quirks.registry] Fail because input cluster mismatch on at least one endpoint
[...]
```

So at least I know that it tries the quirk.

closed 2024-02-17 by github-actions[bot]