zigpy/zha-device-handlers #2192: [BUG] Power delivery summation sensor not updated on Ubisys S1 switch

grubersh, 2023-02-13:
**Describe the bug**
The power summation in kWh is not update for the Ubisys S1 switch (https://www.ubisys.de/produkte/produkte-licht-verbraucher/produkte-licht-verbraucher-s1/). It is only updated on reboot. Firmware is the newest one available (ubisys S1 Firmware Version 2.1.2, ubisys Zigbee Stack 4.0.4 (Hardware Revision >= 2). 

**To Reproduce**
Steps to reproduce the behavior:
1. read current summation sensor value in Home Assistant / ZHA
2. Open device 
3. Open sensor and read summation value

**Expected behavior**
The summation should constantly increase based on the current summation of kWh used.

**Screenshots**

[image]

<details>
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4338, maximum_buffer_size=127, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": []
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0001",
      "in_clusters": [
        "0x0000",
        "0x0003"
      ],
      "out_clusters": [
        "0x0005",
        "0x0006",
        "0x0008"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0501",
      "in_clusters": [
        "0x0000",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": []
    },
    "232": {
      "profile_id": 260,
      "device_type": "0x0507",
      "in_clusters": [
        "0x0000",
        "0x0015",
        "0xfc00"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
[... 848 lines cut ...]
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

TheJulianJES, 2023-02-18:
Can you send a screenshot of the reconfiguration dialog? It looks like the device doesn't send attribute reports properly and ZHA would need to poll it.

freakyjoe8, 2023-02-19:
I've also written to Ubisys support and asked them if this is maybe a firmware bug. They replied no and they stated that they are following Zigbee standards here and sent me this screenshot:

[image]

Not sure what this means as I'm no Zigbee expert. But I'd also suspect this error to occur on their whole switch product line (S1, S1-R, S2, S2-R).

**Screenshot of the reconfigure dialog**

[image]

freakyjoe8, 2023-03-03:
I've worked around this now by polling the attribute manually every 5 minutes via ZHA toolkit and creating an automation. This is the service call if someone else also has this problem.

```json
service: zha_toolkit.attr_read
data:
  ieee: 00:1f:ee:00:00:00:xx:xx
  endpoint: 3
  cluster: 1794
  attribute: 0
  use_cache: false
  state_attr: sensor.steckdose_quooker_summation_delivered
```

closed 2023-09-06 by github-actions[bot]