zigpy/zha-device-handlers #2193: [BUG] TS0601 dimmer not working anymore

Tropaion, 2023-02-13:
**Describe the bug**
I'm not sure since which update the dimmer is not working anymore, but it's definitely a while now.
Displaying state (receiving information) works, but setting a state in the dashboard is not working an gives an error.
@javicalle Maybe you know something about this, since you are the most experienced one with these dimmers :D.

**To Reproduce**
Enable/Disable dimmer or set dimmer level

**Expected behavior**
Change state of dimmer

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_3p5ydos3",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaSingleSwitchDimmerGP"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

Setting dimmer level:
```yaml
2023-02-13 21:48:03.850 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0x0000] Sending reply: Default_Response(command_id=10, status=<Status.SUCCESS: 0>)
2023-02-13 21:48:03.851 DEBUG (MainThread) [zigpy_znp.zigbee.application] Sending packet ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x858F), dst_ep=1, source_route=None, extended_timeout=False, tsn=156, profile_id=260, cluster_id=0, data=Serialized[b'\x18\x9c\x0b\n\x00'], tx_options=<TransmitOptions.ACK: 1>, radius=0, non_member_radius=0, lqi=None, rssi=None)
2023-02-13 21:48:03.852 DEBUG (MainThread) [zigpy_znp.api] Sending request: AF.DataRequestExt.Req(DstAddrModeAddress=AddrModeAddress(mode=<AddrMode.NWK: 2>, address=0x858F), DstEndpoint=1, DstPanId=0x0000, SrcEndpoint=1, ClusterId=0, TSN=156, Options=<TransmitOptions.SUPPRESS_ROUTE_DISC_NETWORK|ACK_REQUEST: 48>, Radius=0, Data=b'\x18\x9C\x0B\x0A\x00')
2023-02-13 21:48:03.872 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataRequestExt.Rsp(Status=<Status.SUCCESS: 0>)
2023-02-13 21:48:03.932 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.DataConfirm.Callback(Status=<Status.SUCCESS: 0>, Endpoint=1, TSN=156)
2023-02-13 21:48:13.606 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139825703042976] NoManufacturerCluster.command() got an unexpected keyword argument 'level'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1787, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1824, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 680, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 968, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 720, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 571, in async_handle_light_on_service
    await light.async_turn_on(**filter_turn_on_params(light, params))
  File "/usr/src/homeassistant/homeassistant/components/zha/light.py", line 320, in async_turn_on
    result = await self._level_channel.move_to_level_with_on_off(
  File "/usr/src/homeassistant/homeassistant/components/zha/core/channels/base.py", line 78, in wrapper
    result = await command(*args, **kwds)
TypeError: NoManufacturerCluster.command() got an unexpected keyword argument 'level'
```

Turning dimmer off:
```yaml
2023-02-13 21:48:27.279 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0x0006] Setting the NO manufacturer id in command: 0
2023-02-13 21:48:27.280 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0x0006] Sending Tuya Cluster Command... Cluster Command is 0, Arguments are ()
2023-02-13 21:48:27.280 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] tuya_mcu_command: cluster_data=TuyaClusterData(endpoint_id=1, cluster_attr='on_off', attr_value=0, expect_reply=True, manufacturer=-1)
2023-02-13 21:48:27.281 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] get_dp_mapping --> found DP: 1
2023-02-13 21:48:27.281 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] from_cluster_data: 1, DPToAttributeMapping(ep_attribute='on_off', attribute_name='on_off', dp_type=<TuyaDPType.BOOL: 1>, converter=None, dp_converter=None, endpoint_id=None)
2023-02-13 21:48:27.281 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] ztype: Bool.false
2023-02-13 21:48:27.282 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] from_value: [1, 0]
2023-02-13 21:48:27.282 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] raw: b'\x00'
2023-02-13 21:48:27.282 DEBUG (MainThread) [zigpy.zcl] [0x858F:1:0xef00] tuya_command: TuyaCommand(status=0, tsn=102)
2023-02-13 21:48:27.289 ERROR (MainThread) [zigpy.zcl] [0x858F:1:0xef00] Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy/quirks/__init__.py", line 180, in command
    return await self.request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 304, in request
    hdr, request = self._create_request(
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 266, in _create_request
    request.serialize()  # Throw an error before generating a new TSN
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 257, in serialize
    chunks.append(value.serialize())
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 231, in serialize
    for field, value in self.assigned_fields(strict=True):
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/struct.py", line 177, in assigned_fields
    raise ValueError(
ValueError: Value for field 'datapoints' is required: TuyaCommand(status=0, tsn=102)
2023-02-13 21:48:27.296 WARNING (MainThread) [homeassistant.components.automation] Error evaluating condition in 'EG_LIGHT_STATE_OFF':
In 'condition' (item 4 of 4):
  In 'state':
    In 'state' condition: unknown entity switch.stiegenhaus_deckenlicht_on_off
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2023-02-16:
I believe that you are using a local quirk for this device.
Your device seems to be already supported, so you should remove the local quirk and use the current version of the quirk.

Tropaion, 2023-02-16:
@javicalle Well, thats awkward, you`re right, looks like I forgot to remove it.

closed 2023-02-16 by Tropaion