zigpy/zha-device-handlers #2194: [BUG]Aqara switch (lumi.switch.b1naus01) shows up as Light

larry-glz, 2023-02-13:
**Describe the bug**
Aqara switch (lumi.switch.b1naus01) shows up as "light" when added via ZHA.  It the "device_type" in the signature shows up as 0x0100 (ON_OFF_LIGHT), but it should be 0x0000 (ON_OFF_SWITCH).  

**To Reproduce**
Add via ZHA

**Expected behavior**
Device should show a "switch" instead of a "light" 

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.b1naus01",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
I've added a separate switch (lumi.switch.b2naus01 - double rocker) and both rockers show as device_type 0x0000.

TheJulianJES, 2023-02-23:
See https://www.home-assistant.io/integrations/zha/#modifying-the-device-type for fixing this yourself.

Add the following to your config (replace `IEEE` with the IEEE of the affected Aqara switch):
```yaml
zha:
  device_config:
    IEEE-1:  # format: ieee-endpointId
      type: "switch"  # corrected device type
```

We could also change this with a quirk, but this would break the implementation for everyone using it as a light.
As this happens with a variety of Xiaomi/Aqara and Tuya devices, it's best not to add a quirk changing the device type for everything.

larry-glz, 2023-02-23:
This worked.  Thanks!

closed 2023-02-23 by larry-glz