zigpy/zha-device-handlers #2196: [BUG] Tuya TRV TS0601 _TZE200_hhrtiq0x get no reply to timesync (command x24) request

rpsainio, 2023-02-15:
I was testing above mentioned TRV and sniffing Zigbee traffic and noticed that 0x24 does not get answered. Switching on debugging revealed following:
 zigpy.zcl[83] DEBUG [0x062E:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, directio
n=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=50, command_id=36, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
 zigpy.zcl[83] DEBUG [0x062E:1:0xef00] Decoded ZCL frame: SiterwellManufCluster:set_time_request(param=[0, 52])
 zigpy.zcl[83] DEBUG [0x062E:1:0xef00] Received command 0x24 (TSN 50): set_time_request(param=[0, 52])
 zigpy.zcl[83] DEBUG [0x062E:1:0xef00] **No explicit handler for cluster command 0x24**: set_time_request(param=[0, 52])

As I used git clone I built in some additional debugging and it seems that "set_time_offset" is not set (has value 0) and this will lead to mentioned errormsg. I changed it to 1970 and verified that now setting the time works.

Remark: I do not have Home Assistent in use rather than I use only the lower layers: zigpy, zigpy-<radio>  and zha-device-handlers directly.

MattWestb, 2023-02-15:
That is interesting !!!

I have one first gen devices with the old `TYST11` Zigbee module and i have not seen that error (perhaps not looking enough).

Can you posting where you was doing the change and i can testing if its working with the old ones ??

Also can you posting the 6 first digits (or all) of your device MAC / IEEE address then i like to knowing if its one new T-Link chips in the Zigbee module ?

rpsainio, 2023-02-15:
I changed set_time_offset = 1970  in tuya/__init__.py for testing purposes, but I think it should done in tuya/ts0601_trv.py as I can see that it gets set once for MoesManufCluster. As the request comes only once in an hour I used the brute force just to verify :)
Here is the partial MAC 84:ba:20

MattWestb, 2023-02-15:
Thanks i shall testing patching my test setup and see its its working OK.

The MAC is one Silicon Laboratories so its one good Zigbee module then the new T-Link is acting little strange and we need leaning how to fixing them.

I reporting back and then taking one race with the devs for implanting it nice.

rpsainio, 2023-02-15:
I was thinking the same place and I can test that on my clone

rpsainio, 2023-02-15:
Seems to work, however using  1970 instead of 2000 was just my guess. 
class SiterwellManufCluster(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of some thermostatic valves."""
    set_time_offset = 1970

    attributes = TuyaManufClusterAttributes.attributes.copy()

MattWestb, 2023-02-15:
Great !!
This TRVs dont have schedule in the hardware so if like doing it you is using HAs automatons so i think its not important if its 1970 or 2000 then its not being so intelligent to knowing the different but i think its good giving it one real time then its requesting it.

MattWestb, 2023-02-15:
I have made on PR for getting it in main ZHA and you dont need patching HA container and using local quirks in the future.

Testing if 2000 is working OK and i changing the PR.

MattWestb, 2023-02-15:
If the `0x0007` attribute is on time cluster its Zigbee standard and setting the  time in the Zigbee module. The `x24` request is from the tuya MCU that also like having its time set and is having more version of offset (= tuya standard).

rpsainio, 2023-02-15:
It seems to me that this TRV uses two methods to get localtime:
- Tuya standard command x24
- Zigbee standard reading 0x0007 attribute
Both requests come from Tuya TRV

TheJulianJES, 2023-02-15:
Just to confirm, https://github.com/zigpy/zha-device-handlers/pull/2197 with ``1970`` (instead of ``2000``) is correct and enough to fix this issue?

rpsainio, 2023-02-15:
Well, I do not know (1970 or 2000) and we miss a trace from original Tuya gateway. In the same file there is another place with 1970. Have to search Internet for an  example. @MattWestb  how was your olderTRV ?

MattWestb, 2023-02-15:
I was testing 1970 and its not complaining but i have not sniffing it.
Then the Moes is using 1970 i think its the right setting then its both the first gen TRVs from tuya.
If the device is acking it OK in the Zigbee network / app layer then you is sniffing it it shall being OK.

PR #2197 (2023-02-16): Adding timesync response for Siterwell RTVs
Fixes: #2196
commit b3b7a58b (2023-02-16): Add timesync response for Siterwell TRVs (#2197)

Fixes #2196
closed 2023-02-16 by TheJulianJES