zigpy/zha-device-handlers #2199: [Device Support Request] Aqara H1 single Rocker with Neutral WKS-03 lumi.switch.n1aeu1

ditheo, 2023-02-16:
i have this one , althought I can control with ZHA for on-off , the temperature shows only 0.3 'C and i can only make single tap , double tap and triple tap does not work. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0702",
        "0x0b04"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.n1aeu1",
  "class": "zigpy.device.Device"
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

j0hann3s, 2023-02-27:
I am also interested in this. For the person looking into a quirk for this device, I posted some further information in this reply to an open issue: https://github.com/zigpy/zha-device-handlers/issues/1369#issuecomment-1446801993
There might be something which can help :smile:

szamfirov, 2023-05-08:
Just raised a PR for this: https://github.com/zigpy/zha-device-handlers/pull/2385

It's work in progress but would appreciate any support.

szamfirov, 2023-05-26:
The PR above was recently merged which should resolve this issue. Feel free to give the quirk a try. I've been using the wall switch in decoupled mode for quite some time and it works like a charm.

closed 2023-11-29 by TheJulianJES