zigpy/zha-device-handlers #2200: [BUG] TS0013 Girier 3-way switch

LordArrin, 2023-02-16:
**Describe the bug**

I have successfully paired a GIRIER 3 gang Zigbee switch. However, it is impossible to turn ON or OFF 1 gang at a time. For example, if I try to turn ON the gang 1 (left) via ZHA all the gangs will turn ON and the same thing happens when turning it OFF.
The same thing happens if i try with the gang 2 and 3.

**Expected behavior**
A clear and concise description of what you expected to happen.
Separated control fo the device's relays
**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
IEEE: a4:c1:38:05:5b:0f:6f:cb
Nwk: 0x3aed
Device Type: EndDevice
LQI: 176
RSSI: -56
Устройство было в сети: 2023-02-16T15:55:12
Источник питания: Battery or Unknown
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Moscow",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "ui_lovelace_minimalist": {
      "version": "v1.1.6-hotfix",
      "requirements": [
        "aiofiles==0.8.0",
        "aiogithubapi>=22.2.4"
      ]
    },
    "xiaomi_gateway3": {
      "version": "3.1.0",
      "requirements": [
        "zigpy>=0.42.0"
      ]
    },
    "ble_monitor": {
      "version": "11.6.2",
      "requirements": [
        "pycryptodomex>=3.14.1",
        "janus>=1.0.0",
[... 464 lines cut ...]
              }
            }
          }
        },
        "out_clusters": {}
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2023-02-16:
I have a quirk porposal for your device.

You would need enable the local quirk folder in your installation. There is a good guide to configure it:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once configured, copy in your `custom_zha_quirks` folder the current `ts001x.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts001x.py

Edit the file and add at the bottom this class:
<details open>
<summary>TuyaTripleGang_var05</summary>

```python
class TuyaTripleGang_var05(EnchantedDevice, TuyaSwitch):
    """Tuya 3 gang light switch (variation 05)."""

    signature = {
        # "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, 
        # user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, 
        # mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, 
        # maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, 
        # descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, 
        # *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, 
        # *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
        MODEL: "TS0013",
        ENDPOINTS: {
            # "profile_id": 260, "device_type": "0x0100",
            # "in_clusters": ["0x0000","0x0003","0x0004","0x0005","0x0006","0xe000","0xe001"],
            # "out_clusters": ["0x000a","0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            },
            # "profile_id": 260, "device_type": "0x0100",
            # "in_clusters": ["0x0004","0x0005","0x0006"],
            # "out_clusters": []
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # "profile_id": 260, "device_type": "0x0100",
            # "in_clusters": ["0x0004","0x0005","0x0006"],
            # "out_clusters": []
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

</details>

Save changes, remove the device from HA, restart HA and pair the device again.

LordArrin, 2023-02-18:
Yeah, it works! Thanks, man! But if it will works out-of-box, it will better)

javicalle, 2023-02-18:
Let's keep open until the quirk is merged to mainstream.
Once the quirk is published wirh a new version, I recommend you to remove the local quirk.

PR #2207 (2023-02-18): Add `TuyaTripleGang_var05` for `TS0013` devices
New `TS0013` quirk variation.

Fixes: #2200
commit f5ed78de (2023-02-18): Add `TuyaTripleGang_var05` for `TS0013` devices (#2207)

* Add `TuyaTripleGang_var05` for `TS0013` devices

New `TS0013` quirk variation.

Fixes: #2200
closed 2023-02-18 by TheJulianJES