zigpy/zha-device-handlers #2202: Aqara FP1 - implement region support

droopanu, 2023-02-16:
**Is your feature request related to a problem? Please describe.**
Would you be able to implement regions  for Aqara FP1?
It makes the sensor actually usable and reduces the false positives.
Zigbee2mqtt has just implemented it but I would rather use zha if possible.

**Describe the solution you'd like**
The following options should be implemented:
- possibility to add/remove regions
- parse region info from sensor and make it available in a event or sensor
- even better, the device could create one sensor for each region and update it's status :D

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0xfff0",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0406",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "aqara",
  "model": "lumi.motion.ac01",
  "class": "zhaquirks.xiaomi.aqara.motion_ac01.AqaraLumiMotionAc01"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

alexisiglesias1, 2023-03-05:
hi @droopanu have you had any updated in this request?

michalmo, 2023-05-16:
I've created PR #2401 that adds support for regions.

pingu2211, 2023-11-12:
The PR resolving this issue remains open

samsonusmc, 2024-01-24:
Would also love to see this implemented in ZHA

BrendanAustinWSP, 2024-07-23:
The PR resolving this issue remains open

masi, 2024-09-06:
Ideally some basic code lives within ZHA that supports regions on a general level. I assume all presence sensors will have the notion of regions.

Wh1t3Rose, 2025-04-20:
Why hasn't this been implement yet?

closed 2025-12-24 by github-actions[bot]