zigpy/zha-device-handlers #2203: [Device Support Request] IKEA controller with 24.X firmware battery readings.

MattWestb, 2023-02-18:
**Is your feature request related to a problem? Please describe.**

All 24.X controllers have fixed battery reporting and dont need the doubling of the % renaming.
Only Styrbar looks reporting 255% and its something wrong in the firmware (it was the first out).

Styrbar and 5 button and old motion sensor update is in the normal IKEA feed.
Dirigera is updating OnOff, OpenClose new motion sensor and blinds and all need being updated.

**Describe the solution you'd like**

The problem is that most of the device is using the same EP / Cluster setting like the privies version so we cant fixing it in the quirk in the normal way.

Is it possible using software version from basic cluster and if its larger then 2.0 we using no battery doubling and if its lower we is doing it (Z2M is start using that) ??
If its possible i suggesting implanting it in IKEA  INIT for not spamming all quirks with the same code.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-02-18:
Matrix before and after updating to 24.X firmware
Device  | Type | TRADFRI | DIRIGERA | Battery | PM
-- | -- | -- | --| -- | --
E1525 | Old Motion | 24.4.5 | 24.4.5 | OK | No Quirk
E1745 | New Motion | 2.0.022 | 24.4.5 | OK | No Quirk
E1743 | OnOff / OpenClose | 2.3.079 | 24.4.5 | 200% | With Quirk
E1524 | Remote 5Bt | 24.4.5 | 24.4.5 | OK
E2001/2| Styrbar | 24.4.5 | 24.4.5 | 255%
E1744 | Symfonisk | 2.3.080 | 24.4.5 | OK | No Quirk
E1812 | Shortcut | 2.3.080 | ? |  ?

All is not made and coming later.

TheJulianJES, 2023-02-18:
Can you also post all the new signatures? (after re-pairing the devices)

MattWestb, 2023-02-18:
I do it then i have getting all updated with Dirigera but i need SWD flashing old firmware on some devices then they is rolling back the latest (OpenClose, Symfonisk and Shortcut) and eating batteries.

Its looks like all is fixed and dont need quirks but then loosing DA and Styrbar is using 0-255 (firmware problem or is it 2 * AAA batteries).

MattWestb, 2023-02-18:
Diag from all firmware and the 2 versions that is using the same (OnOff and OpenClose).
[E1812-IKEA of Sweden TRADFRI SHORTCUT Button.txt](https://github.com/zigpy/zha-device-handlers/files/10774982/E1812-IKEA.of.Sweden.TRADFRI.SHORTCUT.Button.txt)
[E1744-IKEA of Sweden SYMFONISK Sound Controller.txt](https://github.com/zigpy/zha-device-handlers/files/10774983/E1744-IKEA.of.Sweden.SYMFONISK.Sound.Controller.txt)
[E2001-IKEA of Sweden Remote Control N2.txt](https://github.com/zigpy/zha-device-handlers/files/10774986/E2001-IKEA.of.Sweden.Remote.Control.N2.txt)
[E1524-IKEA of Sweden TRADFRI remote control.txt](https://github.com/zigpy/zha-device-handlers/files/10774987/E1524-IKEA.of.Sweden.TRADFRI.remote.control.txt)
[E1766-IKEA of Sweden TRADFRI open_close remote.txt](https://github.com/zigpy/zha-device-handlers/files/10774988/E1766-IKEA.of.Sweden.TRADFRI.open_close.remote.txt)
[E1743-IKEA of Sweden TRADFRI on_off switch.txt](https://github.com/zigpy/zha-device-handlers/files/10774989/E1743-IKEA.of.Sweden.TRADFRI.on_off.switch.txt)
[E1525-IKEA of Sweden TRADFRI old motion sensor.txt](https://github.com/zigpy/zha-device-handlers/files/10774990/E1525-IKEA.of.Sweden.TRADFRI.old.motion.sensor.txt)
[E1745-IKEA of Sweden TRADFRI new motion sensor.txt](https://github.com/zigpy/zha-device-handlers/files/10774991/E1745-IKEA.of.Sweden.TRADFRI.new.motion.sensor.txt)

TheJulianJES, 2023-02-27:
Likely blocked by (for devices which have the same signature):
- https://github.com/zigpy/zigpy/issues/1033

MattWestb, 2023-02-27:
Syrbar reporting 255 % and also voltage that meaning is out of range / not known if looking in ZCL R8 (also with rechargeable batteries).
So for the moment its best fixing the quirk to normal battery (200) and letting it being out of range and waiting for IKEA fixing the device firmware.

jasondefuria, 2023-06-10:
It looks like the E1766 remote does the same on firmware 24.4.6.

MattWestb, 2023-06-10:
The E1766 have not changing any cluster on its endpoints so its loading the quirk and is doubling the power that is not needed for the updated firmware.
The good with the quirk is adding some good DA that some user is using.

I have making one issue for fixing the battery in the quirk but is not possible for the moment and some changes in the core is needed for making it selecting how the batters power is being imprinted.

jasondefuria, 2023-06-10:
@MattWestb That makes sense- I saw the other change mentioned that it will need an upstream based on firmware version. Very interesting! Will have to wait.

MattWestb, 2023-06-19:
Then bot Dirigera and Tradfri is using 24.X in there update feed and ZHA is using the TF feed all user that is updating there devices is getting the same problems with the fixed batteriyregretting.
Many devices is not flashing the update and is looping restarting but its one other problem.
My suggestion is changing the last device class in all device that is loading the quirk with the standard battery reporting and then its working for all device with updated firmware and some is not that is running the old latest firmware but the user can updating the device and all is working OK.

I think OpenClose need update of the quirk for not losing the DA and perhaps some more but i mus looking more (i have all my IKEA controllers updated to latest OTA but have not deleting and adding then new).
Also Styrbar is reporting 255 % = legal / unknown reading but is one firmware problem.

nmeachen, 2023-07-25:
@MattWestb is there a way of working around this with custom quirks until a permanent upstream solution is available? I have the E1743 & E1524.

MattWestb, 2023-07-25:
ZHA cant using the software version from basic cluster for matching the quirks so some device we can see if they is on old or new firmware.
The E1524 shall using the `Quirk: zhaquirks.ikea.fivebtnremote.IkeaTradfriRemote5` then its have getting new clusters but you need deleting it and wait one minute (or restarting HA) and adding it new so the system is getting the new clusters in the DB and it shall working.
For the OnOff and also if you is having OpenCloce you can adding one local quirk that is overriding the ZHA ones but change the replacement using the normal like is being done in the quirk for the E1524 here https://github.com/zigpy/zha-device-handlers/blob/92c9fbc6d01a5d86f78d64183302b906aa7d8215/zhaquirks/ikea/fivebtnremote.py#L430
and do the same in this line
https://github.com/zigpy/zha-device-handlers/blob/92c9fbc6d01a5d86f78d64183302b906aa7d8215/zhaquirks/ikea/twobtnremote.py#L178
for the OnOff switch and it shall working.

I have not doing it in my production and test systems then i like to see  if the battery draining issues / bugs is fixed and its looks mush better then the old firmware's but i was not having so large problems with it as some other users was having.

sergeykad, 2023-08-15:
With the new 0x24040011 firmware, my Fyrtur reporting battery at 84% in the diagnostic section and 34% for the roller blind battery sensor. 
Previously it was over 100% for the battery and the roller blind battery sensor was zero.

It is an improvement, but I am confused what the actual battery percentage.

MattWestb, 2023-08-15:
The Zigbee standard is reporting 1/2° and IKEA was doing wrong in the first and second gen devices firmware and was using 1° but it fixed the the 24.X release so now most IKEA controllers is working without quirks.

Mrcs-o, 2023-08-24:
Hey everyone, I also have the E1743 and tried the modifications, that @MattWestb suggested.
It is running on the latest Firmware 0x24040006.
I applied a custom quirk, which only changes the doubling of the Battery thingy (sorry I am not really familiar with this kind of coding) and the Switches (4 in total) loaded the custom quirk with no problem. After some time, when the first checkin was fired after 55 minutes it seems like the correct battery status is reported with each checkin.
Is there a way of uploading the custom quirk, so everybody else with this kind of problem and the E1743 can solve this problem?
Because I think it will take some time, until the fix is applied globally doesn't it?

[image]

MattWestb, 2023-08-24:
I have suggesting changing the quirk in ZHA for the last firmware but then its wrong for the latest 2.3X firmware but user can updating there device and its no problems (if OTA is working OK on the device) but i have not getting any response from the maintainers if shall doing it or not.
Great that you have all working OK with little work.
I have not doing it in m production system then its very clean and my test system is not needed then i have verifying its working OK and using  testing all IKEA controllers i have (i dont  have the KNYCKLAN Open/Close remote (E1841) but its the same hardware as the 2 button and no OTA released for it).

TheJulianJES, 2024-02-03:
This issue should be fixed in HA Core 2024.2.0 (currently in beta).
Relevant PR:
- https://github.com/zigpy/zha-device-handlers/pull/2948

If you still see new "double battery percentage" reports, wait a day.
You can also try to press a button on the device or manually read the `sw_build_id` attriubte. Re-pairing/reconfiguring should also work.

If it's not fixed for you, please create a new issue and provide a new diagnostics file.

closed 2024-02-03 by TheJulianJES