zigpy/zha-device-handlers #2204: [Device Support Request]

cayote64, 2023-02-18:
The problem
On the 1 gang Moes TS0011 switches there is no option to change the backlight status which on the 1 gang switches are all the opposite of what they should be, i.e go out when light off so you cant see the switch in the dark.

The 2 gang Moes switches ( TS0601 ) display an entity : *.backlight_mode and power on state in the configuration tab in ZHA.

Can the same configurations not be added to the 1 gang TS0011 switches so we can swap the LED status so they react in the correct way ? 

I'm new to HA so not sure if ZHA has files missing or is it not a ZHA error ?

I have been asked to include the signitures or each.

TS0011 1 gang with no backlight feature : 

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_hhiodade",
  "model": "TS0011",
  "class": "zhaquirks.tuya.ts001x.TuyaSingleNoNeutralSwitch_2"
}

TS0601 with fully working features : 

{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0006"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "_TZE200_g1ib5ldv",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_switch.TuyaDoubleSwitchTO"
}

[image]

[image]

javicalle, 2023-02-18:
In the `TS0011` devices the backlight behavior can be set from the device view -> Admin zigbee device:
[image]

Select the `TuyaZBOnOffAttributeCluster` cluster and the `backlight_mode` attribute to get and set the value.

As the attribute is already there, the UI part must be addressed in HA.

javicalle, 2023-02-18:
BTW, some Moes devices accepts different `backlight_mode` values:
* https://github.com/zigpy/zha-device-handlers/blob/8122e9283589f4fff585e4cf03892bf87587e345/zhaquirks/tuya/mcu/__init__.py#L70-L77

javicalle, 2023-02-19:
@cayote64 have you be able to fix the behavior? can the issue be closed?

cayote64, 2023-02-19:
Yes all the switches changed to the correct status.

Many thanks, case closed 😀

closed 2023-02-19 by cayote64

TheJulianJES, 2023-02-20:
We could also leave this open until the proper entities are added in ZHA/HA Core (even though it's technically a HA issue and not a quirks issue), but it's a bit easier to keep track of the issues here.

I think this should/would be fixed if we match quirk names instead of manufacturer names for Tuya devices?

javicalle, 2023-02-20:
> I think this should/would be fixed if we match quirk names instead of manufacturer names for Tuya devices?

I think that would be a clear candidate to be implemented 😉

dr3amr, 2023-02-24:
Been looking for this for ages...would be verry nice to be added to ui. The switch supports other functions also.

GerardUmbert, 2023-03-21:
> e and power on state in the configuration tab in ZHA.
>
> Can the same configurations not be added to the 1 gang TS0011 switches so we can swap the LED status so they react in the correct way ?
>
> I'm [...]

And what are we supposed to write there if we want it disabled all the time?

dr3amr, 2023-03-21:
@GerardUmbert it is mentioned above, write "0x00" (off) in the Value field and click on "Write attribute". You should now be able to see that the light is always off.