zigpy/zha-device-handlers #2205: [Device Support Request] Cleverio / iHorn - Scenario Button (HO-011ZB)

Tagnard, 2023-02-18:
**Is your feature request related to a problem? Please describe.**
Button presses are not recognized

**Describe the solution you'd like**
I want it to be able to handle the single-, dubbel- and long press events

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0006"
      ],
      "out_clusters": [
        "0x0006",
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TYZB01_4qw4rl1u",
  "model": "TS0041A",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Stockholm",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.7",
      "pyserial==3.5",
[... 392 lines cut ...]
[0xC2F4:1:0x0006] Received command 0xFD (TSN 14): b'\x02'
[0xC2F4:1:0x0006] No explicit handler for cluster command 0xfd: b'\x02'
[0xC2F4:1:0x0006]: received '253' command with b'\x02' args on cluster_id '6' tsn '14'
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC2F4), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=56, profile_id=260, cluster_id=6, data=Serialized[b'\x01\x0e\xfd\x02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-65)
[0xC2F4:1:0x0006] Received ZCL frame: b'\x01\x0e\xfd\x02'
[0xC2F4:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=14, command_id=253, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xC2F4:1:0x0006] Unknown cluster command 253 b'\x02'
[0xC2F4:1:0x0006] Received command 0xFD (TSN 14): b'\x02'
[0xC2F4:1:0x0006] No explicit handler for cluster command 0xfd: b'\x02'
[0xC2F4:1:0x0006]: received '253' command with b'\x02' args on cluster_id '6' tsn '14'
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xC2F4), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=57, profile_id=260, cluster_id=6, data=Serialized[b'\x01\x0e\xfd\x02'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-66)
[0xC2F4:1:0x0006] Received ZCL frame: b'\x01\x0e\xfd\x02'
[0xC2F4:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Server_to_Client: 0>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), tsn=14, command_id=253, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
[0xC2F4:1:0x0006] Unknown cluster command 253 b'\x02'
[0xC2F4:1:0x0006] Received command 0xFD (TSN 14): b'\x02'
[0xC2F4:1:0x0006] No explicit handler for cluster command 0xfd: b'\x02'
[0xC2F4:1:0x0006]: received '253' command with b'\x02' args on cluster_id '6' tsn '14'
[0xC2F4](TS0041A): Device seen - marking the device available and resetting counter
[0xC2F4](TS0041A): Update device availability -  device available: True - new availability: True - changed: False
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-02-18:
I have a quirk porposal for your device.

You would need enable the local quirk folder in your installation. There is a good guide to configure it:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once configured, copy in your `custom_zha_quirks` folder the current `ts0041.py` file:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0041.py

Edit the file and add at the bottom this class:
<details>
<summary>TuyaSmartRemote0041_var04</summary>

```python
class TuyaSmartRemote0041_var04(CustomDevice):
    """Tuya 1-button remote device with time on out cluster."""

    signature = {
        MODEL: "TS0041A",
        ENDPOINTS: {
            # "profile_id": 260,
            # "device_type": "0x0000",
            # "in_clusters": ["0x0000","0x0001","0x0006"],
            # "out_clusters": ["0x0006","0x000a","0x0019"]
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    OnOff.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    OnOff.cluster_id,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaSmartRemoteOnOffCluster,
                ],
                OUTPUT_CLUSTERS: [
                    TuyaSmartRemoteOnOffCluster,
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

    device_automation_triggers = {
        (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: SHORT_PRESS},
        (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: LONG_PRESS},
        (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 1, COMMAND: DOUBLE_PRESS},
    }
```

</details>

Not sure if the `TuyaSmartRemoteOnOffCluster` must be in both clusters (input & output).
Save changes, remove the device from HA, restart HA and pair the device again.

Check if device gets the new quirk and attach logs if don't work as expected.

Tagnard, 2023-02-18:
Works like a charm! Thanks for your help

MattWestb, 2023-02-18:
If its one remote and have both in and out try deleting  the in then  its making one not working switch in ZHA GUI.

javicalle, 2023-02-19:
@Tagnard is the quirk creating an useless switch button in HA?
Can you try the suggested change (removing the `TuyaSmartRemoteOnOffCluster` from the input clusters)?

Tagnard, 2023-02-19:
No, no useless button was added with or without TuyaSmartRemoteOnOffCluster.

Tagnard, 2023-02-19:
Should the button be added under diagnostics for the device in hass or are only events triggered?

javicalle, 2023-02-19:
Sorry to insist but, have you test to remove the device from HA?
If not, could you test it?

Remove the device from HA. Wait a couple of minutes. Pair the device again. If it is a battery device, be sure to keep the device awake during all the pairing process just short pressing any button every second or so.

Thanks in advanced.

Tagnard, 2023-02-19:
If i remove TuyaSmartRemoteOnOffCluster from input_cluster, the name of the button in the event is missing. And nothing happens with or without it in output_clusters.

I did 4 tests, and during the tests I removed the device and restarted hass between each test.
[image]

javicalle, 2023-02-20:
@Tagnard Thanks a lot for testing. We really appreciate it.

@MattWestb, I bet that the `ep_attribute` overwritte prevents from creating the HA button:
* https://github.com/zigpy/zha-device-handlers/blob/e1b82c206a2b5ca24e11d2c0ac3f79eb81c44e1f/zhaquirks/tuya/__init__.py#L926

So it seems that the `OUTPUT_CLUSTER` is not really needed here but I would keep it in the quirk because you know, is Tuya 🤦🏻‍♂️

MattWestb, 2023-02-20:
If you like do it.

With TS004F devices was doing but is using `NON_COLOR_CONTROLLER` and the TS0044 is using `REMOTE_CONTROL` as device type.

If its not  doing on not working switch in the GUI is great then  its only making problems for our users.

BoomerBoomers, 2023-07-21:
Are there any news on this? I have a TS0041A button and the same problem of only getting the battery status when adding this.

closed 2024-01-24 by github-actions[bot]

Eth030, 2024-02-05:
i tested the quirk on a TS0041A
von _TYZB01_1xktopx6  and i can confirm i only have following registered.
[image]

MattWestb, 2024-02-05:
Batters % an LQI and RSSI is standard for battery powered controllers and what do you like more to have ?
Your device is sending events that you can using doing  automatons and its the was controllers working in ZHA.