zigpy/zha-device-handlers #2206: [Device Support Request] Aqara Motion Sensor P1

timiman, 2023-02-18:
**Is your feature request related to a problem? Please describe.**
I've just migrate from ConBeeII to SkyConnect and Aqara Motion Sensor P1 just don't pair correctly. I have tried many times to pair and always the scenario is to stop responding to any activity or setting change after been paired through HA O/S. 

**Describe the solution you'd like**

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0406",
        "0x0500",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.ac02",
  "class": "zhaquirks.xiaomi.aqara.motion_ac02.LumiMotionAC02"
}```

</details>

<details>
<!-- Diagnostic information can be acquired by clicking on the "Download Diagnostics" button in the device settings view -->
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Athens",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.26.2",
      "requirements": [
        "aiogithubapi>=22.2.4"
      ]
    }
[... 357 lines cut ...]
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Also the Battery information seems to be empty.
Re-configuring the device fails at "OppleCluster"
[image]
[image]

TheJulianJES, 2023-02-18:
No problem with pairing multiple P1 sensors to a SkyConnect here. Battery information can take quite some time to arrive (the motion sensor sends it in a way that doesn't follow specs).

If you leave the sensor with no motion for over a minute and then wave your hand in front of it, does the status entity change or is it always stuck on "Clear"?

timiman, 2023-02-18:
It stays at Clear even if I have left it without motion for over an hour

timiman, 2023-02-19:
I've added additional logs during the pairing period of the sensor.
I've also tried with fresh batteries in case that was the issue, but no go again. After pairing with Coordinator the icons on the "Device is ready" modal became grey out, like the device "died"(?) / "went offline". It is the exact time that the sensor starting to not respond to quick click of its button (connectivity check).

AmmokSkater, 2023-02-23:
Same issue here....

<details> <summary>device signatur</summary>

```json
"node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0406",
        "0x0500",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0xfcc0"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.motion.ac02",
  "class": "zhaquirks.xiaomi.aqara.motion_ac02.LumiMotionAC02"
```
</details>

<details><summary>diagnostic</summary>
<p>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "multimatic": {
      "version": "1.13.0",
      "requirements": [
        "pymultimatic==0.6.12"
      ]
    },
    "nodered": {
      "version": "1.1.2",
      "requirements": []
    },
    "wundergroundpws": {
      "version": "1.1.1",
      "requirements": []
    },
[... 343 lines cut ...]
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xfcc0": {
            "endpoint_attribute": "opple_cluster",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```
</p>
</details>

[image]

[image]

[image]

AmmokSkater, 2023-02-23:
ok, i have some news. I've done a firmware update on my conbee 2 stick up to the latest version.
before: deCONZ_ConBeeII_0x26720700.bin.GCF 9-Aug-2021
now: deCONZ_ConBeeII_0x26780700.bin.GCF 15-May-2022
source: [https://deconz.dresden-elektronik.de/deconz-firmware/](url)
installation manual: [https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually](url)

Now the motion sensor works fine and it shows me the illuminance:
[image]

But it still doesn't show me the battery.

AmmokSkater, 2023-02-23:
I am well aware of that. but i thought it might help you with your problem, too. sorry for interfering

MattWestb, 2023-02-23:
With EZSP coordinators you you also need running the latest version of HA for getting latest gen Aqara devices working OK (the same reason deCONZ need latest firmware but EZSP is doiing it in the driiver).

timiman, 2023-02-23:
I'm already running HA 2023.3.b0 which also includes zha-quirks v0.0.93. Still Aqara P1 cannot be paired with a SkyConnect coordinator.

timiman, 2023-02-28:
Quick update of my Aqara Motion Sensor P1 pairing issue.
It seems that after an issue with HA 2023.3.0b3 update, which crippled my HA installation and recovered back only via a full VM backup replacement and by using -now- version 2023.3.0b5 it seems that I can now pair Aqara Motion Sensor P1 and use it without any issues at all.
I really don't know if anything from .0b4 or .0b5 helped with that but the only other parameter that was changed on the HA installation was the manual removal of HACS. The removal of HACS -happened some months ago- from within HA UI was not fully completed, because by searching the logs for info regarding the issue I've got with .0b3 update, I've found that HA was reporting that an installation of HACS was present on the system which could not be found on HA UI.
So, I've 'rmdir' the HACS installation folder manually from Terminal.
I repeat, I do not know if this has to do anything with the pairing issue, but it was the only thing that was changed in between.

cammurray, 2023-04-21:
Was there something changed with the ZHA quirks lib that's impacting this device? In 2023.2 home assistant bumped the quirks lib to 0.0.96.

I'm running HA 2023.4.5 and the device is stuck "Clear" and no motion is detected, i've dropped the detection interval down to 5 seconds and still have the same issue.

I too am using a SkyConnect USB dongle, same as @timiman

thos-edison, 2023-06-03:
Same as @cammurray and @timiman  with my install of HA - after many attempts, it pairs...then, it's useless.

timiman, 2023-06-18:
Another late update for the P1. It seems that for some reason it is reporting thousands of lux as illuminance when it is about to become for sure unavailable and started needing a re-pair to become available again. It has nothing to do with network health as in my case it is just 1 meter from the coordinator itself. It is the most un-trusted device of my Zigbee network by far.

egelinotte, 2023-07-02:
Any update on this issue ? It seems that more un more people have this problem (included myself).

PabloGux, 2023-09-24:
Same issue here! Aqara P1 contact sensor (lumi.magnet.ac01) pairs with ZHA but the binary_sensor does not get updated, nor it reads battery information. Please add support in ZHA for this.

Aeilstrup, 2023-10-04:
I too hope for a fix...

mdalacu, 2024-02-06:
Is there any special method to pair the sensor with SkyConnect? I press the side button for more than 5 seconds but id does not show in ZHA what i scan for new devices. In Xiaomi gateway v2 I can add it immediately..
Thanks! :-)

closed 2024-08-11 by github-actions[bot]