zigpy/zha-device-handlers #2209: [Device Support Request] Sunricher 3 in 1 Remote Control ZG2858A

qp68, 2023-02-18:
**Is your feature request related to a problem? Please describe.**
For the moment, only battery and identify are shown for my Sunricher 3 in 1 Remote Control ZG2858A. Direct binding is possible but S1 to S6 buttons can't be use.

**Describe the solution you'd like**
To be able to use buttons as Home Assistant actions to trigger automations

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4644, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0105",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0105",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x0300",
        "0x1000"
      ]
    },
    "3": {
      "profile_id": 260,
      "device_type": "0x0105",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0b05",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
[... 537 lines cut ...]
            "endpoint_attribute": "level",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0300": {
            "endpoint_attribute": "light_color",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

TheJulianJES, 2023-02-19:
If you listen to ``zha_event`` in the HA developer section, do you see any events when you press buttons? (Or do you see events in the HA logbook?)

qp68, 2023-02-19:
So I digged a little more and it seems that event on button press are only passed after pressing on buttons 1-3 and then S1-S5 buttons... But I can see them on HA so there are no problems. Thanks

closed 2023-02-19 by qp68