zigpy/zha-device-handlers #2211: [Device Support Request] Leedarson LDHD2AZW Contact Sensor

larryventura, 2023-02-19:
**Is your feature request related to a problem? Please describe.**
Leedarson LDHD2AZW is mostly properly working, however `battery_percentage_remaining` is not reported.

**Describe the solution you'd like**
Improved support which exposes battery level properly.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4381, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05",
        "0xfd50"
      ],
      "out_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "Leedarson",
  "model": "LDHD2AZW",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
"data": {
    "ieee": "**REDACTED**",
    "nwk": 54204,
    "manufacturer": "Leedarson",
    "model": "LDHD2AZW",
    "name": "Leedarson LDHD2AZW",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4381,
    "power_source": "Battery or Unknown",
    "lqi": null,
    "rssi": null,
    "last_seen": "2023-02-19T11:25:19",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4381, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0402",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0020",
            "0x0402",
            "0x0500",
            "0x0b05",
            "0xfd50"
          ],
          "out_clusters": [
            "0x0019"
          ]
        }
      }
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "binary_sensor.washing_machine_door_iaszone",
        "name": "Leedarson LDHD2AZW"
      },
      {
        "entity_id": "button.washing_machine_door_identify",
        "name": "Leedarson LDHD2AZW"
      },
      {
        "entity_id": "sensor.washing_machine_door_battery",
        "name": "Leedarson LDHD2AZW"
      },
      {
        "entity_id": "sensor.washing_machine_door_temperature",
        "name": "Leedarson LDHD2AZW"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
      {
        "name": "IAS_ZONE"
      }
    ],
    "user_given_name": "",
    "device_reg_id": "e305d2d0470867281c2b87f2ff1bf078",
    "area_id": "",
    "cluster_details": {
      "1": {
        "device_type": {
          "name": "IAS_ZONE",
          "id": 1026
        },
        "profile_id": 260,
        "in_clusters": {
          "0x0000": {
            "endpoint_attribute": "basic",
            "attributes": {
              "0x0004": {
                "attribute_name": "manufacturer",
                "value": "Leedarson"
              },
              "0x0005": {
                "attribute_name": "model",
                "value": "LDHD2AZW"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0001": {
            "endpoint_attribute": "power",
            "attributes": {
              "0x0020": {
                "attribute_name": "battery_voltage",
                "value": 29
              },
              "0x0031": {
                "attribute_name": "battery_size",
                "value": 7
              },
              "0x0033": {
                "attribute_name": "battery_quantity",
                "value": 1
              }
            },
            "unsupported_attributes": {
              "0x0021": {
                "attribute_name": "battery_percentage_remaining"
              }
            }
          },
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0020": {
            "endpoint_attribute": "poll_control",
            "attributes": {
              "0x0000": {
                "attribute_name": "checkin_interval",
                "value": 13200
              }
            },
            "unsupported_attributes": {}
          },
          "0x0402": {
            "endpoint_attribute": "temperature",
            "attributes": {
              "0x0000": {
                "attribute_name": "measured_value",
                "value": 2017
              }
            },
            "unsupported_attributes": {}
          },
          "0x0500": {
            "endpoint_attribute": "ias_zone",
            "attributes": {
              "0x0000": {
                "attribute_name": "zone_state",
                "value": 1
              },
              "0x0001": {
                "attribute_name": "zone_type",
                "value": 21
              },
              "0x0002": {
                "attribute_name": "zone_status",
                "value": 0
              },
              "0x0010": {
                "attribute_name": "cie_addr",
                "value": "00:12:4b:00:2a:2e:82:25"
              }
            },
            "unsupported_attributes": {}
          },
          "0x0b05": {
            "endpoint_attribute": "diagnostic",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0xfd50": {
            "endpoint_attribute": "manufacturer_specific",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-02-19:
According to Z2M battery report can take 24h long to report:
* https://www.zigbee2mqtt.io/devices/LDHD2AZW.html#battery-numeric

Have you waited time enought to verify?

Doridian, 2023-05-20:
I can add my +1 here: I had some of these sensors for a month now, and they just don't report battery percentage. They do report battery voltage (~ 3.0V), quantity (1) and size (CR2). They also do not report battery rated voltage.

closed 2023-11-23 by github-actions[bot]