zigpy/zha-device-handlers #2214: [Device Support Request] IKEA Symfonisk remote

ditheo, 2023-02-22:
I have bought 10 of them to control some media players. I think that maybe there is a new version of that.  The product code is 603.704.80 instead 203.704.82 404.337.80 described on zigbee blackadder page of the product. The remote is connected to zha normally and get the quirk but  I cannot set any functions. Also I cannot get anything on zha_event listener. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

<details>

<summary>Device signature</summary>

```yaml
node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)"
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Athens",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "spotcast": {
      "version": "v3.6.30",
      "requirements": []
    },
    "alarmo": {
      "version": "v1.9.7",
      "requirements": []
    },
    "hacs": {
      "version": "1.30.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "browser_mod": {
      "version": "1.3.1",
      "requirements": []
    },
    "smartir": {
      "version": "1.17.6",
      "requirements": [
        "aiofiles==0.6.0"
      ]
    },
    "bodymiscale": {
      "version": "3.0.7",
      "requirements": [
        "cachetools==5.1.0"
      ]
    }
[... 323 lines cut ...]
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-02-22:
IKEA controllers is fast sleepers and its normal they is going in to sleep before all is being configured OK. 
Your devises is normal ones then the quirk is loaded OK `"quirk_class": "zhaquirks.ikea.symfonisk.IkeaSYMFONISK1"` but not on the latest production firmware.
From the device card clock on `reconfigure` and waking up the device before continuing so its not sleeping.
It can being tricky getting it working and restart of HA can helping but some time its need readding  it by resetting the device and adding it one more time..
Then you is getting events on the e long on the device card it shall  being OK.

ditheo, 2023-02-22:
After your advice i can see the events on listener. Also toss all the batteries. From the 10 boxes only one was ok

closed 2023-02-22 by ditheo

MattWestb, 2023-02-22:
The upcoming 24.X firmware is mush worse (Dirigera is updating them to it) and oft need taking the battery out for getting the reconfigure working OK.

Great that you was getting 10 of them then its the best IKEA controller but its going out for the upcoming Symfonisk 2 that not looks so nice.