zigpy/zha-device-handlers #2221: TS0601 Inconsistent behaviour of zhaquirks.tuya.ts0601_trv.MoesHY368_Type1new

DeanK2022, 2023-02-24:
I am running HA latest and I am seeing a difference in the handling of maximum temp settings on some but not all of my TS0601 TRVs.

Home Assistant 2023.2.5
Supervisor 2023.01.1
Operating System 9.5
Frontend 20230202.0 - latest

Some TRVs have the correct temperature range, others seem to want to go up to 300!
[image]

The device signatures are identical:
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0301",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_b6wax7g0",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_trv.MoesHY368_Type1new"
}

I have deleted and re-added the device back to HA but the range still seems stuck on 0-300.  Any clues as to what to try next?  Would deleting everything and moving to Zigbee2MQTT solve this issue?  I have quite a lot of devices (>20) installed so this would be quite painful.

Can anyone please tell me where to look next?

stueng, 2023-04-14:
Hi, have you manged to get force valve state working with this new TRV? 
I have older TRV's with MoesHY368_Type1 working fine with force state, but not the newer one with MoesHY368_Type1new

DeanK2022, 2023-04-14:
Nope, I control the valve on/off state by setting the target temperature.  I have grouped the TRVs into regions (lower/upper floor and whole house) to make them easier to set en masse.

closed 2023-10-18 by github-actions[bot]

dzervas, 2023-11-20:
I also have the same problem (some reporting 30, some reporting 300 max temp)
@stueng While there is a `valve_force_state` attribute in zigbee (id 0x4006 in cluster MoesThermostatNew id 1, 0x0201, type in)

DeanK2022, 2023-11-21:
With the 30 vs 300 issue, did you manage to see what it is doing?  I have been looking for about a year to try to fix this but nothing works so far...

dzervas, 2023-11-21:
I used @jacekk015 's quirk and everything works lovely - valve & battery reporting, min/max temperatures, everything

https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_beca.py

DeanK2022, 2023-11-21:
I will give that a go, thanks!  Did it fix the 30vs300 problem?

dzervas, 2023-11-22:
yeap! forgot to write that

also look how many stuff you get to tinker with:

[image]

DeanK2022, 2023-11-23:
I added the ts0601_trv_beca.py as a quirk, rebooted and I now get the shiny device console like yours:
[image]

But still, this one goes to 300 :(((
[image]

DeanK2022, 2023-11-23:
Aha!  If I explicitly tweak the min and max like on here and wait 5 mins, I get proper range on the TRV!
[image]

Thank you a million times, @dzervas!

dzervas, 2023-11-23:
happy to help!

now you can open the can of worms of managing the temperature per room (needs an additional temperature sensor/room, need to find a consistent way to tell if the room is used, need to find the correct temperatures, etc.)

If you find an easy solution for all these please drop me a note (an email will do as well, I have it in my gh profile)

oh side note for the radiator vs room temperature and offset calibration madness: https://better-thermostat.org/

I'm waiting for my temp sensors to arrive so I haven't tested it yet

DeanK2022, 2023-11-23:
I have a manual thermometer, I will wait until the radiators have been on for an hour or so then take what the thermometer says vs what the TRV says and then use the difference as the offset.

At a risk of making a rod for my own back, what temp sensors are you buying?

DeanK2022, 2023-11-23:
Oh and thank you once again, this has been doing my head in for well over a year!   I tried everything via web search, haunted the Discord channels on the offchance and even tried reading the code!  How did you find this repo, BTW?  Searching for repos with "_TZE200_b6wax7g0" ?

It is just so strange it works out of the box for one TRV but not the next, and they were all bought at the same time.

dzervas, 2023-11-23:
I'm buying "the cheap ones", SNZB-02. They're known not to be very stable and they don't report the temperature very frequently but I need 8 and I don't have 50eu to spare rn (the whole project's target is to be sure that I use as little diesel as possible). For info around the hardware and how to install the custom firmware: https://zigbee.blakadder.com/Sonoff_SNZB-02.html

If I had the choice I'd buy something better though that works out of the box as it should

`b6wax7g0`, `MoesHY368_Type1new` (if you search the repo you'll see that this is the class that handles this device) `BRT-100` (it's the name that zigbee2mqtt wiki uses for this device - probably ours is a rebrand?)

DeanK2022, 2023-11-24:
Interesting stuff!  Thanks for the link, good luck getting it working smoothly.

And thanks very very much for the link.  I had been using search engines across the web to look for the code support, looks like I should just go straight to git instead.  

I also share your hesitancy for "random internet code", my Python is weak so it takes a while to read and understand it before I install anything!