zigpy/zha-device-handlers #2222: Strange readings from Tongou (Tuya) Circuit Breaker [Zigbee ID: TS011F | _TZ3000_qeuvnohg]

filipeaparicio91, 2023-02-24:
Hi,

I added 4 Tuya Circuit Breakers (manufactured by Tongou) to Z2M and configured them correctly, but 2 of them show readings of 3W and 4W respectively even when they are switched off.

When power is being drawn by the circuit breakers, they update the power value accordingly, but as soon as I turn off the appliances that are connected to them, they return to the 3W and 4W "stand-by" readings.

The issue is that if I turn the switches off, they still show these values, when I assume the readings should be 0W.

This also affects the energy consumption reported by the circuit breakers as over an extended period of time, despite having them turned off, the KWh values keep increasing.

Could this be a calibration issue? If so, how does one get this fixed?

arsenyspb, 2023-03-05:
@filipeaparicio91 me too. I got *_TZ3000_qeuvnohg TS011F* which is [Tongou TO-Q-SY1-JZT](https://www.zigbee2mqtt.io/devices/TS011F_din_smart_relay.html#tuya-ts011f_din_smart_relay) and is supported by Zigbee2MQTT. 

I confirm that 1 out of 4 Tongou TO-Q-SY1-JZT in my batch was also reporting "stale" reading. Mine was high, - anywhere from 30 to 40W, - and as in your case, only after load was off. At restart it was "clear", 0W. I hoped firmware update would help, but:

```
2023-03-05 10:23:33.125 DEBUG (MainThread) [zigpy.zcl] [0x0F25:1:0x0019] OTA upgrade_end handler for '_TZ3000_qeuvnohg TS011F': status=Status.INVALID_IMAGE, manufacturer_id=4098, image_type=5634, file_version=87
```
...@Koenkk says we [need `image_type=5634`](https://github.com/Koenkk/zigbee2mqtt/issues/16584). If you happen to have a TuYa bridge, please consider [contributing](https://www.zigbee2mqtt.io/advanced/more/tuya_xiaomi_ota_url.html) it.

So far, I requested refund from seller, since others in batch work fine with ZHA out of the box, like the 50% of yours.

Re: "kWh increasing", FYI, also it looks like it's a known issue in TZ3000_qeuvnohg itself, since the meter reports summation 100x times higher than real. Also the current is seemingly in mA, not in A. It's already addressed in Z2M's [TS011F_din_smart_relay](https://github.com/Koenkk/zigbee-herdsman-converters/blob/348dc94d3daaba5d7041410e1dda274af236b23f/devices/tuya.js#L3411) by dividing Summation Delivered by `100`, and Current by 1000. But I don't think it's addressed in ZHA just yet. For ZHA, @zleba was working on [pull 2008](https://github.com/zigpy/zha-device-handlers/pull/2008) to support it correctly, see their [issue 2005](https://github.com/zigpy/zha-device-handlers/issues/2005).

Please update this thread if you have more solutions on these DIN circuit breakers.

arsenyspb, 2023-03-05:
@filipeaparicio91 I tried again, here's what I found. When there's no appliance connected the "shadow" load shows up. E.g. if I have an appliance that draws standby current, it's 0W. If it's an airgap / no load, it shows weird random numbers.

patrickli, 2023-05-29:
I also have one of them showing 3W when the appliance is not drawing power. I have another one that doesn't do this. The 3W is not constant, it changes between 3W and 0W but 3W most of the time. Current is always 0.0A. Worst case scenario it is going to skew the summation delivered by 2kWh per month which is not a lot. Can't be bothered to return or exchange it now.

vhascoet, 2023-11-16:
Hello,

Same here with a TO-Q-SY1-JZT 32A:

- 0W/0mA when switched off
- 5W/299-305mA when switched on and induction hob off
- 0W/0mA when switched on and induction hob on (not heating)

That's confusing...

cordvision, 2024-02-28:
Same here with the Earu ts011f Breaker (re-brand). Also noticed that the wattage readings are about 7-8% too high (vs actual load).

Koenkk, 2024-02-29:
Probably the hardware is just not that good, I recommend to use the [calibration feature](https://www.zigbee2mqtt.io/devices/TS011F_din_smart_relay_polling.html#options)

closed 2024-09-03 by github-actions[bot]