zigpy/zha-device-handlers #2223: [Device Support Request] Ikea E2123: Symfonisk sound remote Gen2

MrDodojo, 2023-02-24:
**Is your feature request related to a problem? Please describe.**
Ikea released a new Symfonisk sound remote (gen2) and it is not yet supported by ZHA. The device when added to homeassistant does not have triggers for pressing the various buttons seperately. The only triggers it has are: Device offline, has been pressed and battery level changed(while the battery level does not indicate correctly either).

**Describe the solution you'd like**
Support for all the buttons: Volume up & down, next & previous, play/pause and 2 custom "shortcut" buttons. 

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0006",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x1000",
        "0xfc57"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019",
        "0x1000",
        "0xfc7f"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "SYMFONISK sound remote gen2",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "adaptive_lighting": {
[... 308 lines cut ...]
            "endpoint_attribute": "manufacturer_specific",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
No Debug logs
```

</details>

**Additional context**
[image]
^This is the device, it has a lot of buttons and would be quite nice to use with homeassistant.

MattWestb, 2023-02-24:
Where did you geeing it ??
I hope its being released iin EU very soon then its have being CSA-certified over one year now.

MrDodojo, 2023-02-24:
> Where did you geeing it ?? I hope its being released iin EU very soon then its have being CSA-certified over one year now.
> 
> Edit: Its coming next week to Westbahnhoff in Vienna :-(((

Got it at an Ikea in the Netherlands, they didn't have a lot of them so I was sure to grab one. 2 extra buttons for a total of 7 is nice to have

MattWestb, 2023-02-24:
Its looks very standard of the cluster and is clean but we must getting all the commands its sending.
I think its prity easy getting all Device Automatons (DA) but its needs little work getting it right and testing all is working OK.
Its only 2 IKEAs in Austria that having it on the homepage and no inside Vienna  but i going and looking tomorrow if its some is having it.

One interesting thing its only Dirigera compatible and TF but shall testing that 2.

MattWestb, 2023-02-25:
Do you having ZHA-Toolkit https://github.com/mdeweerd/zha-toolkit  installed ??

If yes bind the cluster Id: `0xfc80` on endpoint 2 and 3 to the coordinator (0x0000) and i think you shall getting the commands from the 2 shortcut buttons then its wot IKEA is doing in Dirigera then pairing it.

MrDodojo, 2023-02-26:
No I haven't. I just did, don't know how to use it yet. I'd appreciate if you could spoon-feed/tell me what to do. Otherwise it will take me a while to understand the docs

SamDeBlock, 2023-02-27:
I have the same device, these are the debug logs for the 'dot' buttons:

1 dot button
```
DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64639, SrcAddr=0xD1BF, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=63, SecurityUse=<Bool.false: 0>, TimeStamp=1453435, TSN=0, Data=b'\x15\x7C\x11\x42\x01\x01\x01', MacSrcAddr=0xD1BF, MsgResultRadius=29)
DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD1BF), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64639, data=Serialized[b'\x15|\x11B\x01\x01\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=63, rssi=None)
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Received ZCL frame: b'\x15|\x11B\x01\x01\x01'
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4476, tsn=66, command_id=1, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
WARNING (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Unknown cluster command 1 b'\x01\x01'
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Received command 0x01 (TSN 66): b'\x01\x01'
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] No explicit handler for cluster command 0x01: b'\x01\x01'
```

2 dot button
```
DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=64639, SrcAddr=0xD1BF, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=57, SecurityUse=<Bool.false: 0>, TimeStamp=2806217, TSN=0, Data=b'\x15\x7C\x11\x43\x01\x02\x01', MacSrcAddr=0xD1BF, MsgResultRadius=29)
DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD1BF), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=64639, data=Serialized[b'\x15|\x11C\x01\x02\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=57, rssi=None)
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Received ZCL frame: b'\x15|\x11C\x01\x02\x01'
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=True, direction=<Direction.Server_to_Client: 0>, disable_default_response=1, reserved=0, *is_cluster=True, *is_general=False, *is_reply=False), manufacturer=4476, tsn=67, command_id=1, *direction=<Direction.Server_to_Client: 0>, *is_reply=False)
WARNING (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Unknown cluster command 1 b'\x02\x01'
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] Received command 0x01 (TSN 67): b'\x02\x01'
DEBUG (MainThread) [zigpy.zcl] [0xD1BF:1:0xfc7f] No explicit handler for cluster command 0x01: b'\x02\x01'
```

MattWestb, 2023-02-27:
For both is the remote on original / factory firmware or is it updated by Dirigera ?
I have looking   little and my have more endpoints and cluster but i was updating it and its now `0x01000032`.
Signature:
```
    },
      }
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 31962,
    "manufacturer": "IKEA of Sweden",
    "model": "SYMFONISK sound remote gen2",
    "name": "IKEA of Sweden SYMFONISK sound remote gen2",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4476,
    "power_source": "Battery or Unknown",
    "lqi": 208,
    "rssi": -58,
    "last_seen": "2023-02-27T12:19:54",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0000",
            "0x0001",
            "0x0003",
            "0x0020",
            "0x1000",
            "0xfc7c"
          ],
          "out_clusters": [
            "0x0003",
            "0x0004",
            "0x0006",
            "0x0008",
            "0x0019",
            "0x1000"
          ]
        },
        "2": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0xfc80"
          ],
          "out_clusters": [
            "0x0003",
            "0x0004",
            "0xfc80"
          ]
        },
        "3": {
          "profile_id": 260,
          "device_type": "0x0006",
          "in_clusters": [
            "0x0000",
            "0x0003",
            "0xfc80"
          ],
          "out_clusters": [
            "0x0003",
            "0x0004",
            "0xfc80"
          ]
        }
      }
    },
```

SamDeBlock, 2023-02-27:
Mine has definitely been updated to the latest firmware.  I first connected it to the dirigera hub and then it automatically started updating.

MrDodojo, 2023-02-27:
I did not connect it to a dirigera hub since I don't have one.

MattWestb, 2023-02-27:
OK then its logic way @MrDodojo is having different configuration.
We must doing 2 device classes so both updated and not updated devices is working OK.
@MrDodojo pleas **dont** updating your device we need it for doing testing then IKEA have locking the chip so its not possible dumping the original firmware (that i normally doing) and flashing it back for doing firmware downgrading.

MattWestb, 2023-02-27:
@TheJulianJES  I have getting the signature working OK and the quirk is loading for both device classes.
The V1 is using only EP1 for all commands and V2 is using EP1 for Light commands and EP2 for button 1 and EP3 for button 2 extra commands from cluster V1 `0xFC7F` / V2`0xFC80`.

From the V1 log:
``` 
[0x0664:1:0xfc7f] Unknown cluster command 1 b'\x01\x01' = B1 Short
[0x0664:1:0xfc7f] Unknown cluster command 1 b'\x02\x01' = B2 Short
[0x0664:1:0xfc7f] Unknown cluster command 1 b'\x01\x02' = B1 Duble
[0x0664:1:0xfc7f] Unknown cluster command 1 b'\x02\x03' = B2 Held
```
The V2 is not reporting then its need EP 2 and 3 being bound and  i dont knowing how to do that.

The next problems is the DA is crazy  dim up is not triggered and left and right is triggered for both.

Can you taking one look if you can see what i have doing wrong ?

[symfonisk2.zip](https://github.com/zigpy/zha-device-handlers/files/10843413/symfonisk2.zip)

Thanks in advance !

MichelBrodatzki, 2023-02-28:
I've just opened a pull request with a basic implementation for the E2123 with (probably) V1. Every button should work as a trigger for automations (and some buttons even with long press and double press).
Currently I'm struggling a bit with passing arguments as both ARGS and PARAMS (or with passing the at all as PARAMS) with the listener_event(...) approach. 

@MattWestb I'm quite new to creating device handlers and I don't have an updated E2123 at hand to test my theories, but maybe the event listener approach might be a way to make implementing V2 easier? (i.e. passing the command actions from EP2 and 3 to the custom LevelControlCluster)

MattWestb, 2023-02-28:
Also Also i think the `PowerConfiguration2AAACluster` is not needed and can using the normal `PowerConfiguration.cluster_id` like all other V24.X firmware devices.
My Sy2 is showing 50 and 82% without doubling and is having unchanged batteries from IKEA that shall need being changed for have 100%.
What is your device reporting and is you using alkaline or rehanged batteries ?

MattWestb, 2023-02-28:
V2 looks using complete different commends but its having release after long press implanted and is sent from EP 2 and 3 for button 1 and 2.
Attaching the frames its sending and have the paring and the button press saved if we need it later.
[Symfonisk2UpdatedDASniff.txt](https://github.com/zigpy/zha-device-handlers/files/10848708/Symfonisk2UpdatedDASniff.txt)

It can being the command is the same as V1 then wireshark is little outdated but i dont have the knowledge analyzing that.

MrDodojo, 2023-02-28:
Anything for me to test atm? I could test the quirk MichelBrodatzki made but I don't understand how to. I added a custom quirk location to my configuration, do I need to add the whole repo or just the changed files?

MattWestb, 2023-03-05:
I have getting all the V2 commands working from the network but i cant getting them being sent to zha_events. The command part looks like this:
```
class ShortcutV2Cluster(CustomCluster):
    """Ikea Shortcut Button Cluster Variant 2."""

    name = "ShortcutClusterV2"
    cluster_id = 0xFC80

    server_commands = {
        0x01: foundation.ZCLCommandDef(
            "shortcut",
            {
                "shortcut_mode": t.int8s,
                "param1": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
       ),
       0x02: foundation.ZCLCommandDef(
            "hold",
            {
                "held": t.uint4_t,
                "param1": t.uint4_t,
               },
            False,
            is_manufacturer_specific=True,
        ),
        0x03: foundation.ZCLCommandDef(
            "press",
            {
                "pressed": t.uint4_t,
                "param1": t.uint4_t,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x04: foundation.ZCLCommandDef(
            "releas",
            {
                "released": t.uint4_t,
                "param1": t.uint4_t,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x05: foundation.ZCLCommandDef(
        "unknown",
            {
                "unknown": t.uint4_t,
                "param1": t.uint4_t,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x06: foundation.ZCLCommandDef(
            "double",
            {
                "double_pressd": t.int8s,
                "param1": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
    }

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: List[Any],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ):
        """Handle a cluster request."""
        self.debug(
            "%s: handle_cluster_request - Command: %s Data: %s",
            self.name,
            hdr.command_id,
            args,
        )

        if hdr.command_id == 0x01:
            self.debug("%s: sending ZHA event", self.name)

            event_args = {
                "step_mode": (
                    LevelControl.StepMode.Up
                    if args.shortcut_mode == 1
                    else LevelControl.StepMode.Down
                ),
                "step_size": args.param1,
                "transition_time": 0,
            }

            self.endpoint.out_clusters[8].listener_event(
                ZHA_SEND_EVENT, "step_with_on_off", event_args
            )
```
The last part i have not touched then i dont understanding it.

@javicalle do you knowing how getting the command to events ?
The log is now looks like this then pressing the button:
```
2023-03-05 14:01:07.573 DEBUG (MainThread) [zigpy.zcl] [0xBD72:2:0xfc80] Decoded ZCL frame: ShortcutV2Cluster:press(pressed=0, param1=0)
2023-03-05 14:01:07.576 DEBUG (MainThread) [zigpy.zcl] [0xBD72:2:0xfc80] Received command 0x03 (TSN 85): press(pressed=0, param1=0)
2023-03-05 14:01:07.580 DEBUG (MainThread) [zigpy.zcl] [0xBD72:2:0xfc80] ShortcutClusterV2: handle_cluster_request - Command: 3 Data: press(pressed=0, param1=0)
```

MattWestb, 2023-03-05:
I was looking how we have doing the TS004X implementation that is only sending events and not using strange cluster and also is doing the arg and parms OK and i like that but i cant doing  the coding for Symfonisk 2.
[class TuyaSmartRemoteOnOffCluster(OnOff, EventableCluster):](https://github.com/zigpy/zha-device-handlers/blob/9402bcf5be5e5c434786c9fef91c0e5aa0d132eb/zhaquirks/tuya/__init__.py#L934)

javicalle, 2023-03-05:
> @javicalle do you knowing how getting the command to events ?

Maybe implementing the `EventableCluster`?
```python
class ShortcutV2Cluster(CustomCluster, EventableCluster):
```
You will need to add the import like:
```python
from zhaquirks import EventableCluster, ...
```

javicalle, 2023-03-05:
Let me take a look at the args part...

MattWestb, 2023-03-05:
(CustomCluster, EventableCluster) was not working so i was trying with (EventableCluster)  and its loading but i dont knowing if its working.

I have copy from the tuya but i dont know how to do it working:
```
    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: List[Any],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ):

        if hdr.command_id == 0x00:
            rotate_type = args[0]
            self.listener_event(
                ZHA_SEND_EVENT, self.rotate_type.get(rotate_type, "unknown"), []
            )
        elif hdr.command_id == 0x03:
            press_type = args[0]
            self.listener_event(
                ZHA_SEND_EVENT, self.press_type.get(press_type, "unknown"), []
            )
```
I do only having different commands 1-6 and the data looks being not interesting then can see from witch EP its coming.

MichelBrodatzki, 2023-03-05:
Sorry for the radio silence, I've been quite busy with university. But I think I just solved the args part! Let me just test one last thing, then I'll push it and comment what I did! Give me 10min :D

MattWestb, 2023-03-05:
For V2 commands looks like this:

01 first frame
02 hold
03 short
04 release
05 ?
06 double

Command 0x06 is the data length 2 all other is data length 1.
Data is always length 1 but looks being irrelevant for us.

And the device is sending one command 1 and then one command 2-6.
If command 2 its sending one command 4 then being released without command 1.
So command 1 is only some button is being pressed.

MattWestb, 2023-03-05:
Only for information i was reading that IKEA is forcing update of version 1 to new firmware and user and we cant reversing it (chip is debug locked)  so V1 is only one "factory delivered version" and not one production version.
Also it is missing release after held.

MichelBrodatzki, 2023-03-05:
So, after testing I can say that V1 should work now. The solution actually was pretty simple, I just haven't interpreted to underlying code correctly.

So for argument passing to work properly, you just have to create a new class which inherits from CommandSchema (which in turn is a subclass of t.Struct) and then you can pass an object of that class to listener_event. The method handles CommandSchema differently to dicts and lists.

MichelBrodatzki, 2023-03-05:
> Only for information i was reading that IKEA is forcing update of version 1 to new firmware and user and we cant reversing it (chip is debug locked) so V1 is only one "factory delivered version" and not one production version. Also it is missing release after held.

That's an interesting information, thanks! I'd say that supporting V1 is still pretty important, as OTA is not on by default for ZHA, so there might still be some people on V1 when adding their E2123 to Home Assistant.

MattWestb, 2023-03-05:
I think that was one TF user that was not getting the OTA then TF dont have it then Dirigera is forcing update in the background.
Also many users dont have one IKEA GW and dont getting updates if not doing it manual in ZHA so V1 its needed !

And i **think** Vindriktning is not being supported by TF GW only Dirigera.

javicalle, 2023-03-05:
@MattWestb if you remove the `handle_cluster_request` function from your `ShortcutV2Cluster` class you will get `ZHA_EVENT`s with the command_id and args from the device. I would expect that you got events like:
```
    "event_type": "zha_event",
    "data": {
        "device_ieee": "14:b4:57:ff:fe:70:c4:bb",
        "unique_id": "14:b4:57:ff:fe:70:c4:bb:1:0x0102",
        "device_id": "a9f1979c6084affe5c7d5a1b6bb72463",
        "endpoint_id": 1,
        "cluster_id": 64640,
        "command": "press",
        "args": [],
        "params": {pressed=0, param1=0}
    },
```

Then I suppose that you could match the event in the `device_automation_triggers`, doesn't you?

But if you want some kind of translation from your device commands to 'standar' commands (like `step`, `step_with_on_off`, etc) that would require some kind of mapping table to translate comands and params. Is that what you want to achieve?

MattWestb, 2023-03-05:
Fast test:
```
event_type: zha_event
data:
  device_ieee: 1c:34:f1:ff:fe:78:57:15
  unique_id: 1c:34:f1:ff:fe:78:57:15:3:0xfc80
  device_id: 7b803a8f3912b70e1949639ccf9e0ad5
  endpoint_id: 3
  cluster_id: 64640
  command: hold
  args:
    - 0
    - 0
  params:
    held: 0
    param1: 0
origin: LOCAL
time_fired: "2023-03-05T20:58:22.078751+00:00"
context:
  id: 01GTSS2ZFYZ342GQ3D8SGDVC6T
  parent_id: null
  user_id: null

event_type: zha_event
data:
  device_ieee: 1c:34:f1:ff:fe:78:57:15
  unique_id: 1c:34:f1:ff:fe:78:57:15:3:0xfc80
  device_id: 7b803a8f3912b70e1949639ccf9e0ad5
  endpoint_id: 3
  cluster_id: 64640
  command: releas
  args:
    - 1
    - 0
  params:
    released: 1
    param1: 0
origin: LOCAL
time_fired: "2023-03-05T20:58:24.264093+00:00"
context:
  id: 01GTSS31M8D72JS443SF19PRK7
  parent_id: null
  user_id: null
```
Need fixing the commands to Matter standard names and testing all of them but that is for tomorrow.
I going for doing the Matter compatible and doing DA for the shortcut cluster and normal for the Light control commands.

Its no use doing other not true commands better using the raw and the user can using them in automatons and doing what they like and its more flexible.

## Great thanks @javicalle you have make my week and the next to coming 2 !!

MattWestb, 2023-03-06:
@javicalle I is using `int8s` for parms (taken from IKEA scene cluster implanting) and its working OK for all commands but command `0x06` is 2 bytes and all other is 1.
So i is using int16s for it or i getting error in the HA log that data is renaming.
But the parm dot look as expected. 
From wirschark:
```
ZigBee Cluster Library Frame, Mfr: Ikea of Sweden (0x117c)
    Frame Control Field: Cluster-specific (0x15)
    Manufacturer Code: Ikea of Sweden (0x117c)
    Sequence Number: 66
    Command: Unknown (0x06)
Data (2 bytes)
    Data: 0002
    [Length: 2]
```
And in HA the parm is:
```
event_type: zha_event
data:
  device_ieee: 1c:34:f1:ff:fe:78:57:15
  unique_id: 1c:34:f1:ff:fe:78:57:15:2:0xfc80
  device_id: 7b803a8f3912b70e1949639ccf9e0ad5
  endpoint_id: 2
  cluster_id: 64640
  command: MultiPressComplete
  args:
    - 512
  params:
    Data: 512
origin: LOCAL
time_fired: "2023-03-06T07:34:13.957233+00:00"
context:
  id: 01GTTXF9650M39ZZ9MDGPVDVC0
  parent_id: null
  user_id: null
```
and i think it shall being 2.
Is i using the wrong type for getting the data then its doing it like 0x02, 0x00  = 512 dec instead 0x00, 0x02 = 002 dec ?
Its no braking thing then we dont need the parm then all commands is unique and is being sent from respective endpoint OK but im sure its wrong and can making problem in the future and its better making it right from the beginning.

All commands is working OK and only need do more testing and making one new quirk for the device and all the DAs.

Thanks in advance for your help :-)))

MattWestb, 2023-03-06:
Working Variant 2 with updated firmware.
Configure local quirk and make one `ikea` directory in it and putting the 2 files in it.
Shall looking like this:
```
/config/custom_zha_quirks/ikea/__init__.py
/config/custom_zha_quirks/ikea/symfonisk2.py
```
[symfonisk2.zip](https://github.com/zigpy/zha-device-handlers/files/10897562/symfonisk2.zip)

Restarting HA.
Your need binding EP 2 and 3 with ZHA-Toolkit until i have implemented it in the quirk.
DA for the light commands in implanted for Shortcut not but its coming and you can using `zha_events` as trigger in automatons.

Battery info fixed and also for upcoming gen 2 2 button devices is implanted.

### For our devs: 
How do i making ZHA binding EP2 and 3 with the Chortcut cluster in the quirk ?

Is its possible using the DAs from the common DAs and adding device specific ones so not need having 2 block with 2/3 the same code ?

Thanks in advance !!

javicalle, 2023-03-06:
> Is i using the wrong type for getting the data then its doing it like 0x02, 0x00 = 512 dec instead 0x00, 0x02 = 0002 dec ?
> Its no braking thing then we dont need the parm then all commands is unique and is being sent from respective endpoint OK but im sure its wrong and can making problem in the future and its better making it right from the beginning.

Hi @MattWestb, that sounds me to all the big/little endian stuff but I have very little knowledge in this mater.
I would suggest to you to try with a `int16s_be` type and see if that make it work:
```
        0x06: foundation.ZCLCommandDef(
            "double",
            {
                "double_pressd": t.int8s,
                "param1": t.int16s_be,
            },
            False,
            is_manufacturer_specific=True,
        ),
```

But it is just a random change to get the expected value.

javicalle, 2023-03-06:
Just thinking, could it be that command `0x06` takes 2 `int8s` params?
It seems extrange to me that just one command would take an `int16s_be` param.

Could it be that command `0x06` would be something like this?:
```
        0x06: foundation.ZCLCommandDef(
            "double",
            {
                "double_pressd": t.int8s,
                "param1": t.int8s,
                "param2": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
```

MattWestb, 2023-03-06:
With BE is looks great:
```
event_type: zha_event
data:
  device_ieee: 1c:34:f1:ff:fe:78:57:15
  unique_id: 1c:34:f1:ff:fe:78:57:15:2:0xfc80
  device_id: 7b803a8f3912b70e1949639ccf9e0ad5
  endpoint_id: 2
  cluster_id: 64640
  command: MultiPressComplete
  args:
    - 2
  params:
    Data: 2
origin: LOCAL
time_fired: "2023-03-06T13:00:02.817222+00:00"
context:
  id: 01GTVG3VW1NWQB85JE4MHBA86F
  parent_id: null
  user_id: null
```
but your thinking its the right !!
I can being 2 parms and we dont knowing that.
Im planing only using the commands and no arg/parms in the DA then the switch is sending unique commands / EP so if we later changing how we is doing it in INIT the DA shall **not** being broken only user that is using `zha_event` can getting broken automatons.

I think making one protocol with 64K possible repeated  key pressings is little over shutting so 2 args is more realistic (CSA-IOT have much brain capacity for doing things good).

Thanks for helping !

MattWestb, 2023-03-06:
Im going for 2 args and the events then looks like this:
```
event_type: zha_event
data:
  device_ieee: 1c:34:f1:ff:fe:78:57:15
  unique_id: 1c:34:f1:ff:fe:78:57:15:3:0xfc80
  device_id: 7b803a8f3912b70e1949639ccf9e0ad5
  endpoint_id: 3
  cluster_id: 64640
  command: MultiPressComplete
  args:
    - 0
    - 2
  params:
    Data1: 0
    Data2: 2
origin: LOCAL
time_fired: "2023-03-06T13:14:57.459885+00:00"
context:
  id: 01GTVGZ5HKM9SREKZ6VQ5Z4GHK
  parent_id: null
  user_id: null
```
If having wrong we is knowing it in next year or so and can fixing it then ;-)

MattWestb, 2023-03-06:
I have founding the ZZZ code that current Matter is using and its 2 args for it:
https://github.com/project-chip/connectedhomeip/blob/7e7b7afbef6bc89be966ed6cafe543f81fcf1199/zzz_generated/app-common/app-common/zap-generated/cluster-objects.h#L11246-L11253
```
    uint8_t previousPosition            = static_cast<uint8_t>(0);
    uint8_t totalNumberOfPressesCounted = static_cast<uint8_t>(0);
```
So i making 2 `int8s` args :-)))

MattWestb, 2023-03-07:
Im trying reusing DA from the V1 and using them for V2 and adding the extra one and have getting this working so good that ZHA is starting but i only getting the V1 DA.
```

    device_automation_triggers = IkeaSYMFONISKgen2V1.device_automation_triggers.copy()
    device_automation_triggers.update(
        {
            (SHORT_PRESS, BUTTON_1): {ENDPOINT_ID: 2, COMMAND: COMMAND_BUTTON_SINGLE},
            (DOUBLE_PRESS, BUTTON_1): {ENDPOINT_ID: 2, COMMAND: COMMAND_BUTTON_DOUBLE},
            (LONG_PRESS, BUTTON_1): {ENDPOINT_ID: 2, COMMAND: COMMAND_BUTTON_HOLD},
            (SHORT_PRESS, BUTTON_2): {ENDPOINT_ID: 3, COMMAND: COMMAND_BUTTON_SINGLE},
            (DOUBLE_PRESS, BUTTON_2): {ENDPOINT_ID: 3, COMMAND: COMMAND_BUTTON_DOUBLE},
            (LONG_PRESS, BUTTON_2): {ENDPOINT_ID: 3, COMMAND: COMMAND_BUTTON_HOLD},
        }
    )

```
Is im doing somthing wrong or is not not possible "importing" DAs and using it on one other class and adding more ADs for the new class ??

javicalle, 2023-03-07:
> Is im doing somthing wrong or is not not possible "importing" DAs and using it on one other class and adding more ADs for the new class ??

I don't see any problem in your code. IMO that must work. Have you deleted the `__pycache__` folder?

MattWestb, 2023-03-07:
_The cach_ i **never** deleting then it can being god old code storage in it that i can reusing !!  ;-))

I trying doing the good old cach deletion and restarting HA and not only reloading ZHA that is the easy to do then having problems with syntax (that i often dont understand).

Thanks for helping J !!

MattWestb, 2023-03-07:
Was getting problem with the Zram log is full and the system is not liking it :-((

But i can now doing 20 DA for the device !!!
```
Do something when…
Device offline
"Toggle" pressed
"Dim up" pressed
"Dim up" continuously pressed
"Dim down" pressed
"Dim down" continuously pressed
"Right" pressed
"Left" pressed
"button_1" InitialPress
"First button" pressed
"First button" double clicked
"First button" continuously pressed
"First button" released after long press
"button_2" InitialPress
"Second button" pressed
"Second button" double clicked
"Second button" continuously pressed
"Second button" released after long press
IKEA SYMFONISK sound remote gen2 Upgraded Identify has been pressed
IKEA SYMFONISK sound remote gen2 Upgraded Battery battery level changes
```
Dim up and down hold is repeating commands and right and left is only one command sent.
All is tested with automatons made from the device DA.

Only left how getting EP 2 and 3 bond to the coordinator from the quirk so the getting the commands.

MattWestb, 2023-03-07:
Updated dirty quirk only for V2 (updated fimware) if some like testing.
[symfonisk2.zip](https://github.com/zigpy/zha-device-handlers/files/10913074/symfonisk2.zip)

cooljimy84, 2023-03-08:
I think i'm on the old firmware for the gen2  (i've not enabled zha update or connected it to any thing other than HA)
HA reports Firmware: 0x00010012, I'll see if i can work out this custom quirk thing and test (if that would be any help?)

MattWestb, 2023-03-08:
Your remote is on factory firmware and using  V1 quirk.
I have getting the events working on both V1 and V2 but i having problem getting the command parsing working OK for using them in DA in one not to dirty way.
I have sniffing Dirigera updating one V1 remote and dome working file that is stgned and OK and the device is accepting it = OK for updating to V2 but is for the moment not recommended upgrading it then we need both V1 and V2 for testing but you can do it if you like.

cooljimy84, 2023-03-08:
I'll stick on v1 firmware for now, as it does all i need for now.
I've just tested the posted zip and i no longer get the "unknown cluster command" for the single dot and double dot buttons. But they also don't show any presses in the device log in Home Assistant.
Sorry if that was expected, but thought I'd report back

MattWestb, 2023-03-08:
The above posted quirk is only working with V2 and an can getting it doing events for V1 but i cant getting the DA working with V1.
Working but not so nice implanted is here https://github.com/zigpy/zha-device-handlers/pull/2236.

MattWestb, 2023-03-08:
@javicalle I can getting nice zha_events from both version with this code in INIT:
```
class ShortcutV1Cluster(EventableCluster):
    """Ikea Shortcut Button Cluster Variant 1."""

    name = "ShortcutClusterV1"
    cluster_id = IKEA_SHORTC_CLUSTER_ID

    server_commands = {
        0x01: foundation.ZCLCommandDef(
            "ShortCut",
            {
                "shortcut_mode": t.int8s,
                "Data": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
    }

class ShortcutV2Cluster(EventableCluster):
    """Ikea Shortcut Button Cluster Variant 2."""

    name = "ShortcutClusterV2"
    cluster_id = IKEA_MSWITCH_CLUSTER_ID

    server_commands = {
        0x00: foundation.ZCLCommandDef(
            "SwitchLatched",
            {
                "NewPosition": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x01: foundation.ZCLCommandDef(
            "InitialPress",
            {
                "NewPosition": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x02: foundation.ZCLCommandDef(
            "LongPress",
            {
                "PreviousPosition": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x03: foundation.ZCLCommandDef(
            "ShortRelease",
            {
                "PreviousPosition": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x04: foundation.ZCLCommandDef(
            "LongRelease",
            {
                "PreviousPosition": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x05: foundation.ZCLCommandDef(
            "MultiPressOngoing",
            {
                "NewPosition": t.int8s,
#                 "CurrentNumberOfPressesCounted": t.int8s, # Not implanted by IKEA for the moment.
            },
            False,
            is_manufacturer_specific=True,
        ),
        0x06: foundation.ZCLCommandDef(
            "MultiPressComplete",
            {
                "PreviousPosition": t.int8s,
                "TotalNumberOfPressesCounted": t.int8s,
            },
            False,
            is_manufacturer_specific=True,
        ),
    }
```
And events for V2:
```
  endpoint_id: 2
  cluster_id: 64640
  command: LongRelease
  args:
    - 0
  params:
    PreviousPosition: 0  

  endpoint_id: 2
  cluster_id: 64640
  command: ShortRelease
  args:
    - 0
  params:
    PreviousPosition: 0
```
And V1:
```
  endpoint_id: 1
  cluster_id: 64639
  command: ShortCut
  args:
    - 1
    - 1
  params:
    shortcut_mode: 1
    Data: 1

  endpoint_id: 1
  cluster_id: 64639
  command: ShortCut
  args:
    - 1
    - 3
  params:
    shortcut_mode: 1
    Data: 3
```
How shall i doing the V1 DA triggering on the  `command: ShortCut` and using the args ?
Also for V2 is doing different commands and i cont need the args but how do i getting them in the DA ??

I have patching commends from constants in the quirk so the names is matching and its working but its wrong way.

MattWestb, 2023-03-08:
I was defining one new command in the quirk like this `COMMAND_SHORTCUT = "ShortCut"` and doing the DA like this:
```
    device_automation_triggers = (
        Symfonisk2CommonTriggers.device_automation_triggers.copy()
    )
    device_automation_triggers.update(
        {
            (SHORT_PRESS, BUTTON_1): {
                COMMAND: COMMAND_SHORTCUT,
                PARAMS: {"shortcut_mode": 1, "Data": 1},
            },
            (DOUBLE_PRESS, BUTTON_1): {
                COMMAND: COMMAND_SHORTCUT,
                PARAMS: {"shortcut_mode": 1, "Data": 2},
            },
            (LONG_PRESS, BUTTON_1): {
                COMMAND: COMMAND_SHORTCUT,
                PARAMS: {"shortcut_mode": 1, "Data": 3},
            },
            (SHORT_PRESS, BUTTON_2): {
                COMMAND: COMMAND_SHORTCUT,
                PARAMS: {"shortcut_mode": 2, "Data": 1},
            },
            (DOUBLE_PRESS, BUTTON_2): {
                COMMAND: COMMAND_SHORTCUT,
                PARAMS: {"shortcut_mode": 2, "Data": 2},
            },
            (LONG_PRESS, BUTTON_2): {
                COMMAND: COMMAND_SHORTCUT,
                PARAMS: {"shortcut_mode": 2, "Data": 3},
            },
        },
    )
```
And the DA for button 1 and 2 is working OK.

Is this the right way to do it ?
Can i getting V2 parsing command type 1 - 6 to one command with type as args with the same  `COMMAND_SHORTCUT` so i only need one new command implanted (i have 5 done now and its feel wrong) ?

MrDodojo, 2023-03-09:
okay, will test when I'm at home. I don't understand, this remote uses zigbee and not matter? Why is it involved

MrDodojo, 2023-03-09:
With the quirk installed(without binding EP2 and 3), my other Ikea light remote stops working.

MattWestb, 2023-03-09:
@javicalle Thanks for help restarting HA and deleting cash was fixing changed DA not working.
Can you taking one look on the linked files and see if i have doing it right and if its OK running one test for getting all missing deletion and sort fixed (I can only doing black no isort and other tests) ?

javicalle, 2023-03-09:
This one?
https://github.com/zigpy/zha-device-handlers/issues/2223#issuecomment-1458673478

I'll take a look ASAP

MattWestb, 2023-03-09:
Its the right and its VERY DIRTY code !!
But i think its better you is looking before i trying doing one PR and must redoing  many things.

No hurry not so many user is having the device ;-))

javicalle, 2023-03-09:
I have some comments about the DA implementation.
First I must say that I haven't play around with all the DA too much and my assumptions can be wrong.

From what I understand in the DA are 2 parts:
* the action exposed to user to make the automation
* the device event that will trigger the automation

I'll take a simplified version from the quirk (just as an academic example to explain myself):
```python
    device_automation_triggers.update(
        {
            (PRESSED, BUTTON_1):        {ENDPOINT_ID: 2, COMMAND: COMMAND_PRESS},
            (SHORT_PRESS, BUTTON_1):    {ENDPOINT_ID: 2, COMMAND: COMMAND_BUTTON_SINGLE},
            (DOUBLE_PRESS, BUTTON_1):   {ENDPOINT_ID: 2, COMMAND: COMMAND_BUTTON_DOUBLE},
            (LONG_PRESS, BUTTON_1):     {ENDPOINT_ID: 2, COMMAND: COMMAND_BUTTON_HOLD},
            (LONG_RELEASE, BUTTON_1):   {ENDPOINT_ID: 2, COMMAND: COMMAND_RELEASE},
            (PRESSED, BUTTON_2):        {ENDPOINT_ID: 3, COMMAND: COMMAND_PRESS},
            (SHORT_PRESS, BUTTON_2):    {ENDPOINT_ID: 3, COMMAND: COMMAND_BUTTON_SINGLE},
            (DOUBLE_PRESS, BUTTON_2):   {ENDPOINT_ID: 3, COMMAND: COMMAND_BUTTON_DOUBLE},
            (LONG_PRESS, BUTTON_2):     {ENDPOINT_ID: 3, COMMAND: COMMAND_BUTTON_HOLD},
            (LONG_RELEASE, BUTTON_2):   {ENDPOINT_ID: 3, COMMAND: COMMAND_RELEASE},
        },
    )
```

In my head, the DA definition is a sort of map between the event that the device triggers and a identifier for helping users to make automations.

The expressions to the right must match the events that the device raises. They cannot be chosen freely (unless we use some implementation like the `TuyaSmartRemoteOnOffCluster` one).

The expressions to the left will be the ones exposed in HA and the ones that user will see. They must help to identify the action that generate the event (ie: "up button pressed", "double click left button", etc).

In the implementation _flake8_ complains about this:
```
zhaquirks/ikea/symfonisk2.py:66:1: F811 redefinition of unused 'COMMAND_PRESS' from line 16
zhaquirks/ikea/symfonisk2.py:67:1: F811 redefinition of unused 'COMMAND_BUTTON_DOUBLE' from line 16
zhaquirks/ikea/symfonisk2.py:69:1: F811 redefinition of unused 'COMMAND_BUTTON_SINGLE' from line 16
```

These constants are imported from `zhaquirks.const` but also redefined in the quirk.
That is the 'redefinition' (and some extra ones):
```python
COMMAND_PRESS = "InitialPress"  # New
COMMAND_BUTTON_DOUBLE = "MultiPressComplete"
COMMAND_BUTTON_HOLD = "LongPress"
COMMAND_BUTTON_SINGLE = "ShortRelease"
COMMAND_RELEASE = "LongRelease"
```

As all these already exists in the `const` file we must use another identifiers to avoid confussion.
My suggestion would be something like:
```python
COMMAND_INITIAL_PRESS = "InitialPress"  # New
COMMAND_MULTI_PRESS_COMPLETE = "MultiPressComplete"
COMMAND_LONG_PRESS = "LongPress"
COMMAND_SHORT_RELEASE = "ShortRelease"
COMMAND_LONG_RELEASE = "LongRelease"
```
And if I had understant correctly the code, these constants must be used in the DA part.

About the left expressions I'm confused about that `InitialPress` action. I'm not sure if users would understand the meaning.
From the total ignorance of the device and the actions it supports, my proposal would be to use `SHORT_PRESS` as 'InitialPress' and `SHORT_RELEASE` as 'ShortRelease' (both from the `const`).

So, it would be more or less:
```python
    device_automation_triggers.update(
        {
            (SHORT_PRESS, BUTTON_1):    {ENDPOINT_ID: 2, COMMAND: COMMAND_INITIAL_PRESS},
            (SHORT_RELEASE, BUTTON_1):  {ENDPOINT_ID: 2, COMMAND: COMMAND_SHORT_RELEASE},
            (DOUBLE_PRESS, BUTTON_1):   {ENDPOINT_ID: 2, COMMAND: COMMAND_MULTI_PRESS_COMPLETE},
            (LONG_PRESS, BUTTON_1):     {ENDPOINT_ID: 2, COMMAND: COMMAND_LONG_PRESS},
            (LONG_RELEASE, BUTTON_1):   {ENDPOINT_ID: 2, COMMAND: COMMAND_LONG_RELEASE},
            (SHORT_PRESS, BUTTON_2):    {ENDPOINT_ID: 3, COMMAND: COMMAND_INITIAL_PRESS},
            (SHORT_RELEASE, BUTTON_2):  {ENDPOINT_ID: 3, COMMAND: COMMAND_SHORT_RELEASE},
            (DOUBLE_PRESS, BUTTON_2):   {ENDPOINT_ID: 3, COMMAND: COMMAND_MULTI_PRESS_COMPLETE},
            (LONG_PRESS, BUTTON_2):     {ENDPOINT_ID: 3, COMMAND: COMMAND_LONG_PRESS},
            (LONG_RELEASE, BUTTON_2):   {ENDPOINT_ID: 3, COMMAND: COMMAND_LONG_RELEASE},
        },
    )
```

MattWestb, 2023-03-10:
Thanks for trying getting my understanding the the logic !!
I was trying using the command i was defining in the INIT for V1 on the DA directly and it was not working only after doing one define of the command in the quirk the DA was working (and deleting the cache and restarting HA as we is knowing).

Then its little more political if i shall putting in the "matter switch commands" in const for getting one more clean implementation then i like keeping the matter names that is making it easy implanting new switches that is using  the standard and can doing long and multiple press very easy, and not doing one mix of old standard and new that is braking many thing and making it crappy in the future.

All commands sent from the switch is starting with one initial press and and then one short release, long held or multi press continuing and is the the last 2 its flowing by more commands as long release or multi press complete or continuing so all versions can being done if reading the args right.

'InitialPress' cant being user as short  press as you is getting it for all button pressings but the 'ShortRelease' is working 110% OK.
I can skipping the DA for 'InitialPress'  but i can see some user case for it then users is understanding  how its working and then we is back shall we doing matter commands in the const and doing it good implanted for the future or only putting it it so its working with partly old names ?

From HA log the doing single double and long press with DA being triggered from V2:
```

Unknown device Zha Event was fired triggered by automation 1LR triggered by event 'zha_event'
10:15:52 - Now
1LR triggered by event 'zha_event'
10:15:52 - Now
IKEA SYMFONISK sound remote gen2 Upgraded Remote Button Long Release - Button 1 event was fired with parameters: {'PreviousPosition': 0}
10:15:52 - Now
Unknown device Zha Event was fired triggered by automation 1H triggered by event 'zha_event'
10:15:51 - 2 seconds ago
1H triggered by event 'zha_event'
10:15:51 - 2 seconds ago
IKEA SYMFONISK sound remote gen2 Upgraded Remote Button Long Press - Button 1 event was fired with parameters: {'PreviousPosition': 0}
10:15:51 - 2 seconds ago
Unknown device Zha Event was fired triggered by automation 1I triggered by event 'zha_event'
10:15:50 - 2 seconds ago
1I triggered by event 'zha_event'
10:15:50 - 2 seconds ago
IKEA SYMFONISK sound remote gen2 Upgraded Initialpress - Button 1 event was fired with parameters: {'NewPosition': 0}
10:15:50 - 2 seconds ago
Unknown device Zha Event was fired triggered by automation 12 triggered by event 'zha_event'
10:15:48 - 5 seconds ago
12 triggered by event 'zha_event'
10:15:48 - 5 seconds ago
IKEA SYMFONISK sound remote gen2 Upgraded Remote Button Double Press - Button 1 event was fired with parameters: {'PreviousPosition': 0, 'TotalNumberOfPressesCounted': 2}
10:15:48 - 5 seconds ago
Unknown device Zha Event was fired triggered by automation 1I triggered by event 'zha_event'
10:15:48 - 5 seconds ago
1I triggered by event 'zha_event'
10:15:48 - 5 seconds ago
IKEA SYMFONISK sound remote gen2 Upgraded Initialpress - Button 1 event was fired with parameters: {'NewPosition': 0}
10:15:48 - 5 seconds ago
Unknown device Zha Event was fired triggered by automation 1 triggered by event 'zha_event'
10:15:24 - 1 minute ago
1 triggered by event 'zha_event'
10:15:24 - 1 minute ago
IKEA SYMFONISK sound remote gen2 Upgraded Remote Button Short Press - Button 1 event was fired with parameters: {'PreviousPosition': 0}
10:15:24 - 1 minute ago
Unknown device Zha Event was fired triggered by automation 1I triggered by event 'zha_event'
10:15:23 - 1 minute ago
1I triggered by event 'zha_event'
10:15:23 - 1 minute ago
IKEA SYMFONISK sound remote gen2 Upgraded Initialpress - Button 1 event was fired with parameters: {'NewPosition': 0}
10:15:23 - 1 minute ago
Unknown device Zha Event was fired triggered by automation Toggle triggered by event 'zha_event'
10:15:21 - 
```
PS: the extra commands i was defined in the quirk with matter types was done for being putted in the const then have all working OK and doing the final PR.

I must doing more testing and see if i can doing it better but i think i going for matter commands in the const but is no use if you is saying on then its only wasted time for my side.

MattWestb, 2023-03-10:
I was looking on command for const or in IKEA INIT and i think this looks OK and is easy to use:
```
COMMAND_M_SWLATCHED = "SwitchLatched"
COMMAND_M_INITIAL_PRESS = "InitialPress"
COMMAND_M_LONG_PRESS = "LongPress"
COMMAND_M_SHORT_RELEASE = "ShortRelease"
COMMAND_M_LONG_RELEASE = "LongRelease"
COMMAND_M_MULTI_PRESS_ONGOING = "MultiPressOngoing"
COMMAND_M_MULTI_PRESS_COMPLETE = "MultiPressComplete"
```
The Matter"ZCL" looks like this:
```
1.11.7. Events
This cluster SHALL support these events:
Id Name Access Conformance
0 SwitchLatched V LS
1 InitialPress V MS
2 LongPress V MSL
3 ShortRelease V MSR
4 LongRelease V MSL
5 MultiPressOngoing V MSM
6 MultiPressComplete V MSM
```
I was putting one _M_ after COMMAD for marking its one Matter command.
I see 2 ways one is putting then in const and it can being used for new devices that is coming from IKEA and also from  other manufactures if doing it nice and right. 
(Or HA Matter integration shall taking care of the commands ???
That is one question for the maintainers for not getting one more tuya MCU chaos)
The other way is pitting then in IKEA INT and importing them to the quirk for getting it working OK.

Then i hove only 2 that is not so logic how to doing with: 
```
TOGGLE = "Toggle"
COMMAND_SHORTCUT = "ShortCut"
```
The TOGGLE you is not like but i need it for getting the logic working for the user.
The commands from V1 i must using one command for getting the DA working.
Is it  working if i putting it in the IKEA INIT like so the code is "extracting" the command to "ShortCut" and the DA can using it in the quirk after importing it from the IKEA INIT ?
I testing if its working then is no use putting local custom commands in the const its only making it  crappy.

MattWestb, 2023-03-10:
I have doing Matter commands in IKEA INIT and only 2 "local" const in the quirk and all is working OK with one small cosmetic thing. [symfonisk2B.zip](https://github.com/zigpy/zha-device-handlers/files/10941924/symfonisk2B.zip)

All DA for V1 and V2 is working OK !!
But im using local const for `PRESSED = "InitialSwitchPress"` and using it in the DA like this `(PRESSED, BUTTON_1): {ENDPOINT_ID: 2, COMMAND: COMMAND_M_INITIAL_PRESS},`
The DA is working but ZHA cant finding the alias for `BUTTON_1` (its "First button") and its not looking nice in the GUI then open adding DA as the other DA for the same button:
```
"button_1" InitialSwitchPress
"First button" pressed
"First button" double clicked
"First button" continuously pressed
"First button" released after long press
```
I was trying importing from const `PRESS_TYPE = "press_type"` and using `PRESS_TYPE` in the DA but it was doing the same and i dont knowing way.

If accepting the cosmic glitch is all working OK and i can making one PR.
But i like have `TOGGLE = "Toggle"` and `PRESSED = "InitialSwitchPress"` in const for using with other quirks in the future. Also can the Matter Switch commands can being moved to const in the future but is not needed for getting it working OK.

@javicalle Can you then you having time doing one precommit checking in your dev env so isort and import is OK and dont need doing so many fixes in the PR ?
I think Black is OK then i have running black blacground on the files.

Thanks in advance !!

MrDodojo, 2023-03-10:
The quirk you just sent works(and my other remote also works again) but doesn't have long press released triggers and right, left and toggle don't have continuously released nor long press released triggers, is this intentional?

MrDodojo, 2023-03-10:
Factory firmware.

It makes sense they didn't add long press to left, right and toggle because it isn't used by ikea either, only advanced users would get a use out of it.

MattWestb, 2023-03-10:
I think its little sad they can have implanted the same functions as with the first curtcut button on the toggle with toggle dim up and dim down then we was having little more functions to pay with. Also right and left from the old 5 buttons scene command have giving more functions but this new devices is 110% made for Dirigera and its the ZBGW that is doing all the magic that the controller was doing before.

The updated firmware is giving more functions on the shortcut buttons but is little more tricky to using well (initial button pressed and reseal after hold).

MattWestb, 2023-03-10:
V1 is only sending 101 = Button 1 short, 102 = Button double and 103 = Button 1 held and the same for button 2 with commands 201,202 and 203 so no long reseal only if you updating the firmware but i recommending waiting to do that until IKEA have putting it out for TF with release notes then we **cant** roiling back the firmware in one easy way if its not good.

You can outing in debug log for ZHA and you can see in the log the command the device is sending to the system.

PS: We can roiling back to the V2 firmware then i have sniffing the Dirigera OTA and have building one working firmware of it but its not for normal users if getting problems but V1 we dont having and cant getting then the chip is debug locked.

MrDodojo, 2023-03-10:
[debuglogZHA.log](https://github.com/zigpy/zha-device-handlers/files/10943453/debuglogZHA.log)
I continously pressed the UP button for 10 seconds and then released it. What a bummer V1 doesnt have long press release

[29 more replies]

PR #2273 (2023-03-25): Add IEKA Symfonisk Gen2 v1/v2 support
Add DA for Symfonisk Gen 2 with shipping firmware (V1) and first OTA updated via Dirigera (V2).

The V1 is using raw commands on EP1.
V2 is using Matter Switch ZCL commands on 2 extra EP but with IKEA manufacture cluster then the Matter cluster is not in ZCL R8. V2 also need the 2 extra EP being bounded and one PR is made in ZHA for doing it nice but this version is using OnOff Custer as one temporary solution until the ZHA PR is merged.

Also V1 is using IKEA old battery reporting (100%) and V2 is using Zigbee standard (200%) so have adding new updated po [... cut]
closed 2023-03-25 by TheJulianJES