zigpy/zha-device-handlers #2224: [BUG] Aqara Contact Sensor E1 not showing state & battery

dodekaphilist, 2023-02-25:
**Describe the bug**
I connected the Aqara Door & Window Sensor E1 via ZHA which was successful and it's showing up in the dashboard. The Identify button works, but the state is always off and the battery is always unavailable. I tried it with a second sensor with the same behaviour.

**To Reproduce**
Steps to reproduce the behavior:
1. Pair Sensor
2. State & Battery info not updating

**Expected behavior**
Actual sensor information available

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.magnet.acn001",
  "class": "zhaquirks.xiaomi.aqara.magnet_acn001.MagnetE1"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.2.5",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "4.4.180+",
    "run_as_root": true
  },
  "custom_components": {
    "openrgb": {
      "version": "2.2",
      "requirements": [
[... 525 lines cut ...]
[0x4899:3:0x0300] Attribute report received: color_temperature=250
[0x8048](lumi.weather): Device seen - marking the device available and resetting counter
[0x8048](lumi.weather): Update device availability -  device available: True - new availability: True - changed: False
[0x719F](TS0505B): Device seen - marking the device available and resetting counter
[0x719F](TS0505B): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x8048), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=None, profile_id=260, cluster_id=0, data=Serialized[b'\x18\xce\n\x01\xffB%\x01!\x9f\x0b\x04!\xa8C\x05!\xe7\x05\x06$\x01\x00\x01\x00\x00d)\x8a\x07e!\x0e\x1cf+\x0f\x82\x01\x00\n!\x99H'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-62)
[0x8048:1:0x0000] Received ZCL frame: b'\x18\xce\n\x01\xffA%\x01!\x9f\x0b\x04!\xa8C\x05!\xe7\x05\x06$\x01\x00\x01\x00\x00d)\x8a\x07e!\x0e\x1cf+\x0f\x82\x01\x00\n!\x99H'
[0x8048:1:0x0000] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=206, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
[0x8048:1:0x0000] Decoded ZCL frame: BasicCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0xFF01, value=TypeValue(type=LVBytes, value=b'\x01!\x9f\x0b\x04!\xa8C\x05!\xe7\x05\x06$\x01\x00\x01\x00\x00d)\x8a\x07e!\x0e\x1cf+\x0f\x82\x01\x00\n!\x99H'))])
[0x8048:1:0x0000] Received command 0x0A (TSN 206): Report_Attributes(attribute_reports=[Attribute(attrid=0xFF01, value=TypeValue(type=LVBytes, value=b'\x01!\x9f\x0b\x04!\xa8C\x05!\xe7\x05\x06$\x01\x00\x01\x00\x00d)\x8a\x07e!\x0e\x1cf+\x0f\x82\x01\x00\n!\x99H'))])
[0x8048:1:0x0000] Attribute report received: 0xFF01=b'\x01!\x9f\x0b\x04!\xa8C\x05!\xe7\x05\x06$\x01\x00\x01\x00\x00d)\x8a\x07e!\x0e\x1cf+\x0f\x82\x01\x00\n!\x99H'
[0x8048:1:0x0001] Voltage mV: [Min]:2820 < [RAW]:2975 < [Max]:3100, Battery Percent: 55.5
[0x9449](TS0601): last_seen is 2476751.4956195354 seconds ago and ping attempts have been exhausted, marking the device unavailable
[0x9449](TS0601): Update device availability -  device available: False - new availability: False - changed: False
[0x4899](CLA60 RGBW OSRAM): Device seen - marking the device available and resetting counter
[0x4899](CLA60 RGBW OSRAM): Update device availability -  device available: True - new availability: True - changed: False
[0xD8D2](lumi.sensor_switch.aq2): Device seen - marking the device available and resetting counter
[0xD8D2](lumi.sensor_switch.aq2): Update device availability -  device available: True - new availability: True - changed: False
[0x8048](lumi.weather): Device seen - marking the device available and resetting counter
[0x8048](lumi.weather): Update device availability -  device available: True - new availability: True - changed: False
[0x719F](TS0505B): Device seen - marking the device available and resetting counter
[0x719F](TS0505B): Update device availability -  device available: True - new availability: True - changed: False
```

</details>

TheJulianJES, 2023-02-25:
What coordinator are you using? If it's a Conbee or TI coordinator, make sure the firmware is somewhat recent.

dodekaphilist, 2023-02-25:
@TheJulianJES I am using a Conbee II stick. I couldn't figure out, which firmware I was running so I updated it to the newest available version using GCFFlasher. After re-pairing the sensor, it now works flawlessly.
Would be a nice thing, if ZHA would actually be able to do these kinds of firmware updates. But that's another story. Thanks for your help!

closed 2023-02-26 by dodekaphilist