zigpy/zha-device-handlers #2226: [Device Support Request] Philips Hue White Ambiance Still ceiling lamp

carp3-noctem, 2023-02-26:
**Is your feature request related to a problem? Please describe.**
Device is listed as unknown device and unknown Manufacturer
https://www.philips-hue.com/de-de/p/hue-white-ambiance-still-deckenleuchte/8719514341357

**Describe the solution you'd like**
Add device to ZHA Database.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=128, server_mask=11264, maximum_outgoing_transfer_size=128, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "11": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xfc03"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "unk_manufacturer",
  "model": "unk_model",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
[... 532 lines cut ...]
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
no additional logs
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

TheJulianJES, 2023-02-26:
Hmm, I'm pretty sure that device should be supported out-of-the-box. My guess is that it didn't pair completely.
Could you try to remove it from ZHA and then re-pair that device?

Note: If it still identifies as ``unk_manufacturer`` or ``unk_model``, make sure your Zigbee coordinator is on an USB extension. Maybe there's some kind of interference issues that prevent ZHA and the light to complete the interview.

carp3-noctem, 2023-02-26:
I did have a Extension (the one that comes with the SkyConnect Stick) between the Pi and the Stick.
I will try the Repair as soon as I'm back on location and will report back.

carp3-noctem, 2023-02-26:
Ok, this can be closed, adding it again, first showed the same behaviour.
Resetting the lamp with the Phillips Hue remote and redo the same now shows a working instance with parameters. Thanks for the Hint!

closed 2023-02-26 by carp3-noctem