zigpy/zha-device-handlers #2233: [Device Support Request]

rahulpdev, 2023-02-27:
**Is your feature request related to a problem? Please describe.**
[Samotech SM323](https://zigbee.blakadder.com/Samotech_SM323.html) is recognised by ZHA but is not performing some basic functions correctly in Home Assistant. 

When toggling the state of the light from Home Assistant, if the state is turned On the device turns On but the state reported in Home Assistant toggles itself back to Off. Home Assistant now considers the state of the device to be Off and so the device can't be turned Off by the control in Home Assistant. 

The same thing happens the other way around, if the state is turned Off the device turns Off but the state reported in Home Assistant toggles itself back to On. Home Assistant now considers the state of the device to be On and so the device can't be turned On by the control in Home Assistant. 

The physical device otherwise performs as normal, and the brightness control in Home Assistant also performs as normal.

**Describe the solution you'd like**
Toggling the light control in Home Assistant results in the correct state persisting in Home Assistant.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4107, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0101",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0b05",
        "0x1000"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "Samotech",
  "model": "HK_DIM_A",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
[... 773 lines cut ...]
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

The device displays in Home Assistant as HK_DIM_A by Samotech. I also have in Home Assistant a [HK_DIM_A by Candeo](https://zigbee.blakadder.com/Candeo_HK-DIM-A.html) that is a physically identical device and that works perfectly.

JohnHL12, 2023-02-28:
Strangely enough i have a HK-DIM-A Candeo device that is behaving in the exact same way as your Samotech device.

Through messing about i have found the "correct" codes to turn the device on and off and it reflect in HA those being

Clusters
OnOff (Endpoint id: 1, Id: 0x0006, Type: in)

Commands of the selected cluster
on (id: 0x0001)  for on 

and

Commands of the selected cluster
off (id: 0x0000) for off.

These work perfectly in HA when issued manually however rewriting the code is a little beyond me.

I am therefore wondering if these switching are using different Zcl versions meaning some work and some don't

Are you able to check the Zcl version of your working and non working versions?

rahulpdev, 2023-03-01:
> Strangely enough i have a HK-DIM-A Candeo device that is behaving in the exact same way as your Samotech device.
>
> Through messing about i have found the "correct" codes to turn the device on and off [...]

I'm still fairly new to Home Assistant, what specifically do you mean by **Zcl version**? Thanks!

JohnHL12, 2023-03-01:
If you go into the device in ZHA and click on the 3 dots next to reconfigure it will show you the clusters and under the attributes drop down menu there is a zcl version you should be able to click read on and will give you a value.

rahulpdev, 2023-03-01:
> If you go into the device in ZHA and click on the 3 dots next to reconfigure it will show you the clusters and under the attributes drop down menu there is a zcl version you should be able to click read on and will give you a value.

Ah, thanks. The Samotech reports a value of 8 and Candeo reports a value of 3.

JohnHL12, 2023-03-01:
> Ah, thanks. The Samotech reports a value of 8 and Candeo reports a value of 3.

Hmmm intresting, my new Candeo reports a value of 8 and has the exact same behaviour as your Samotech device. It seems at some point the HK-DIM-A devices have changed and thats broken the ZHA intergration? However if this was updated all the devices with 3 would maybe stop working. Might be fixed with a blue print or script but sadly that's getting outside my level of experience.

rahulpdev, 2023-03-02:
For completeness :

<google-sheets-html-origin><style type="text/css"></style>

Attribute | id | Samotech | Candeo
-- | -- | -- | --
app_version | 0x0001 | 0 | 3
cluster_revision | 0xfffd | 3 | 1
date_code | 0x0006 | 20220531 | 20210128
hw_version | 0x0003 | 0 | 1
manufacturer | 0x0004 | Samotech | Candeo
model | 0x0005 | HK_DIM_A | HK_DIM_A
stack_version | 0x0002 | 0 | 6
sw_build_id | 0x4000 | 1 | 1
zcl_version | 0x0000 | 8 | 3

JohnHL12, 2023-03-03:
So after messing about with this i found a workaround but it's far from ideal. It does seem that the device itself does inform HA that it's status has changes but this takes about 15mins after which the toggle in HA goes to the correct position which is just silly.

The work around for now is i have created an automation using the ZHA tool kit that polls the device for its status after the device is triggered this seems to require a 7 second delay to function.

rahulpdev, 2023-03-04:
Oh great! Could you share the yaml and I'll see if it works on the Samotech?

rahulpdev, 2023-03-04:
> So after messing about with this i found a workaround but it's far from ideal. It does seem that the device itself does inform HA that it's status has changed but this takes about 15mins after which the toggle in HA goes to the correct position which is just silly.
> 
> The work around for now is i have created an automation using the ZHA tool kit that polls the device for its status after the device is triggered this seems to require a 7 second delay to function.

Oh great! Could you share the yaml and I'll see if it works on the Samotech?

TheJulianJES, 2023-03-04:
If you have [zha-toolkit](https://github.com/mdeweerd/zha-toolkit), you could try to call something like this to poll on/off:
```yaml
service: zha_toolkit.attr_read
data:
  ieee: light.your_light
  endpoint: 1
  cluster: 6
  attribute: 0
```
Also, you could you start a reconfiguration of the device and send a screenshot of what the reconfiguration dialog looks like?

JohnHL12, 2023-03-05:
> Oh great! Could you share the yaml and I'll see if it works on the Samotech?

As TheJulianJes said but with the trigger:

alias: Dimmer testing
description: ""
trigger:
  - platform: event
    event_type: call_service
condition: []
action:
  - delay:
      hours: 0
      minutes: 0
      seconds: 7
      milliseconds: 0
    enabled: true
  - service: zha_toolkit.attr_read
    data:
      ieee: light.shyugj_hk_dim_a_light
      cluster: 6
      attribute: 0
mode: single

You'll need to install the zha toolkit first (https://github.com/mdeweerd/zha-toolkit) the device will still flip itself due to the delay but should set correctly once the automation fires. The 7 second delay seemed to be the minimum i could set to get it to work

TheJulianJES, 2023-03-05:
@JohnHL12 Could you also start a reconfiguration of the device and send a screenshot of the reconfiguration dialog? (Settings -> Integrations -> ZHA: Configure -> Devices -> select your device -> Reconfigure button)
(Also see if the issue changes after that)

Also, next to the reconfiguration button, there are three dots. Can you click that and then "Manage Zigbee device" -> Bindings tab -> select your Zigbee coordinator in the first drop-down and click "Bind". (See if the issue changes)
And maybe try "unbind", wait 10 seconds, and then "bind" again and see if the issue changes.

JohnHL12, 2023-03-05:
> Oh great! Could you share the yaml and I'll see if it works on the Samotech?

@rahulpdev Just as @TheJulianJES said with the trigger just for ease of use, you need to add the took kit linked above first.

There will still be some flapping of the toggle as couldn't get it to work with a delay less than the 7 seconds but least it will settle in the correct place once the automation fires.

(@TheJulianJES Deleted my previous comment to add this in the correct format to enable copying )

```yaml
alias: Dimmer testing
description: ""
trigger:
  - platform: event
    event_type: call_service
condition: []
action:
  - delay:
      hours: 0
      minutes: 0
      seconds: 7
      milliseconds: 0
    enabled: true
  - service: zha_toolkit.attr_read
    data:
      ieee: light.shyugj_hk_dim_a_light
      cluster: 6
      attribute: 0
mode: single

JohnHL12, 2023-03-05:
@TheJulianJES have done the reconfigure before and doesn't make a difference to the way to device acts. Have also now tried unbinding and rebinding  didn't do anything even after another reconfigure.

Just strange to me that if you send the cluster commands through manage device manually it responds fine as I'd assume (maybe wrongly) that those are the commands normally sent.

 I'll run the debug logs and send them across

[image]

JohnHL12, 2023-03-05:
@TheJulianJES 

I've sent the logs of attempting to turn the device on and then off.

I maybe wrong but the line 

2023-03-05 09:00:32.094 DEBUG (MainThread) [zigpy.zcl] [0x499C:1:0x0006] Decoded ZCL frame: OnOff:Default_Response(command_id=1, status=<Status.UNSUP_CLUSTER_COMMAND: 129>)

does seem to me like there is an issue with the response from the device?

javicalle, 2023-03-07:
I seems that `0x499C` device doesn't accept the HA commands. 
Are you sure that this is the same device? 
You can check it from the `Nwk` value, device view --> "Zigbee info"

rahulpdev, 2023-03-12:
@TheJulianJES I reconfigfured my Samotech device and returned the same results as @JohnHL12

andhya, 2023-03-13:
I've got the same Samotech 323, and trying any of the OnOff commands in 'manage ZigBee device' I also get <Status.UNSUP_CLUSTER_COMMAND: 129>

[image]

dhc25, 2023-03-21:
@rahulpdev @JohnHL12 
Can you please both advise the "#" number on the back of your Candeo modules?
It appears that different "#" batches are performing differently with ZHA.

Teevolution, 2023-03-22:
Just had a response from Samotech on this issue this morning.

Hello,

 

Yes, the issues are now fixed in the latest firmware that will be available next month. Please get back in touch by the end of next month.

 

Kind regards,

 

SAMOTECH

JohnHL12, 2023-03-22:
I unfortunately returned mine so can't supply anymore details

thestraycat, 2023-03-31:
Hi guys - Eagerly following this... I currently have a candeo dimmer which was bought about a week ago... I'm looking at buying the samotechs as there cheaper and as mentioned above releasing a new firmware. Do we know if the candeo dimmers already have this firmware and whether samotech are playing catchup? Or is this a new firmware that neither candeo or samotech have had previously?

leomarkt, 2023-04-12:
Heads up! I just got my dimmers back from samotech after a firmware upgrade, and now they're all working like a charm in ZHA. Hands down that's the best customer service around!

**Before firmware upgrade:**

[image]

**After firmware upgrade:**

[image]

rahulpdev, 2023-04-17:
@leomarkt that's frustrating as mine came back from Samotech a few weeks ago and still had the same problem. I ended up returning the devices that Samotech was good enough to give me a refund.

rahulpdev, 2023-05-01:
My unit has the `date_code` 20220531 and I'm also experiencing Problem 2 so it seems there is no permanent fix for this device.

leomarkt, 2023-05-01:
> My unit has the `date_code` 20220531 and I'm also experiencing Problem 2 so it seems there is no permanent fix for this device.

The fix is to get it reflashed with the latest firmware from Samotech.

yankeeinlondon, 2023-06-11:
is there no way to flash these devices at home? I am only in the UK for a few more days and would like to close this loop

thestraycat, 2023-12-13:
My issue is still happening on the latested version of HA and Z2M.

leomarkt, 2023-12-13:
> My issue is still happening on the latested version of HA and Z2M.

and what's your issue?

closed 2024-06-17 by github-actions[bot]