zigpy/zha-device-handlers #2239: [Device Support Request] Immax NEO Smart remote control v2

lafriks, 2023-02-28:
**Is your feature request related to a problem? Please describe.**

Device can be connected but no buttons are working. I got it working using such quirks definition:
https://gist.githubusercontent.com/tetele/66b620677bcf0f09caa38e5d940fd6b7/raw/f5afb533936f7938fe5e9f7fefab98c25f262cc7/_TYZB01_7qf81wty_TS1001.py

**Describe the solution you'd like**

Would be nice to add it this repository to support it without need to add special quirks. Signature below is with quirks from above link applied.

<details>
<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0104",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x1000"
      ],
      "out_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0019",
        "0x1000"
      ]
    }
  },
  "manufacturer": "_TYZB01_7qf81wty",
  "model": "TS1001",
  "class": "_TYZB01_7qf81wty_TS1001.ImmaxNeo07087_2"
}
```

</details>

<details>
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.2.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Riga",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.01.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
[... 319 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0006": {
            "endpoint_attribute": "on_off",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0008": {
            "endpoint_attribute": "level",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x1000": {
            "endpoint_attribute": "lightlink",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

lafriks, 2023-03-01:
If you could give a hint about what would be correct dir name and filename I could prepare PR for this

javicalle, 2023-03-02:
I'm going to suggest:
* filename: `zhaquirks/tuya/ts1001.py`
* class name: `TuyaRemoteTS1001`
* If you want keep the 'Immax NEO 07087-2' reference beside the `MODELS_INFO`:
```python
class TuyaRemoteTS1001(TuyaDimmerSwitch):
    """TS1001 remote control with triggers."""

    signature = {
        MODELS_INFO: [("_TYZB01_7qf81wty", "TS1001")],  # Immax NEO 07087-2 remote control
        ENDPOINTS: {
```
* file header: `"""Tuya TS1001 devices."""`

lafriks, 2023-09-01:
👍

lafriks, 2024-02-28:
👍

lafriks, 2024-08-27:
👍🏻

closed 2025-08-20 by github-actions[bot]