zigpy/zha-device-handlers #2241: [BUG] Errors in custom zha quirks breaking ZHA

ScratMan, 2023-03-01:
**Describe the bug**
When there is an error in a custom ZHA quirks, Home Assistant fails to start ZHA.
Faced it today when updating from HA 2023.2.5 to 2023.3.0
I have a custom quirks for a tuya valve that was importing     TuyaDPType from zhaquirks.tuya.mcu ; but is has been moved to zhaquirks.tuya in latest release.
When starting, ZHA failed to load with error message:

```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/zhaquirks/ts0601_TZE200_akjefhj5_valve.py:25
First occurred: 00:28:30 (1 occurrences)
Last logged: 00:28:30

Error setting up entry ConBee II, s/n: DE2463913 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/zhaquirks/ts0601_TZE200_akjefhj5_valve.py", line 25, in <module>
    from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/mcu/__init__.py)

 

```
**To Reproduce**
Steps to reproduce the behavior: (use 2023.3.0)
1. Insert an import error in a custom ZHA quirk
2. Reload ZHA integration or restart HA
3. Check state of ZHA or load the log
4. See error

**Expected behavior**
Errors in custom ZHA quirks should only break device's operation, but ZHA should continue starting and operate normally.

<details>
<summary>Additional logs</summary>

```
Logger: homeassistant.config_entries
Source: custom_zha_quirks/zhaquirks/ts0601_TZE200_akjefhj5_valve.py:25
First occurred: 00:28:30 (1 occurrences)
Last logged: 00:28:30

Error setting up entry ConBee II, s/n: DE2463913 - dresden elektronik ingenieurtechnik GmbH for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 100, in async_setup_entry
    setup_quirks(config)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/__init__.py", line 409, in setup
    importer.find_module(modname).load_module(modname)
  File "<frozen importlib._bootstrap_external>", line 548, in _check_name_wrapper
  File "<frozen importlib._bootstrap_external>", line 1063, in load_module
  File "<frozen importlib._bootstrap_external>", line 888, in load_module
  File "<frozen importlib._bootstrap>", line 290, in _load_module_shim
  File "<frozen importlib._bootstrap>", line 719, in _load
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_zha_quirks/zhaquirks/ts0601_TZE200_akjefhj5_valve.py", line 25, in <module>
    from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/mcu/__init__.py)

```

</details>

TheJulianJES, 2023-03-02:
I think this was actually intended, but judging from the amount of people that use custom quirks, this may not be the greatest idea.

See this PR:
- https://github.com/zigpy/zha-device-handlers/pull/2242

aquemy, 2023-03-02:
Same error here.

javicalle, 2023-03-02:
Just remove all the `dp_type=TuyaDPType.XXXXXX` from the `dp_to_attribute` part:
```python
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            MotionSensorCluster.ep_attribute,
            "zone_status",
            dp_type=TuyaDPType.BOOL,   # <-- REMOVE FROM ALL DPs
            converter=lambda x: IasZone.ZoneStatus.Alarm_1 if not x else 0,
        ),
        .../...
```

Remove also the `TuyaDPType` from the `import` part.

antoweb, 2023-03-02:
> Just remove all the `dp_type=TuyaDPType.XXXXXX` from the `dp_to_attribute` part:
>
> ```python
>     dp_to_attribute: Dict[int, DPToAttributeMapping] = {
>         1: DPToAttributeMapping( [...]

Removing these lines zha start wothout erros but my device tuya t601 wont work

ScratMan, 2023-03-02:
I solved the issue by importing TuyaDPType from zhaquirks.tuya and remove the dp_type= in attributes mapping.
See here : https://github.com/zigpy/zha-device-handlers/discussions/1660#discussioncomment-5172286

javicalle, 2023-03-03:
But you haven't removed the references:
```python
    dp_to_attribute: Dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
            TuyaDPType.BOOL,  # <-- these
        ),
```

All these `TuyaDPType.XXXX` aren't used any more. Remove it and the import.

ScratMan, 2023-03-03:
Yes, i discovered later on that it causes some issues at runtime.

besiktas97, 2023-03-07:
@javicalle I did what you said, but it still doesnt work. Can you check if the code is good ?

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

from zhaquirks.tuya import (
    NoManufacturerCluster,
    TuyaLocalCluster,
    TuyaNewManufCluster,
)
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""
    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""

    def __init__(self, *args, **kwargs):
        """Init."""
        super().__init__(*args, **kwargs)
        self._update_attribute(
[... 237 lines cut ...]
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarMaxRange,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
        }
    }

closed 2023-09-14 by github-actions[bot]