zigpy/zha-device-handlers #2243: [BUG] Tuya ZHA quirk: data type missing

SanchYESS, 2023-03-02:
https://github.com/zigpy/zha-device-handlers/blob/dda8d8a0e99692813f34d7e0518eecfdd7a02ece/zhaquirks/tuya/__init__.py#L252

Got an error message after updating core from 2023.2.5 to 2023.3.0:
```
Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 592, in on_af_message
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 875, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 381, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
    self.endpoints[src_ep].handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 485, in handle_cluster_request
    zvalue = ztype(tuya_data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 93, in __new__
    n = super().__new__(cls, *args, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 253, in __int__
    return ints[self[0]].deserialize(bytes(reversed(self[1:])))[0]
KeyError: 8
```
Seems like my power meter (_TZE200_lsanae15, [custom quirk](https://github.com/zigpy/zha-device-handlers/issues/1768)) sends 8 bytes of data but its type is missing in the __int__() function.
I guess it must be like `ints = {1: t.int8s, 2: t.int16s, 3: t.int24s, 4: t.int32s, 8: t.int64s}` at least and maybe with other types as well.

javicalle, 2023-03-03:
@SanchYESS could you attach a log from that kind of messages, please?
Just enable the debug logs and catch a couple of them (and its meanings if possible).

Thanks in advanced.

SanchYESS, 2023-03-03:
@javicalle 
<details><summary>Part of log</summary>

```
2023-03-03 12:02:12.338 DEBUG (MainThread) [zigpy_znp.api] Received command: ZDO.SrcRtgInd.Callback(DstAddr=0x6B83, Relays=[])
2023-03-03 12:02:12.349 DEBUG (MainThread) [zigpy_znp.api] Received command: AF.IncomingMsg.Callback(GroupId=0x0000, ClusterId=61184, SrcAddr=0x6B83, SrcEndpoint=1, DstEndpoint=1, WasBroadcast=<Bool.false: 0>, LQI=168, SecurityUse=<Bool.false: 0>, TimeStamp=7276745, TSN=0, Data=b'\x09\x71\x02\x07\x74\x06\x00\x00\x08\x08\x4E\x00\x00\x97\x00\x00\x1B', MacSrcAddr=0x6B83, MsgResultRadius=29)
2023-03-03 12:02:12.360 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x6B83), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=61184, data=Serialized[b'\tq\x02\x07t\x06\x00\x00\x08\x08N\x00\x00\x97\x00\x00\x1b'], tx_options=<TransmitOptions.NONE: 0>, radius=29, non_member_radius=0, lqi=168, rssi=None)
2023-03-03 12:02:12.362 DEBUG (MainThread) [zigpy.zcl] [0x6B83:1:0xef00] Received ZCL frame: b'\tq\x02\x07t\x06\x00\x00\x08\x08N\x00\x00\x97\x00\x00\x1b'
2023-03-03 12:02:12.366 DEBUG (MainThread) [zigpy.zcl] [0x6B83:1:0xef00] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.CLUSTER_COMMAND: 1>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=0, reserved=0, *is_cluster=True, *is_general=False, *is_reply=True), tsn=113, command_id=2, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-03 12:02:12.372 DEBUG (MainThread) [zigpy.zcl] [0x6B83:1:0xef00] Decoded ZCL frame: TuyaManufClusterDinPower:set_data_response(param=Command(status=7, tsn=116, command_id=6, function=0, data=[8, 8, 78, 0, 0, 151, 0, 0, 27]))
2023-03-03 12:02:12.375 DEBUG (MainThread) [zigpy.zcl] [0x6B83:1:0xef00] Received command 0x02 (TSN 113): set_data_response(param=Command(status=7, tsn=116, command_id=6, function=0, data=[8, 8, 78, 0, 0, 151, 0, 0, 27]))
2023-03-03 12:02:12.377 DEBUG (MainThread) [zhaquirks.tuya] [0x6b83:1:0xef00] Received value [8, 78, 0, 0, 151, 0, 0, 27] for attribute 0x0006 (command 0x0002)
2023-03-03 12:02:12.378 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/usr/local/lib/python3.10/site-packages/zigpy_znp/zigbee/application.py", line 592, in on_af_message
self.packet_received(
File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 875, in packet_received
self.handle_message(
File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 381, in handle_message
return sender.handle_message(
File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
self.endpoints[src_ep].handle_message(
File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
handler(hdr, args, dst_addressing=dst_addressing)
File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 485, in handle_cluster_request
zvalue = ztype(tuya_data)
File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 93, in __new__
n = super().__new__(cls, *args, **kwargs)
File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 253, in __int__
return ints[self[0]].deserialize(bytes(reversed(self[1:])))[0]
KeyError: 8
```

</details>

Seems the meter reports 3 values inside TUYA_POWER_ATTR, I get them like that:

<details><summary>ts0601_din_power.py (part)</summary>

```python
TUYA_POWER_ATTR = 0x0006        #0x0213
#...
class TuyaManufClusterDinPower(TuyaManufClusterAttributes):
    """Manufacturer Specific Cluster of the Tuya Power Meter device."""

    attributes = {
        #...
        TUYA_POWER_ATTR: ("power", t.uint64_t, True),
        #...
    }

    def _update_attribute(self, attrid, value):
        #...
        elif attrid == TUYA_POWER_ATTR:
            self.endpoint.electrical_measurement.voltage_reported((value >> 48) & 0xffff)
            self.endpoint.electrical_measurement.current_reported((value >> 24) & 0xffffff)
            self.endpoint.electrical_measurement.power_reported(value & 0xffffff)
        #...
```

</details>

So, voltage in `[8, 78, 0, 0, 151, 0, 0, 27]` is 8*256 + 78 = 2126 (212,6V), current  is 151mA and power is 27W.

Rarely I get some values in TUYA_VOLTAGE_ATTR (0x0004) like 0x01, 0x02 or 0x13, don't know what it supposed to mean.
TUYA_TOTAL_ENERGY_ATTR (0x0201) works fine before and after the update.
Did not get any TUYA_CURRENT_ATTR (0x0011) messages.

PR #2244 (2023-03-03): Enable `ints` up to 8 bytes
Update the `Data` `__int__` to enable up to 8 bytes

Fixes: #2243
closed 2023-03-03 by TheJulianJES