zigpy/zha-device-handlers #2247: [Device Support Request] Aqara E1 Leak Sensor (lumi.flood.acn001)

jjross, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
Aqara E1 Leak Sensor (lumi.flood.acn001) does not present the battery level.

**Describe the solution you'd like**
I would like to see the battery level with this device so I can ensure that it is still working.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.flood.acn001",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

TheJulianJES, 2023-03-05:
1. The water leak sensor entity itself works correctly?
2. Can you check what kind of battery the E1 sensor takes? (guessing `CR2032`)
3. (Where did you find these sensors? 😄)

jjross, 2023-03-06:
1. Yes, the leak sensor works totally fine but without the battery I don't trust it.  :-)
2. Yes, CR2032
3. I bought it on Amazon Canada here:

https://www.amazon.ca/gp/product/B07D39MSZS/

The odd thing is that the senor is mentioned on the Blakadder site and it states that the senor works with ZHA and exposes the battery and the leak sensor but I don't see the battery:

https://zigbee.blakadder.com/Aqara_SJCGQ13LM.html

TheJulianJES, 2023-03-07:
Download this file: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/tjj/aqara_e1_water_leak_battery/zhaquirks/xiaomi/aqara/water_acn001.py
and install it as a custom quirk (see this tutorial): https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274
If you need more help, let me know.

If the battery entity doesn't show up, you might need to remove and re-pair the device. It can then still take some time until the battery measurement shows up (the entity should be created though). You could try pressing (the button) on it once.
If this works for you, let me know and it'll make it into the 2023.4.0 HA release. (You should remove the custom quirk by then --> just delete the file. You can keep the entry in `configuration.yaml` though (for future testing)).

--- 

Also very interesting that you got that from Amazon. As far as I know, these are China-only devices (but consider yourself lucky, because they're Zigbee 3.0). Just curious, is the box in Chinese writing only?

closed 2023-03-09 by TheJulianJES

TheJulianJES, 2023-03-14:
You should be able to remove the custom quirk when updating to HA Core 2022.3.4 or later (which will be available for download in the next hour or so).

jjross, 2023-03-14:
Yup!  Seems to work perfectly.  I see this quirk applied:

zhaquirks.xiaomi.aqara.water_acn001.WaterE1

and the values are being decoded properly