zigpy/zha-device-handlers #2248: [Device Support Request] smoke detector TS0601 _TZE200_ntcy3xu1 not working but the quirk gets recognized

pbvdven, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
My smoke detector wont show any useful entities. the only one that it show is iaszone and its always clear
i looked in the quirk zhaquirks.tuya.ts0601_smoke.TuyaSmokeDetector0601 but my devices seems already added but not working as expected.

[image]

**Describe the solution you'd like**
working entities.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0500",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ntcy3xu1",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_smoke.TuyaSmokeDetector0601"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.3.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "5.10.0-21-amd64",
    "supervisor": "2023.01.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "23.0.1",
    "chassis": "desktop",
    "run_as_root": true
  },
  "custom_components": {
    "helium": {
      "version": "0.3.8",
[... 386 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

digistep, 2023-03-04:
Same situation for the TS0601 / _TZE200_m9skfctm

0anton, 2023-03-09:
Same here for  `zhaquirks.tuya.ts0601_smoke` - device paired, but only one `laszone` sensor added, which is dysfunctional. Device does not communicate with the controller in any form. Re-pairing does not help.

```
IEEE: xxx
Nwk: xxx
Device Type: EndDevice
LQI: 135
RSSI: Unknown
Last Seen: 2023-03-09T18:01:26
Power Source: Battery or Unknown
Quirk: zhaquirks.tuya.ts0601_smoke.TuyaSmokeDetector0601
```

[image]

no single data point at any sensor

0anton, 2023-04-03:
I've found the same device working fine with _Z2M_
[image]

The device reports in Z2M (in contrast to ZHA) sensors: battery, alarm, tamper.

Translation code in Z2M apparently provided by:

- https://github.com/Koenkk/zigbee2mqtt/issues/11119
- https://github.com/Koenkk/zigbee2mqtt/issues/12769

Is it possible to port it to ZHA?

b2un0, 2023-04-24:
same for me. have 4 of them.

b2un0, 2023-04-24:
same for me.
have 4 of theme.
here is a pairing log:

[home-assistant_zha_2023-04-24T12-37-16.094Z.log](https://github.com/zigpy/zha-device-handlers/files/11310495/home-assistant_zha_2023-04-24T12-37-16.094Z.log)

<details>
<summary>diagnostic</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.4.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": true
  },
  "custom_components": {
    "zha_toolkit": {
      "version": "v0.8.38",
      "requirements": [
        "packaging>=20.8",
        "pytz"
      ]
    },
    "xiaomi_miio_fan": {
      "version": "2022.8.0.0",
      "requirements": [
        "construct==2.10.56",
        "python-miio>=0.5.12"
      ]
    },
    "localtuya": {
      "version": "5.0.0",
      "requirements": []
    },
    "avanza_stock": {
      "version": "1.2.0",
      "requirements": [
        "pyavanza==0.6.0"
      ]
    },
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "auto_backup": {
      "version": "1.3.2",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
[... 230 lines cut ...]
          },
          "0xef00": {
            "endpoint_attribute": "tuya_manufacturer",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x000a": {
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```
</details>

git-peterk, 2023-09-10:
You can give a try to this custom quirk, hope that works for you too:
https://github.com/zigpy/zha-device-handlers/issues/1838#issuecomment-1712805018

ViperRNMC, 2023-10-04:
> You can give a try to this custom quirk, hope that works for you too: [#1838 (comment)](https://github.com/zigpy/zha-device-handlers/issues/1838#issuecomment-1712805018)

Thanks, only need to change model info to TZE200 in the py file. Can now see tamper and battery.

yoshz, 2023-12-30:
Same problem with `_TZE200_m9skfctm`. The smoke_detected attribute is not logged when I press the test button, nor is the "clear" logged.

@git-peterk That quirk just adds tampering and battery state and doesn't change anything to the smoke_detected attribute.

My guess is that these smoke detectors are just a peace of Chinese crap.

hellebauer, 2024-02-28:
_TZE200_aycxwiau   (Woox Smoke Detector R7049)

No luck with above model.  Pairing works, but nothing works beyond that.

TGMivo, 2024-03-21:
I have the TZE200_rccxox8p also no luck
Pairing works, but see only LQI and RSSI

closed 2024-09-24 by github-actions[bot]