zigpy/zha-device-handlers #2249: [Device Support Request] ADEO LDSENK08 contact and vibration detector

dranih, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
ADEO (Lexman) LDSENK08 device is a contact sensor with a vibration sensor.
The device only shows a IasZone sensor in ZHA with the contact and vibration events mixed. You cannot know if it is a vibration or an open contact event.
The type of event can be found by reading the zone_status property :
- if ZONE_STATUS first bit is set, it's an open contact
- if ZONE_STATUS second bit is set, it's a vibration event

Integration has been done in zigbee2mqtt : https://www.zigbee2mqtt.io/devices/LDSENK08.html

PS : the device also gives tamper and low battery details on bits 3 and 4 but I guess this will be addressed here : https://github.com/home-assistant/core/issues/72828

**Describe the solution you'd like**
I would like to separate contact and vibration sensors in ZHA.
I've made a quirk which works by adding 2 endpoints for the contact and vibration sensors and using a bus to pass the events depending on the Iaszone zone_status bits.
It's my first quirk and I am not sure it's done the right way. I would like your opinion before opening a PR.

```yaml
"""Device handler for ADEO Lexman LDSENK08 smart door window sensor with vibration"""
import logging

from typing import Any, List, Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice, CustomCluster
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Ota,
    PollControl,
    PowerConfiguration,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl import foundation

from zhaquirks import Bus, LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_STATUS,
    ZONE_TYPE,
    ON,
    OFF,
)

_LOGGER = logging.getLogger(__name__)

class IasMultiZoneCluster(CustomCluster, IasZone):
    """Multi Zone Cluster."""
    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Standard_CIE}

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: List[Any],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ):
        """Handle a cluster command received on this cluster."""
        if hdr.command_id == 0:
            #If we have bit 2, it's a vibration
            if (args[0] & 2) > 0:
                self.endpoint.device.ias_bus.listener_event("vibration_event", ON)
[... 300 lines cut ...]
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**

TheJulianJES, 2023-03-09:
Doesn't look bad! Nice job.

The docs are a bit outdated though, you might be able to remove all `Bus` stuff and update the attributes on another cluster directly. Normally it's done like this: `self.endpoint.ias_zone.update_attribute(ZONE_STATUS, ON)`
However, since you have multiple `ias_zone` clusters and they're on different endpoints, that probably won't work.

I'll need to have a look later, but maybe you already want to experiment a bit.
Perhaps you could try something `self.endpoint.device.endpoints[2].ias_zone.update_attribute(ZONE_STATUS, ON)` and see if that works?

`VibrationCluster` and `ContactCluster` could have their ``__init__`` and ``x_event`` functions removed then, so they would just contain `_CONSTANT_ATTRIBUTES`.

dranih, 2023-03-10:
Thanks for the feedback !

Your solution is definitively working :clap: , the code is a lot cleaner :
```python
"""Device handler for ADEO Lexman LDSENK08 smart door window sensor with vibration"""
from typing import Any, List, Optional, Union

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice, CustomCluster
import zigpy.types as t
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Ota,
    PollControl,
    PowerConfiguration,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl import foundation

from zhaquirks import LocalDataCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    ZONE_STATUS,
    ZONE_TYPE,
    ON,
    OFF,
)

class IasMultiZoneCluster(CustomCluster, IasZone):
    """Multi Zone Cluster."""
    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Standard_CIE}

    def handle_cluster_request(
        self,
        hdr: foundation.ZCLHeader,
        args: List[Any],
        *,
        dst_addressing: Optional[
            Union[t.Addressing.Group, t.Addressing.IEEE, t.Addressing.NWK]
        ] = None,
    ):
        """Handle a cluster command received on this cluster."""
        if hdr.command_id == 0:
            #If we have bit 2, it's a vibration
            if (args[0] & 2) > 0:
                self.endpoint.device.endpoints[2].ias_zone.update_attribute(ZONE_STATUS, ON)
            else:
                self.endpoint.device.endpoints[2].ias_zone.update_attribute(ZONE_STATUS, OFF)
            #If we have bit 1, it's open
            if (args[0] & 1) > 0:
                self.endpoint.device.endpoints[3].ias_zone.update_attribute(ZONE_STATUS, ON)
            else:
                self.endpoint.device.endpoints[3].ias_zone.update_attribute(ZONE_STATUS, OFF)
        super().handle_cluster_request(hdr, args, dst_addressing=dst_addressing)

class VibrationCluster(LocalDataCluster, IasZone):
    """Vibration Sensor."""
    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Vibration_Movement_Sensor}

class ContactCluster(LocalDataCluster, IasZone):
    """Open/close Sensor."""
    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Contact_Switch}

class AdeoLdsenk08(CustomDevice):
    """Adeo contact and vibration sensor controller."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=1026
        #  device_version=1
        #  input_clusters=[0, 1, 3, 32, 1280, 2821]
        #  output_clusters=[25]>
        MODELS_INFO: [("ADEO", "LDSENK08")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,  # 260
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,  # 2048
                INPUT_CLUSTERS: [
                    Basic.cluster_id,  # 0
                    PowerConfiguration.cluster_id,  # 1
                    Identify.cluster_id,  # 3
                    PollControl.cluster_id,  # 32
                    IasZone.cluster_id,  # 1280
                    Diagnostic.cluster_id  # 2821
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id  # 25
                ],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,  # 0
                    PowerConfiguration.cluster_id,  # 1
                    Identify.cluster_id,  # 3
                    PollControl.cluster_id,  # 32
                    IasMultiZoneCluster,  # 1280
                    Diagnostic.cluster_id  # 2821
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id  # 25
                ],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    VibrationCluster,  # 1280
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    ContactCluster,  # 1280
                ],
                OUTPUT_CLUSTERS: [],
            },
        },
    }
```

I have 2 questions left before maybe opening a PR : 
- In the quirk, is it possible to change the default display names of the added Iaszones ? It may be confusing to have 3 sensors named "Iaszone" even if the icons indicate the zone type.
- The vibration sensitivity of the device can be adjusted. I have managed to do so by reading and writing the `current_zone_sensitivity_level` attribute while the device is active. Is it possible (and a good idea) to add a configuration item for this in the UI ? Also the device sends a `checkin` command on cluster_id 32 every hour or so ; my guess is that it is used to update the configuration of the device in the manufacturer solution (documentation - in french - https://enkiapp.zendesk.com/hc/fr/articles/360018121540-Utilisation-du-d%C3%A9tecteur-de-vibration-Lexman-avec-Enki)

Anyway, those 2 points would just be nice to have ; the main problem is fixed (distinguish vibration and open/close sensors).

TheJulianJES, 2023-03-10:
> is it possible to change the default display names of the added Iaszones ?

At the moment, I don't think it's possible to change the default sensor name. ZHA should probably generate it from the device class. See:
https://github.com/home-assistant/core/blob/96bd7143643a813689c86f6f7512008c187a4766/homeassistant/components/zha/binary_sensor.py#L159-L175

> add a configuration item for this in the UI ?

It would probably be possible to create a config entity for this, but it'll also only write if the device is awake. It also has to be added in ZHA itself at the moment (see select.py with some examples for motion sensitivity which write a different attribute).

From what I can see, the device actually follows spec pretty closely, but ZHA has no way to create different sensor for this one attribute at the moment (and it would generate it for all IAS devices). It's definitely something that should be supported better.

Also, I think you might be able to shorten `if (args[0] & 2) > 0:` to `if args[0] & 2`

dranih, 2023-03-11:
Thanks again for your explanations and another code improvement suggestion.
I need to look how to add some tests and after that I will open a PR referencing this issue.

TheJulianJES, 2023-09-17:
Reopening support request, as the PR is still open:
- https://github.com/zigpy/zha-device-handlers/pull/2298

closed 2024-03-22 by github-actions[bot]