zigpy/zha-device-handlers #2251: [Device Support Request] Aqara H1 EU Switch Double Rocker WS-EUK04

j0hann3s, 2023-03-03:
**Is your feature request related to a problem? Please describe.**
I can not capture any used energy measurements in my home assistant instance's history due to missing energy reporting. I also miss the feature to be able to get the device temperature, set the power outage memory setting, turn the indicator led on and off on the device, and set the operation mode to coupled/decoupled.

**Describe the solution you'd like**
An added attribute publishing the _sum of consumed energy_ (kWh) as well as an _instantaneous power measurement_. Moreover, a _device temperature_ report, being able to set the _power outage memory_, enable/disable the _indicator led_, and choosing the _operation mode._

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009",
        "0x0012",
        "0x0702",
        "0xfcc0"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0000",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": []
    },
    "21": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "31": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x000c"
      ],
      "out_clusters": []
    },
    "41": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0012"
      ],
[... 835 lines cut ...]
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Here is a screenshot of how it looks in ZHA right now. I am using the SkyConnect adapter.
[image]

Here is a screenshot of how the same switch looks inside Z2M and what attributes it exposes.
[image]

amoshydra, 2023-03-13:
I just figured out this yesterday.

Apparently the implementation is already done, see: https://github.com/zigpy/zha-device-handlers/blob/0.0.93/zhaquirks/xiaomi/aqara/opple_switch.py#L56-L61

# How to enable decoupled mode:

## Step 1 - Manage Zigbee Device

1. In your Device view, under Device info, click on the 3 dots menu
2. Manage zigbee device

[image]

## Step 2 - Choose `OppleSwitchCluster`

[image]

Inside the cluster dropdown menu, you will find:
- OppleSwitchCluster (Endpoint id: 1) - This correspond to button_1 (left side)
- OppleSwitchCluster (Endpoint id: 2) - This correspond to button_2 (right side)

Select OppleSwitchCluster and choose the attribute `operation_mode (id: 0x0200)`

[image]

You can read the current value by pressing "READ ATTRIBUTE"

[image]

The operation_mode will be one of:
- OppleOperationMode.Decoupled - `0`
- OppleOperationMode.Coupled - `1`

## Step 3 - Change Operation Mode

You can change the operation mode by writing a numeric value into the `value` like this:

[image]

[image]

----

You can repeat the same for the another button

timoei, 2023-03-25:
> I just figured out this yesterday.
> 
> Apparently the implementation is already done for `zhaquirks.xiaomi.aqara.opple_switch.XiaomiOpple2ButtonSwitchFace1`, see: https://github.com/zigpy/zha-device-handlers/blob/0.0.93/zhaquirks/xiaomi/aqara/opple_switch.py#L56-L61
> # How to enable decoupled mode:
> ## Step 1 - Manage Zigbee Device
> 
>     1. In your Device view, under Device info, click on the 3 dots menu
> 
>     2. Manage zigbee device
> 
> 
> [image]
> ## Step 2 - Choose `OppleSwitchCluster`
> 
> [image]
> 
> Inside the cluster dropdown menu, you will find:
> 
>     * OppleSwitchCluster (Endpoint id: 1) - This correspond to button_1 (left side)
> 
>     * OppleSwitchCluster (Endpoint id: 2) - This correspond to button_2 (right side)
> 
> 
> Select OppleSwitchCluster and choose the attribute `operation_mode (id: 0x0200)`
> 
> [image]
> 
> You can read the current value by pressing "READ ATTRIBUTE"
> 
> [image]
> 
> The operation_mode will be one of:
> 
>     * OppleOperationMode.Decoupled - `0`
> 
>     * OppleOperationMode.Coupled - `1`
> 
> 
> ## Step 3 - Change Operation Mode
> 
> You can change the operation mode by writing a numeric value into the `value` like this:
> 
> [image]
> 
> [image]
> 
> You can repeat the same for the another button

As I was looking for this option quite long: It is important to have the ZHA quirks enabled in the config.yaml!

tarosk, 2023-07-09:
It seems support for single rocker version is already merged in issue #1369 pull requeste #2385 

Can this be replicated to double rocker version as well?

pfolta, 2023-09-30:
@amoshydra, I managed to set my WS-EUK04 switch into decoupled mode using your instructions but I am not getting any `zha_event`s for when either button is pressed, see https://github.com/home-assistant/core/issues/101136

Any ideas?

danielshiloah, 2023-10-01:
I have the same problem of missing the attributes power, temp and configurations.
has there been any updates? cant find anything on the web but this device support request?

Jolanrensen, 2023-10-22:
Thanks @timoei!
I was looking for the decoupled mode :)
It looks like all supported features in zigbee2mqtt for this switch need to be enabled here too :)

salos1982, 2024-04-06:
After the latest update of Home Assistant 2024.4 decoupling does not work. After the update, decoupling for the already decoupled device was reset. I cannot decouple new device in the same way I did for previous devices

sattath, 2024-06-21:
Is there any way to control the LEDs in decoupled mode (as described in this comment: https://github.com/zigpy/zha-device-handlers/issues/2251#issuecomment-1483929748)?

ncandido, 2024-08-22:
> Is there any way to control the LEDs in decoupled mode (as described in this comment: [#2251 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2251#issuecomment-1483929748))?

Regarding the control of the LEDs, is possible to disable the LEDs in decoupled mode?

salos1982, 2024-08-22:
> > Is there any way to control the LEDs in decoupled mode (as described in this comment: [#2251 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2251#issuecomment-1483929748))?
> 
> Regarding the control of the LEDs, is it possible to disable the LEDs in decoupled mode?

Yes, you can use click and double-click events to turn on/off whatever you want. I use it to switch fans and ceiling lamps

teeten, 2024-09-07:
Using ZHA "manage zigbee device" as shown above, I was able to enable decoupled mode for both buttons of my Aqara H1 WS-EUK04 device. I can control the 2 switches (relays) from HA without problem. However the HA Logbook does not show any event when I press a button. 

HA Device info shows quirk is "zhaquirks.xiaomi.aqara.switch_h1_double.AqaraH1DoubleRockerSwitchWithNeutral4".

What should I check first? Quirk file, version? Sonoff Dongle Plus firmware? Most articles I have read are about Z2M but I only have ZHA at the moment.

closed 2025-08-20 by github-actions[bot]