zigpy/zha-device-handlers #2254: [Device Support Request]

andreasc1, 2023-03-06:
**Is your feature request related to a problem? Please describe.**
It seems that ZHA cannot discover these tuya zigbee wall switches: https://www.ebay.co.uk/itm/354407918982?var=623817126041

i have already preordered skyconnect to replace cc2531 but wondering if there's anything i can do until it arrives and if skyconnect will actually work with those switches.

**Describe the solution you'd like**
to be discovered by zha and be able to work.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
Paste the diagnostic information here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

MattWestb, 2023-03-06:
They is general sported, but tuya is making new version of them every week.
And not posting `Device signature` **and** `Diagnostic information` we cant helping you getting them working in ZHA.

javicalle, 2023-03-06:
If it is a battery powered device, you need to keep the device awake during all the pairing process. That can be done just short pressing any button every second or so until the pairing proccess ends.

closed 2023-09-12 by github-actions[bot]