zigpy/zha-device-handlers #2258: Support for Aqara wireless 2 gang switch (LUMI lumi.remote.b28ac1)

sophipl, 2023-03-07:
I just bought Aquara switch, however it was not recognized in Home Assistant with any buttons.
After each zha reload, the device is recognized as unknown device.
I noticed that there is already quirk for this device in `remote_h1.py`
But my device has signature with 3 empty ... rows?
So I edited the quirk, and it worked fine with buttons, until I reloaded zha, and device broke again.

<details>
<summary>Debug before quirk</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.1.6",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.7",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Warsaw",
    "os_name": "Linux",
    "os_version": "5.15.84-v8+",
    "run_as_root": true
  },
  "custom_components": {},
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "config_flow": true,
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "requirements": [
      "bellows==0.34.6",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.90",
      "zigpy-deconz==0.19.2",
      "zigpy==0.53.0",
      "zigpy-xbee==0.16.2",
      "zigpy-zigate==0.10.3",
      "zigpy-znp==0.9.2"
    ],
    "usb": [
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*2652*",
        "known_devices": [
          "slae.sh cc2652rb stick"
        ]
      },
      {
        "vid": "1A86",
        "pid": "55D4",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus v2"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
        "description": "*sonoff*plus*",
        "known_devices": [
          "sonoff zigbee dongle plus"
        ]
      },
      {
        "vid": "10C4",
        "pid": "EA60",
[... 1069 lines cut ...]
            },
            3: {
                # "3": {
                #   "profile_id": 260, "device_type": "0x0103",
                #   "in_clusters": [ "0x0003" ],
                #   "out_clusters": [ "0x0006" ] }
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT_SWITCH,
                INPUT_CLUSTERS: [Identify.cluster_id],
                OUTPUT_CLUSTERS: [OnOff.cluster_id],
            },
            4: {
            },
            5: {
            },
            6: {
            },
        },
    }
    replacement = RemoteH1DoubleRocker1.replacement
    device_automation_triggers = RemoteH1DoubleRocker1.device_automation_triggers

```

</details>

javicalle, 2023-03-07:
That signature seems to be incomplete. 
If it is a battery powered device, you need to keep the device awake during all the pairing process. That can be done just short pressing any button every second or so until the pairing proccess ends.
Can you try to pair the device again?

TheJulianJES, 2023-03-07:
Yep, and what coordinator are you running? (And with what firmware?)

Also, there were already two new major HA revisions since 2022.1.6.

sophipl, 2023-03-07:
Yes, It's battery powered, CR2450 to be exact, made some photos of the device.
I was trying to re-pair the device, and was clicking furiously, and it seem to work now, it was paired with different quirk.
I'm running ZBDongle-E with original firmware, I don't know which version and don't know how to find out.
My HA is a docker container 2023.1.6
But since it's working, at least now, thank You @javicalle and @TheJulianJES , never before I had a device signature be incomplete. It seems that device is already supported in quirk. 

```json
                "org.opencontainers.image.authors": "The Home Assistant Authors",
                "org.opencontainers.image.created": "2023-01-20 01:09:16+00:00"
                "org.opencontainers.image.version": "2023.1.6"
```

[image]
[image]
[image]

closed 2023-03-07 by sophipl

sofianegargouri, 2023-05-18:
Hi ! 
Is your switch working well with double / triple / long presses ? Mine has the right quirk setup but only the single press seems to be triggered, and the double press is flagged as 2x single presses

sophipl, 2023-05-18:
Hi
I'm not currently using the device, but I have the same quirk as you pointed out.
I didn't have any problems, but didn't check every function.
If You have problems, I have the ability to wireshark the protocol and fix the quirk.