zigpy/zha-device-handlers #2259: [BUG] Tuya Outlet _TZ3000_cehuw1lw goes offline within 1min of pairing

davet2001, 2023-03-07:
**Describe the bug**
The zigbee socket does not pair reliably, and goes unavailable within 1min of pairing.  The problem is identical on two devices, both of which work fine on zigbee2mqtt.

**To Reproduce**
Steps to reproduce the behavior:
1. Power on a Tuya UK power monitoring socket, identifying as _TZ3000_cehuw1lw
2. Click 'add devices' in Home assistant.
3. Long press the button on the side of the _TZ3000_cehuw1lw to enter pairing mode
4. Wait for the device to pair.

**Expected behavior**
The device should join the network, the blue LED on the side should stop flashing, and all parameters such as on/off, voltage, current, power, cumulative energy should be reliably visible.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0702",
        "0x0b04",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_cehuw1lw",
  "model": "TS011F",
  "class": "zhaquirks.tuya.ts011f_plug.Plug"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/London",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
[... 485 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
The device works ok on zigbee2mqtt.  It is observed that the blue led stops flashing at the end of pairing.  With ZHA the blue led keeps flashing.

davet2001, 2023-03-07:
The original seller seems to have stopped selling this, but the following appears to be an identical device:
https://www.aliexpress.com/item/1005003956775658.html

thambl, 2023-03-20:
I’m experiencing identical issues with Tuya device _TZ3000_gjnozsaz TS011F.

davet2001, 2023-04-15:
My two devices that have this problem have the following IEEE addresses:
```
0xa4c13811********
0xa4c1382c********
```

MattWestb, 2023-04-15:
Both is Telink Semiconductor (Taipei) Co. Ltd. 
Its little new and not so well tested and the old Silabs chips that was used in all devices before.
We expect son bugs but i have 2 devices with it and looks working OK.

MaxMSchneider, 2023-04-20:
Also coming in here for _TZ3000_gvn91tmx. Picked up three of them, showing the same behavior like described by @davet2001 

IEEE: a4:c1:38:88:7f:e8:de:45

MattWestb, 2023-04-20:
They is having Telink chips.

MaxMSchneider, 2023-04-20:
@MattWestb meaning? Anything i can do?

MattWestb, 2023-04-20:
We dont knowing way this devices cant paring but the common looks being the chip used in them.
We need one sniffing of the paring for see what is going wrong / what the device dont like with our systems.

davet2001, 2023-05-17:
I don't understand what the difference is here but here's what I find:
I will define 'working' as 'pairs via the UI, remains connected, and flashing light stops after pairing'.

- Ubuntu docker dev branch 10/5/2023- cc2531 - working
- Raspberry pi HA OS 2023.5.2 zigbee2mqtt - zzh stick - working
- Raspberry pi HA OS 2023.5.2 zha - skyconnect - not working (flashing light remains, then device becomes unavailable after ~1min)

So this device definitely can work, it just doesn't fully yet.

shanelord01, 2023-08-02:
Hi team. Looks like I have the same issue witha "TS011F by _TZ3000_dd8wwzcy" as per here:
https://github.com/home-assistant/core/issues/97122

uffevedenbrant, 2023-08-06:
3 different setups..  All latest HA and with ZHA

A little older ( latest HA / RPI4 / Skyconnect )  to which i tried to add Nous plug ( seems to be the same, TZ3000_2putqrmw TS011F ) , i have messed around with this setup starting with a Aeotec Zi-stick , migrated to Skyconnect, most other zigbee stuff works except the Nous plugs.
Nous plugs ends with flashing pairing LED , on/off works for about a minute

HA-test #1 
Total fresh setup with RPI3 , added the Aeotec Zi-stick and the 4 Nous plugs.. Pairing was perfect
Added some Sunricher 2CH switches , still working
Nous plugs ends with steady pairing LED , on/off works for still after 24 hours
Scrapped for HA-test #2000 

HA-test #2
Total fresh setup with RPI4 , added the Aeotec Zi-stick and the 4 Nous plugs.. Pairing was perfect
Nous plugs ends with steady pairing LED , on/off works for still after 30 mins
Added 2  Sunricher 2CH switches , still working

I would then assume that something have happened with the ZHA config that makes pairing to fail.
I.e. a fresh installation works nice , a older installation may fail to pair them

Hope that the above could give a hint on whats going on..

Ill try now to add one of the Nous plugs to the initial setup to see if the lates HA updates made any diff..

uffevedenbrant, 2023-08-06:
Update.. Tried to add the Nous plugs to the older HA setup ( see previous post ) still a fail ( i.e. pairing fails )
Re-paired with the new HA from an hour ago and works as expected. ( i.e. pairing works and stable connectivity )
Old is a then with a Skyconnect zigbee dongle and new is with Aeotec Zi-stick , same chip as i understand for Zigbee in them.

frias0, 2023-08-06:
Using Conbee 1 and ZHA, 
I had the same problem with these devices for months. However, after I got home after a power outage, the devices were paired and working.
```

TS110E by _TZ3210_k1msuvg6 (dimmer 1 gang)
IEEE: a4:c1:38:4e:fa:d3:e8:6a
Quirk: ts110e_custom.DimmerSwitchWithNeutral1Gang

TS0001 by _TZ3000_tqlv4ug4 (switch 1 gang) 
IEEE: a4:c1:38:f4:10:e4:a4:f9
Quirk: zhaquirks.tuya.ts000x.Switch_1G_Metering

TS0001 by _TZ3000_tqlv4ug4 (switch 1 gang)
IEEE: a4:c1:38:0f:c8:88:50:62
Quirk: zhaquirks.tuya.ts000x.Switch_1G_Metering

```
I was waiting to add a dimmer 2 gang, after wiring it up, it is now having the same problem.

```
TS110E by _TZ3210_3mpwqzuu  (dimmer 2 gang)
IEEE: a4:c1:38:d1:6d:7f:6e:c9
Quirk: ts110e_custom.DimmerSwitchWithNeutral2Gang
```

Maybe the power outage did something, that can be reproduced?

MattWestb, 2023-08-06:
Its one common thing with all devices they is having Telink Semiconductor (Taipei) Co. Ltd chips and the original was having Silabs chips.

frias0, 2023-08-06:
> Its one common thing with all devices they is having Telink Semiconductor (Taipei) Co. Ltd chips and the original was having Silabs chips.

Ok, that what I thought, and why I included the ieee. I tried to read the z2m source, to find if they do something special during interview, but I could never get my head around the code.

Nor did I see how a quirk can change the pairing/interview, when I made some modifications to the attributes for my devices.

phil-lipp, 2023-09-28:
Are there any updates on this issue? Over at https://github.com/zigpy/zha-device-handlers/issues/2573 there was a "easy fix" for handling another tuya plug that looks identical to the others, so basically the same situation as in this issue here.

@MattWestb  I'd be glad to help where I can, but I'm new to creating quirks or contributing to zha in general. Where should I start to look for the errors that cause the disconnect? Just watch the Homeassistant debug log when I'm trying to pair the plug?

davet2001, 2023-09-28:
Ok, this one is still a problem for me, the behaviour that @uffevedenbrant described exactly matches my scenario. A completely brand new config works fine, but an existing config that has had problems, never works, no matter how much I try to delete and re-add the device. It's as though there's some bad data or memory in the config that's tripping it up.

I wonder about some way to completely irradiate all memory of a particular MAC address from both the stick eeprom and the HA config.  That seems to be needed sometimes, is it available somehow?

davet2001, 2023-09-28:
I think trying with 2023.10 is a good idea since some related quirks have been added to HA.

phil-lipp, 2023-10-19:
> @davet2001 Had the same issue, updating the skyconnect firmware did the trick. See here: [home-assistant/core#98624 (comment)](https://github.com/home-assistant/core/issues/98624#issuecomment-1688222346)

Fixed it for me too - thanks for the hint! :)

thambl, 2023-10-19:
I can confirm, updating the silicon labs zigbee chip to the latest firmware on my yellow, fixed my pairing issues.

MattWestb, 2023-10-19:
Witch is the bad version ?
I like updating the wiki with recommended version and RCP we is always running  the latest version then its running in the addon and cant running on lower ones.

phil-lipp, 2023-10-20:
> Witch is the bad version ? I like updating the wiki with recommended version and RCP we is always running the latest version then its running in the addon and cant running on lower ones.

I upgraded the SkyConnect from the firmware it shipped with (7.1.1.0 build 273) to the latest available beta (7.3.1.0)

thambl, 2023-10-20:
I’m not sure which version shipped with my yellow, but did the upgrade to latest v7.3.1.0

davet2001, 2023-12-01:
This issue is no longer causing a problem for me.

I recently upgraded the sky connect firmware although I think the problem stopped before then.

Please reopen if you experience an issue.

closed 2023-12-01 by davet2001