zigpy/zha-device-handlers #2260: [BUG] BTicino N4411C (Legend dimmer)

lucagessi, 2023-03-07:
Hi. I use a BTicino dimmer N4411C which works with Legrand app. It has a custom manufacturer cluster for enabling dimmer functionality.
It is shown in HA, it's custom cluster is present but I cannot write its attributes. I would like to enable the dimmer but I couldn't do it. 
The image attached shows that there aren't attributes on the list.
If I try to do the reconfiguration from ZHA it fails somewhere.
[image]

Below the signature.

`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=1, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4129, maximum_buffer_size=89, maximum_incoming_transfer_size=63, server_mask=10752, maximum_outgoing_transfer_size=63, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000f",
        "0x0301",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0000",
        "0x0005",
        "0x0006",
        "0x0019",
        "0xfc01"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": " Legrand",
  "model": " Dimmer switch with neutral",
  "class": "zigpy.device.Device"
}`

TheJulianJES, 2023-03-07:
Yes, you'll need to add a list of the manufacturer attributes to a custom quirk.
Here's a generated basic quirk structure for the signature you provided: https://dpaste.org/M1NSD/raw

You'll then want to create a custom cluster similar to this: https://github.com/zigpy/zha-device-handlers/blob/9c1bc72ecfbba6aa947cca553777d75c9bfd7574/zhaquirks/xiaomi/aqara/motion_ac02.py#L38-L46
(Use `CustomCluster` instead of `XiaomiAqaraE1Cluster`, leave out the methods, just put your manufacturer attributes in there, and then replace `0xfc01` in the `replacement` part of the signature with the class name of custom cluster)
You can then test the quirk as a custom quirk: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

"Edit:"
Although a possible match for your signature is also located at: https://github.com/zigpy/zha-device-handlers/blob/9c1bc72ecfbba6aa947cca553777d75c9bfd7574/zhaquirks/legrand/dimmer.py#L223
It seems like the manufacturer attributes are already implemented there.

I've made some modifications, so download this file: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/tjj/legrand_dimmer/zhaquirks/legrand/dimmer.py
and install it as a custom quirk, then report back if it works. (If it works, we'll merge it for the 2023.4.0 release and you should remove your custom quirk when you install that.)

lucagessi, 2023-04-30:
Hi. Than you for your time. 
But how can I add a custom quick in home assistant installation? There is a folder to edit? Thank you

lucagessi, 2023-05-01:
I did other tests.
I think using the quirk I can write into manufacturer cluster. I have the problem I wrote in the last post so I try to write manually using zha_toolkit. Also there there is a bug in util.py, changing some line of code I succeeded in enabling/disabling dimmer feature.
The problem is that the description of all clusters in not correct I think. I see the light fade in and out during powering but there is some wrong parameter to change but I don't know where. 
For example cluster 8 is for level control. Attribute 0 reports current level used by the dimmer. I should use level_on (attr 0x11) in order to change the actual dimmer level but nothing change. So I think that there are some other params, like max_level/min_level which are not correctly mapped. Maybe actually level_on is a min/max level, I don't know.

closed 2023-11-04 by github-actions[bot]

jafar-atili, 2024-09-17:
@TheJulianJES, I tried your custom quirks and now the dimmer device do expose the `LegrandCluster` with the 

[image]

I wonder if you can contribute this quirk and help us achieve the same for the K4003C (regualr switch) and for the K4027C (Shutter Siwtch) as Z2M exposes this config for both devices.