zigpy/zha-device-handlers #2264: [BUG] TS0501 Fan Controller changes behaviour once paired

Brazen00, 2023-03-09:
**Describe the bug**
I have installed 7x Mercator Ikuu TS0501 Fan Controllers in my home. Before they are paired, the physical "fan speed" button on the controller will cycle the fan between low, medium and high speeds. Once paired to ZHA, the physical "fan speed" button on the controller will only cycle between low and high speeds, skipping the medium speed. If I unpair the controller, the default behaviour returns. On Home Assistant itself, once paired I can change the fan speed between high, medium and low. The incorrect behaviour only occurs on the physical button.

The fix is likely to be to set ```fan_mode_sequence``` to ```FanModeSequence.Low_Med_High``` in a quirk.

This was previously raised [here](https://github.com/zigpy/zha-device-handlers/issues/1577), and was fixed in Z2M [here](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5454).

**Expected behavior**
I expect the physical button to cycle through the 3 speeds.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0202",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_lzqq3u4r",
  "model": "TS0501",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Brisbane",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
[... 633 lines cut ...]
            "endpoint_attribute": "time",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

Brazen00, 2023-03-09:
Yes, performing those steps fixes the issue.

TheJulianJES, 2023-03-09:
Can you try this custom quirk: https://raw.githubusercontent.com/TheJulianJES/zha-device-handlers/tjj/tuya_fan_override_modes_on_bind/zhaquirks/tuya/ts0501_fan_switch.py

Instructions for installing: https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

When re-pairing (or using the "reconfigure" option in the UI), this should write the attribute appropriately now. Can you please check that it works, so we can merge it into zha-quirks?

Brazen00, 2023-03-09:
Thanks. Unfortunately, that has not fixed the issue. It's still reporting as ```Low_High```

TheJulianJES, 2023-03-09:
Did you re-configure the device through the UI (or re-pair it completely) with the quirk in place?
Is the quirk loaded? (Can be seen at the bottom of the device signature)

Brazen00, 2023-03-09:
Yes, I re-configured, which didn't work, so I re-paired it completely.

It is showing as ```Quirk: ts0501_fan_switch.TS0501FanController```

TheJulianJES, 2023-03-09:
Ah, duh. Can you re-download the file and try again? (And make sure it says `FanCluster` in the `replacement` part -- should appear two times in that file)

Brazen00, 2023-03-09:
Ok, that has fixed it!

closed 2023-03-09 by TheJulianJES

TheJulianJES, 2023-03-14:
@Brazen00 You should be able to remove the custom quirk when updating to HA Core 2022.3.4 or later (which will be available for download in the next hour or so).

dracon80, 2023-10-27:
@Brazen00 sorry to dredge up an old bug that has been closed. But I've just had 4 of these installed. While the integration with home-assistant works flawlessly, the controllers themselves are very disappointing at present. If I use the physical touch buttons to control the fan speed or lights the whole controller will often flash white and do what looks like a reset. This turns the fan and the light off and fails to update home-assistant of the change in state. 
Did you have this problem at all? I don't believe its a ZHA problem, as the controllers do it even when not paired with anything.