zigpy/zha-device-handlers #2266: [Device Support Request] Aqara H1 Knob - lumi.remote.rkba01 - ZNXNKG02LM

oxc, 2023-03-10:
This has been reported as https://github.com/zigpy/zha-device-handlers/issues/929 which has been closed as stale. It's still relevant, so I'll try to combine my own information with those from that bug report.

**Is your feature request related to a problem? Please describe.**
When pairing the device to my Home Assistant ZHA integration using ConBee II, it only exposes LQI and RSSI sensors.

**Describe the solution you'd like**
Support for: Battery, Dimmer, Press, Double Press, Press and Hold, Press and turn.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0103",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003"
      ],
      "out_clusters": [
        "0x0003",
        "0x0006",
        "0x0019"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.remote.rkba01",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.3.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.76-v8+",
    "supervisor": "2023.03.1",
    "host_os": "Debian GNU/Linux 11 (bullseye)",
    "docker_version": "20.10.21",
    "chassis": "",
    "run_as_root": true
  },
  "custom_components": {
    "google_home": {
      "version": "1.9.17",
      "requirements": [
        "glocaltokens==0.7.0"
      ]
    },
[... 337 lines cut ...]
    {
        zigbeeModel: ['lumi.remote.rkba01'],
        model: 'ZNXNKG02LM',
        vendor: 'Xiaomi',
        description: 'Aqara knob H1 (wireless)',
        meta: {battery: {voltageToPercentage: '3V_2850_3000'}},
        exposes: [e.battery(), e.battery_voltage(),
            e.action(['single', 'double', 'hold', 'release', 'start_rotating', 'rotation', 'stop_rotating']),
            exposes.enum('operation_mode', ea.ALL, ['event', 'command']).withDescription('Button mode'),
            exposes.numeric('action_rotation_angle', ea.STATE).withUnit('*').withDescription('Rotation angle'),
            exposes.numeric('action_rotation_angle_speed', ea.STATE).withUnit('*').withDescription('Rotation angle speed'),
            exposes.numeric('action_rotation_percent', ea.STATE).withUnit('%').withDescription('Rotation percent'),
            exposes.numeric('action_rotation_percent_speed', ea.STATE).withUnit('%').withDescription('Rotation percent speed'),
            exposes.numeric('action_rotation_time', ea.STATE).withUnit('ms').withDescription('Rotation time'),
        ],
        fromZigbee: [fz.xiaomi_on_off_action, fz.xiaomi_multistate_action, fz.xiaomi_basic, fz.aqara_opple, fz.aqara_knob_rotation],
        toZigbee: [tz.aqara_opple_operation_mode],
        onEvent: preventReset,
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
        },
    },
```
</detail>

oxc, 2023-03-18:
Attached is my custom quirk

What works:
* Rotate (_start_rotation_, _rotation_, _stop_rotation_) (requires replacement endpoint 71)
* Hold and rotate (_hold_start_rotation_, _hold_rotation_, _stop_hold_rotation_) (requires replacement endpoint 72)
* Single press (_1_single_)
* Double press (_1_double_)
* Hold and release (_1_hold_, _1_release_)
* Battery

What doesn't work:
* In command mode, I'm not able to trigger anything on single press
* I can't figure out how to change the sensitivity of the rotation (I assume it only affects the percent value, where this is relevant because it is capped to 100). There's probably some attribute that can be set, either on endpoint 1 or maybe on 71/72, but this is just me guessing.

Other thoughts:
* I'm not sure if the _rotate_left_/_rotate_right_ events make sense. Is there a way to trigger a device automation trigger based on "greater than"/"less than"? Would it make more sense to add a "direction" event arg and use that to define the device_automation_trigger on stop_rotation?
* I've copied the event names from zigbee-herdsman-converters, but renamed the "_speed" suffix to "_delta". This seems more appropriate to me, it is the respective attribute value change between _this_ rotation event and the last, whereas the unsuffixed attriute is the change between _start_rotation_ and _end_rotation_. 

<details>
<summary>Source code: xiamo/aqara/remote_h1_knob.py</summary>

```python
from typing import Optional, Union, List, Any

import zigpy.types as t
from zigpy.profiles import zha
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import Basic, Identify, OnOff, \
    PowerConfiguration, Ota
from zigpy.zdo.types import NodeDescriptor, LogicalType

from zhaquirks.const import (
    ALT_DOUBLE_PRESS,
    ARGS,
    BUTTON,
    COMMAND,
    COMMAND_OFF,
    COMMAND_TOGGLE,
    DEVICE_TYPE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    ENDPOINTS,
    INPUT_CLUSTERS,
    LONG_PRESS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    LONG_RELEASE,
    ZHA_SEND_EVENT, ALT_LONG_PRESS, NODE_DESCRIPTOR,
)
from zhaquirks.xiaomi import (
    LUMI,
    BasicCluster,
    XiaomiAqaraE1Cluster,
    XiaomiCustomDevice,
)
from zhaquirks.xiaomi.aqara.opple_remote import (
    COMMAND_1_DOUBLE,
    COMMAND_1_HOLD,
    COMMAND_1_SINGLE,
    MultistateInputCluster, COMMAND_1_RELEASE,
)
from zhaquirks.xiaomi.aqara.remote_h1 import PowerConfigurationClusterH1Remote

START_ROTATION = "start_rotation"
ROTATION = "rotation"
STOP_ROTATION = "stop_rotation"
HOLD_START_ROTATION = "hold_start_rotation"
HOLD_ROTATION = "hold_rotation"
[... 187 lines cut ...]
                DEVICE_TYPE: zha.DeviceType.SHADE_CONTROLLER,
                INPUT_CLUSTERS: [
                    KnobManuSpecificCluster,
                ],
            }
        },
    }

    device_automation_triggers = {
        # triggers when operation_mode == event
        (SHORT_PRESS, BUTTON): {COMMAND: COMMAND_1_SINGLE},
        (DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_1_DOUBLE},
        (LONG_PRESS, BUTTON): {COMMAND: COMMAND_1_HOLD},
        (LONG_RELEASE, BUTTON): {COMMAND: COMMAND_1_RELEASE},
        (ROTATE_LEFT, BUTTON): {COMMAND: ROTATE_LEFT},
        (ROTATE_RIGHT, BUTTON): {COMMAND: ROTATE_RIGHT},
        (HOLD_ROTATE_LEFT, BUTTON): {COMMAND: HOLD_ROTATE_LEFT},
        (HOLD_ROTATE_RIGHT, BUTTON): {COMMAND: HOLD_ROTATE_RIGHT},
        # triggers when operation_mode == command
        ## single press does not trigger anything
        (ALT_DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_TOGGLE, ENDPOINT_ID: 1, ARGS: []},
        (ALT_LONG_PRESS, BUTTON): {COMMAND: COMMAND_OFF, ENDPOINT_ID: 1, ARGS: []},
    }
```
</details>

oxc, 2023-03-24:
I've put the code into a gist to track the changes for now: https://gist.github.com/oxc/754d6436ce62d92af660d3671acd9346

I decided that I don't like the separate events for rotate_left/right, and added a rotation_direction arg. This feels more appropriate (but is not tested yet).

If I wanted to create a PR for this, what else would be needed?

oxc, 2023-03-24:
> One thing I noticed, the direction is determined based on the net gain in either direction. For example, turning the dial +1 click to the left and then +4 clicks to the right in one motion is a +3 net rotation right, so is considered "rotate right", even though I started by rotating left. If I rotate left +4 and then right +1, it's a +3 net rotation left, and that's considered "rotate left".

That's correct. This is something that's handled internally by the device itself.

I thought that it sends a stop_rotation and a new start_rotation when the direction changes, but in my latest test (with a different Knob than the one I initially developed the quirk with), this was indeed not the case.

It's even funnier if you compare percent to angle: when you do two full turns to the left, you will be at -720deg but only at -100%, because that is capped at 100. If you now do 1.5 turns to the right, you will be at -180deg but +50% :) That's why I thought it would be nice to be able to change the sensitivity, but I'm no longer even sure if that's indeed controlled in the device, or just in the Aqara software that implements this quirk (probably the Hub, I guess?).

nwithan8, 2023-03-24:
In the meantime, was able to crank out a few blueprints for the device:
https://community.home-assistant.io/t/aqara-h1-rotary-dimmer-switch-dial-remote/551909
https://community.home-assistant.io/t/aqara-h1-rotary-dimmer-switch-dial-media-controls/551917

Thanks @oxc for the great work, and I vouch that support for this should get adopted ASAP

sleepy-salmon, 2023-05-14:
@oxc Thank you so much for your work on this. I think that I am having trouble because I am stuck in command mode but I've not been able to get it into event mode.
Do you have any advice on how to achieve this? What I've tried so far is to go on manage zigbee device, do the quick 5 button press and then read attribute, but this fails. I try inputting the value as 1 and the manufacturer override code and writting the attribute, this gives me a tick but the operation of the device doesn't change. How I am I supposed to go about this correctly?

[image]

sleepy-salmon, 2023-05-17:
My problem solved itself, it just decided to allow me to change the operation mode one day and now I've got it into event mode.

Thanks again for your quirk!

oxc, 2023-05-17:
Sorry I didn't reply before. I don't remember how I did it exactly, I believe I pressed 5 times afterwards instead of before. IIRC, I had to press at least once before the commands would return. But I'm really not sure how exactly I didn't it 😂

0rangutan, 2023-06-02:
@oxc - many thanks for supplying this.
I have the quirk successfully applied according to the H1 knob's Device Zigbee info, but it still only shows Battery, Identify, LQI and RSSI in ZHA, no other capabilities.  I have restarted HA and removed/re-added the device countless times. 
Are there any other steps that I need to complete?
Many thanks

oxc, 2023-06-02:
@0rangutan, if the quirk loaded correctly, you can now configure triggers for the device, and will also receive zha_events. Also have a look at the blueprints from @nwithan8 in the comment above.

0rangutan, 2023-06-02:
Thanks - no zhe_events are firing and @nwithan8's blueprints are set up but not working.
Any ideas?

karrui, 2023-08-10:
Just got this knob too. Quirk works great for events. Cannot figure out how to continuously dim a light as the knob is turned though. Is that possible? Setting `Decrease <light> brightness` on turn event just reduces it by a static 10%, and only after the knob has stopped turning (and thus only when the Rotated event fires). Am I missing something?

oxc, 2023-08-10:
You will have to use the "rotation" zha event for that.

karrui, 2023-08-10:
> You will have to use the "rotation" zha event for that.

Thanks for the quick reply! Good to know it was user error. Will experiment more.

oxc, 2023-08-10:
Perhaps it makes sense to expose the ongoing rotation as events too. I would have appreciated some input from the maintainers on that, but that seems hard to come by.

nwithan8, 2023-08-10:
> Thanks - no zhe_events are firing and @nwithan8's blueprints are set up but not working. Any ideas?

Hey, just noticed today that the some of the commands are reporting differently. Specifically, hold and double are now different (will update the blueprint accordingly), but single press isn't registering at all with ZHA (device lights up, so I know something got triggered).

nwithan8, 2023-08-10:
> > Thanks - no zhe_events are firing and @nwithan8's blueprints are set up but not working. Any ideas?
> 
> Hey, just noticed today that the some of the commands are reporting differently. Specifically, hold and double are now different (will update the blueprint accordingly), but single press isn't registering at all with ZHA (device lights up, so I know something got triggered).
> 
> EDIT: I think I figured it out, device got set into a different mode. Potentially the same issue for you @0rangutan
> 
> @oxc offhand, do you know how to toggle between the two modes?
> 
> https://gist.github.com/oxc/754d6436ce62d92af660d3671acd9346#file-remote_h1_knob-py-L253
> 
> EDIT 2: Code comments help, lol: https://gist.github.com/oxc/754d6436ce62d92af660d3671acd9346#file-remote_h1_knob-py-L77
> 
> Unfortunately, can't get my devices out of the alt mode. I have three of them and even the ones I haven't touched in forever are now in this alt mode. I'm starting to suspect a firmware update...

In the meantime, I have made COMMAND mode variants of the blueprints @0rangutan (see "Edit" section of each original post):

https://community.home-assistant.io/t/aqara-h1-rotary-dimmer-switch-dial-remote/551909
https://community.home-assistant.io/t/aqara-h1-rotary-dimmer-switch-dial-media-controls/551917

oxc, 2023-08-10:
I believe you have to set the operation_mode attribute and then press 5 times. Or the other way around. I cannot say for sure, but somehow like that I managed to switch. The same is reported for some other Aqara devices, and at least back then I could not find a definitive guide on how to do it.

karrui, 2023-08-11:
> Click "Write" where? Is this a Dev Tools service I'm calling, or on the device page?

https://github.com/zigpy/zha-device-handlers/issues/2251#issuecomment-1466326352 should help

0rangutan, 2023-08-15:
@oxc and @nwithan8 - Thanks for all your efforts on this.
I have this working now in ZHA and my dimmer set to Event mode successfully.
It works well, but with the one big difference vs native integration in Z2M - the ability to dim on the fly as you rotate the knob, as discussed above.
Looking at how this works in Z2M, the rotation % is exposed as a sensor and then dimming can work dynamically as the knob is rotated based on this sensor.
Is there any way of exposing the rotation % as a sensor in this quirk?

This was the code I was using for dimming under Z2M:

```
alias: Study Lights Dimming
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.study_rotary_dimmer_action_rotation_percent
    from: unknown
condition: []
action:
  - service: light.turn_on
    data_template:
      brightness: >-
        {{((state_attr('light.study_lights','brightness')|int(1))+((trigger.to_state.state|float(2))/2.55)|round(2)*3)|int(1)}}
    target:
      entity_id: light.study_lights
    enabled: true
mode: single
```

nwithan8, 2023-08-17:
> @oxc and @nwithan8 - Thanks for all your efforts on this. I have this working now in ZHA and my dimmer set to Event mode successfully. It works well, but with the one big difference vs native [...]

I'm seeing `rotation_percent` and `rotation_percent_delta` exposed in my ZHA event (granted, I'm still in COMMAND mode; couldn't get the attributes updated :( )

oxc, 2023-08-17:
Yes, they are exposed as zha event properties. Exposing them a sensor would mean having to keep an internal state of the sensor value, which is weird because of the behavior mentioned above.
It should be easy to fill a custom state helper with the desired values based on the ZHA events, though.

0rangutan, 2023-08-18:
Many thanks.

I can see the start_rotation, rotation and stop_rotation events as zha_events, eg.

```
event_type: zha_event
data:
  device_ieee: 54:ef:44:10:00:7f:62:23
  unique_id: 54:ef:44:10:00:7f:62:23:71:0xfcc0
  device_id: 5e0a167c93599c08f8fbd5798471976a
  endpoint_id: 71
  cluster_id: 64704
  command: rotation
  args:
    rotation_time_delta: 500
    rotation_time: 2200
    rotation_angle_delta: 24
    rotation_angle: 228
    rotation_percent_delta: 6.666666507720947
    rotation_percent: 63.33333206176758
    action: 2
  params: {}
origin: LOCAL
time_fired: "2023-08-18T07:37:07.238438+00:00"
context:
  id: 01H83S75D6F2FV0X1YQV5T3HDJ
  parent_id: null
  user_id: null
```

So the data is available :)

I'd like to try to create an automation that adjusts the brightness on the fly as the rotation changes.  This should work in theory as there is a stream of events that come in as the knob rotates and the various rotation arguments are available.

I'm definitely beyond my level of expertise here.  Any ideas how I might trigger based on these events?  I have started with the following but this doesn't trigger in the tracing when I rotate the knob.

```
alias: Study Lights Dimming
description: ""
trigger:
  - platform: event
    event_type: zha_event
    event_data:
      device_ieee: 54:ef:44:10:00:7f:62:23
      endpoint_id: 71
      command: rotation
      args:
        - rotation_percent 
```

Any ideas?

nwithan8, 2023-08-19:
> Many thanks.
>
> I can see the start_rotation, rotation and stop_rotation events as zha_events, eg.
>
> ```
> event_type: zha_event
> data:
>   device_ieee: 54:ef:44:10:00:7f:62:23
>   unique_id: [...]

I belive the arg you have there is misconfigured.

You can look at a blueprint I made to trigger and handle the ongoing rotation of another ZHA dial device for maybe reference: https://github.com/nwithan8/configs/blob/9ab215f54a3d6ed0e42de6ddc5f57cd132a3a243/home_assistant/blueprints/automations/ers_rotary_dial_light_control_zha.yaml#L158

That said, I don't recall this device consistently sending our rotation events. I remember it being only one per rotation (you'd have to stop and start a new rotation), but I could be wrong.

oxc, 2023-08-19:
> That said, I don't recall this device consistently sending our rotation events. I remember it being only one per rotation (you'd have to stop and start a new rotation), but I could be wrong.

It does for me, and since there are two different events for "percent in this piece of the rotation" `rotation_delta` and "prevent in the full rotation" `rotation`, I strongly assume that's how it works for everybody (unless yours/its firmware does not report these properties).

oxc, 2023-08-19:
Here is a (slightly redacted) Automation I'm using in production, it controls the light when rotating (uses ongoing rotation), and a curtain when hold+rotating (no ongoing rotation)

```yaml
alias: Knob
description: ""
trigger:
  - platform: event
    event_type: zha_event
    event_data:
      device_id: xxxxxxxxxx
condition: []
action:
  - variables:
      command: "{{ trigger.event.data.command }}"
  - choose:
      - conditions:
          - alias: Hold and rotate
            condition: template
            value_template: |
              {{ command == "hold_stop_rotation" }}
        sequence:
          - variables:
              percent: "{{ trigger.event.data.args.rotation_percent | int(0) }}"
              current_position: |
                {{ state_attr('cover.cover', 'current_position') }}
              target_position: |
                {{ min(max(current_position - percent*2, 0), 100) }}
          - service: cover.set_cover_position
            data:
              position: "{{ target_position }}"
            target:
              entity_id: cover.cover
      - conditions:
          - alias: Rotate
            condition: template
            value_template: >
              {{ command == "start_rotation" or command == "rotation" or command
              == "stop_rotation" }}
        sequence:
          - variables:
              percent_delta: "{{ trigger.event.data.args.rotation_percent_delta | int(0) }}"
          - service: light.turn_on
            data:
              brightness_step_pct: "{{ percent_delta }}"
            target:
              entity_id:
                - light.light1
                - light.light2
      - conditions:
          - alias: Single press
            condition: template
            value_template: |
              {{ command == "1_single" }}
        sequence:
          - if:
              - condition: or
                conditions:
                  - condition: state
                    entity_id: light.light1
                    state: "on"
                  - condition: state
                    entity_id: light.light2
                    state: "on"
            then:
              - service: light.turn_off
                data: {}
                target:
                  entity_id:
                    - light.light1
                    - light.light2
            else:
              - service: light.turn_on
                data: {}
                target:
                  entity_id:
                    - light.light1
                    - light.light2
mode: queued
max: 10
```

ToracX, 2023-10-27:
Im having trouble changing operating modes of my H1 dimmer. Does anyone know how to check what mode it is in now?
Also trying te write attributes im getting this error. 
Failed to set attribute: value: EVENT attribute: 9 cluster_id: 64704 endpoint_id: 1

arthpeps, 2023-12-23:
Hello,

I can't even get the custom quirk working. I'v done following :

- I added the following lines to my configuration.yaml :

zha:
  custom_quirks_path: /config/custom_zha_quirks/
  database_path: /config/zigbee.db
  enable_quirks: true

- I created a file named lumi.remote.rkba01.py into the folder custom_zha_quirks with the content of the quirk

I can add the H1 device but interaction goes only from ZHA to the device, not the opposite.

Can anyone help ?

Thanks

arthpeps

ToracX, 2023-12-23:
> Hello,
>
> I can't even get the custom quirk working. I'v done following :
>
> * I added the following lines to my configuration.yaml :
>
> zha: custom_quirks_path: /config/custom_zha_quirks/ database_path: [...]

Did you edit configuration.yaml with the new directory for zha quirks?

oxc, 2024-06-21:
This is not stale and I very much would still like to get it added to the code base

mike-nott, 2025-02-22:
I also would very much like this device added to the core ZHA codebase

adamcaz, 2025-05-16:
I too would like this device added to the core ZHA database please. :)

nwithan8, 2025-11-13:
Not stale.

oxc, 2026-02-16:
Unfortunately I don't have time to look at this this month. Is there anything in the startup logs that would indicate a problem?

trfr3ak, 2026-02-20:
Same for me, worked fine for so long, now after the last update it is broken =/

peersil, 2026-09-26:
Claude developed that quirk for me and it worked out quite well

<details>
<summary>aqara_knob_h1_wireless.py</summary>

```
"""Custom ZHA quirk for Aqara Smart Rotary Knob H1 wireless (lumi.remote.rkba01, ZNXNKG02LM).

Based on the Aqara Dimmer H2 knob handling (zha-device-handlers PR #5346)
and the Zigbee2MQTT lumi converter:
- Presses:   MultistateInput present_value on endpoint 1
             (0=hold, 1=single, 2=double, 255=release)
- Rotation:  Aqara cluster 0xFCC0, action in 0x023A (sent last),
             angle 0x022E, angle speed 0x0230, time 0x0231,
             percent speed 0x0232, percent 0x0233
- Settings:  operation_mode 0x0009 (0=command, 1=event), sensitivity 0x0234

Rotation is reported on endpoint 71 (72 while the knob is held); the
cluster is also added on endpoint 1. Tested with HA 2026.9 / zha-quirks 2.2.2.
"""

from typing import Final

from zigpy import types
from zigpy.zcl.clusters.general import MultistateInput
from zigpy.zcl.foundation import BaseAttributeDefs, ZCLAttributeDef
from zigpy.zdo.types import NodeDescriptor

from zhaquirks import CustomCluster
from zhaquirks.builder import QuirkBuilder
from zhaquirks.const import (
    ATTR_ID,
    BUTTON,
    COMMAND,
    COMMAND_DOUBLE,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    COMMAND_SINGLE,
    DOUBLE_PRESS,
    ENDPOINT_ID,
    LONG_PRESS,
    LONG_RELEASE,
    PRESS_TYPE,
    SHORT_PRESS,
    VALUE,
    ZHA_SEND_EVENT,
)
from zhaquirks.xiaomi import XiaomiAqaraE1Cluster, XiaomiPowerConfiguration

# rotation event names (the same strings Zigbee2MQTT uses)
CMD_START_ROTATING = "start_rotating"
CMD_ROTATION = "rotation"
CMD_STOP_ROTATING = "stop_rotating"
ROTARY_KNOB = "rotary_knob"

BUTTON_ACTIONS = {
    0: COMMAND_HOLD,
    1: COMMAND_SINGLE,
    2: COMMAND_DOUBLE,
    255: COMMAND_RELEASE,
}

ROTATION_ACTIONS = {
    0: CMD_START_ROTATING,
    1: CMD_START_ROTATING,
    2: CMD_ROTATION,
    3: CMD_STOP_ROTATING,
}
ROTATION_PRESSED_BIT = 0x80

[... 135 lines cut ...]
    .number(
        KnobOppleCluster.AttributeDefs.sensitivity.name,
        KnobOppleCluster.cluster_id,
        min_value=180,
        max_value=720,
        step=10,
        translation_key="sensitivity",
        fallback_name="Rotation sensitivity",
    )
    .device_automation_triggers(
        {
            (SHORT_PRESS, BUTTON): {COMMAND: COMMAND_SINGLE},
            (DOUBLE_PRESS, BUTTON): {COMMAND: COMMAND_DOUBLE},
            (LONG_PRESS, BUTTON): {COMMAND: COMMAND_HOLD},
            (LONG_RELEASE, BUTTON): {COMMAND: COMMAND_RELEASE},
            (CMD_START_ROTATING, ROTARY_KNOB): {COMMAND: CMD_START_ROTATING},
            (CMD_ROTATION, ROTARY_KNOB): {COMMAND: CMD_ROTATION},
            (CMD_STOP_ROTATING, ROTARY_KNOB): {COMMAND: CMD_STOP_ROTATING},
        }
    )
    .add_to_registry()
)
```

</details>