zigpy/zha-device-handlers #2268: [Device Support Request] ZY-M100-S Human Presence Detector

r-flash, 2023-03-12:
**Is your feature request related to a problem? Please describe.**
Only Occupancy and Illumimance are exposed.

**Describe the solution you'd like**
According to https://zigbee.blakadder.com/Tuya_ZY-M100-S.html , this device should also support the configuration of sensitivity, range, and delays.

<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0107",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x000c",
        "0x0400",
        "0x0406",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_ztc6ggyl",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_motion.MmwRadarMotion"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Prague",
    "os_name": "Linux",
    "os_version": "5.15.84-v8",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "xiaomi_cloud_map_extractor": {
      "version": "v2.2.0",
      "requirements": [
        "pillow",
        "pybase64",
[... 476 lines cut ...]
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**

[image]

The support for this device was added in https://github.com/zigpy/zha-device-handlers/pull/1928 . There is a custom quirk shared in the discussion which some people reported has prblems but can be a good starting point.

javicalle, 2023-03-16:
Can you test the quirk from #2275?
I believe that has all this implemented.

bemble, 2023-03-16:
> Can you test the quirk from #2275? I believe that has all this implemented.

I tried, it works for me but I have some comments on it (I'll add them in the PR directly).

closed 2023-09-19 by github-actions[bot]