zigpy/zha-device-handlers #2269: Yooksmart D10110 constantly 'opening' or 'closing'

rb2k, 2023-03-12:
I installed yoolax zigbee blinds and was able to set them up with ZHA in Homeassistant.
Both of them show "Firmware: 0x13013001"

Aside from having to switch orientation on one of them, the percentage controls and automations with them work great.
My only problem is that their state never seems to change from 'opening' or 'closing' once their done physically moving.

This leads to the 'stop' button being constantly visible and the inability for me to use the up/down button depending on which state they think they're in:

[image]

Pressing 'stop' doesn't seem to do anything.
I can still set the percentage and they will respond.

Any ideas on how to adapt the quirk / fix this behavior?

TheJulianJES, 2023-03-12:
There was a long-standing issue with all Yooksmart blinds being inverted:
- https://github.com/zigpy/zha-device-handlers/issues/1306
- https://github.com/zigpy/zha-device-handlers/pull/1664

This will be fixed in HA 2023.4.0.

As you didn't fill-out the issue template: please attach your device signature and device diagnostics for the blinds.

rb2k, 2023-03-12:
Huh, no idea how I got past that bug report selector, weird.

Either way, device signature:

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0020",
        "0x0102"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "yooksmart",
  "model": "D10110",
  "class": "zhaquirks.yooksmart.D10110blinds.D10110Blinds"
}
```

And Diagnostics:

```
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/New_York",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "vm",
    "run_as_root": true
  },
  "custom_components": {
    "thethingsnetwork": {
      "version": "0.2.0",
      "requirements": []
    },
    "hacs": {
      "version": "1.31.0",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "dahua": {
      "version": "0.5.0",
      "requirements": []
    },
    "fullykiosk": {
[... 340 lines cut ...]
              "0x0008": {
                "attribute_name": "current_position_lift_percentage",
                "value": 55
              }
            },
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

jayfunk, 2023-08-20:
I seem to be running into a similar issue on HA 2023.7.2. It seems to be happening for only one of my Yooksmart blinds periodically. It will report the wrong state of the blind and I will have to manually press the up or down button (depending on the state it is supposed to be in) to get it to be up to date. All of my other blinds seem to be working fine. 

```
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.7.2",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.11.4",
    "docker": true,
    "arch": "x86_64",
    "timezone": "America/Chicago",
    "os_name": "Linux",
    "os_version": "5.19.0-50-generic",
    "run_as_root": true
  },
  "custom_components": {
    "hacs": {
      "version": "1.32.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "alexa_media": {
      "version": "4.6.5",
      "requirements": [
        "alexapy==1.26.8",
        "packaging>=20.3",
        "wrapt>=1.12.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "onboarding",
      "usb"
    ],
    "codeowners": [
      "@dmulcahey",
      "@adminiuga",
      "@puddly"
    ],
    "config_flow": true,
    "dependencies": [
      "file_upload"
    ],
    "documentation": "https://www.home-assistant.io/integrations/zha",
    "iot_class": "local_polling",
    "loggers": [
      "aiosqlite",
      "bellows",
      "crccheck",
      "pure_pcapy3",
      "zhaquirks",
      "zigpy",
      "zigpy_deconz",
      "zigpy_xbee",
      "zigpy_zigate",
      "zigpy_znp"
    ],
    "requirements": [
      "bellows==0.35.8",
      "pyserial==3.5",
      "pyserial-asyncio==0.6",
      "zha-quirks==0.0.101",
      "zigpy-deconz==0.21.0",
[... 249 lines cut ...]
            "unsupported_attributes": {}
          },
          "0x0005": {
            "endpoint_attribute": "scenes",
            "attributes": {},
            "unsupported_attributes": {}
          }
        },
        "out_clusters": {
          "0x0003": {
            "endpoint_attribute": "identify",
            "attributes": {},
            "unsupported_attributes": {}
          },
          "0x0019": {
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

closed 2024-02-23 by github-actions[bot]