zigpy/zha-device-handlers #2270: Support for Aqara 2 Gang switch without neutral (lumi.switch.l2aeu1)

sophipl, 2023-03-12:
I would like to have the ability to trigger on double button press, which I read that this device have in Zigbee2MQTT mode.
The device has automation abilities in HA on state changes (light on, light off), but not of button presses, apparently you can have the ability to disconnect button from it's internal relay and automate it differently as well.

I didn't see any pre-existing quirks.
Sadly I understand too zigbee protocol too little to write this quirk myself.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l2aeu1",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.1.6",
[... 396 lines cut ...]
          }
        },
        "out_clusters": {}
      },
      "242": {
        "device_type": {
          "name": "unknown",
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

sophipl, 2023-03-24:
Well, I think I got as far as I could
The cluster commands works fine, I can detach relay, invert LEDs, disable anti-flicker
But buttons doesn't work
I sit with wireshark and HA logs, didn't manage to make it work.
And it's maddening, everything is a string in quirk `device_automation_triggers`, how can you know how to match things and from where they came from? There is something to be said about strongly typed languages.

<details>
<summary>Quirk so far</summary>

```python
import copy
from enum import Enum

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zhaquirks import CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    DeviceTemperature,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
    MultistateInput,
    GreenPowerProxy,
    Alarms,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    TRIPLE_PRESS,
    LONG_PRESS,
    ALT_DOUBLE_PRESS,
    DOUBLE_PRESS,
    ALT_SHORT_PRESS,
    LEFT,
    RIGHT,
    COMMAND_TOGGLE,
    COMMAND,
    BUTTON,
    ENDPOINT_ID,
    ARGS,
    CLUSTER_ID,
    ATTR_ID,
    PRESS_TYPE,
    VALUE,
)
from zhaquirks.xiaomi import (
    LUMI
)

COMMAND_SINGLE = "single"
COMMAND_DOUBLE = "double"
COMMAND_BUTTON_HOLD = "button_hold"

from zhaquirks.xiaomi.aqara.opple_switch import (OppleSwitchCluster, BOTH_BUTTONS)
from zhaquirks.xiaomi.aqara.opple_remote import (
    MultistateInputCluster,
    OppleCluster,
    OPPLE_CLUSTER_ID
)
[... 140 lines cut ...]
        (COMMAND_BUTTON_HOLD, BOTH_BUTTONS): {ENDPOINT_ID: 1, CLUSTER_ID: XiaomiAqaraMasterCluster.cluster_id, ARGS: {ATTR_ID: 0x00FC, VALUE: 0, PRESS_TYPE: COMMAND_BUTTON_HOLD}},
    }
```

</details>

The buttons work like this, left button is source_endpoint 41, right button is source_endpoint 42, both buttons is 51
It is sending value of uint16 of 1 or 2, for one press or 2 presses, attribute id 0x0055 (Preset value)
When you press 3 times, it does something funny, it introduces itself, sends string value with model information.
Holding a button sends data for cluster 0xfcc0 with value false of boolean, attribute id 0x00fc

It seems that ZHA cannot decode the values as event, I don't know, just a packet

```log

2023-03-24 22:06:34.973 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD3A6), src_ep=41, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000),dst_ep=1, source_route=None, extended_timeout=False, tsn=116, profile_id=260, cluster_id=18, data=Serialized[b'\x18>\nU\x00!\x02\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-34)
2023-03-24 22:05:06.802 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD3A6), src_ep=41, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000),dst_ep=1, source_route=None, extended_timeout=False, tsn=109, profile_id=260, cluster_id=18, data=Serialized[b'\x188\nU\x00!\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-33)
2023-03-24 22:05:08.801 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD3A6), src_ep=41, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000),dst_ep=1, source_route=None, extended_timeout=False, tsn=110, profile_id=260, cluster_id=18, data=Serialized[b'\x189\nU\x00!\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-33)
2023-03-24 22:39:43.094 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD3A6), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=161, profile_id=260, cluster_id=64704, data=Serialized[b'\x1c_\x11L\n\xfc\x00\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-34)
2023-03-24 22:05:09.654 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD3A6), src_ep=42, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000),dst_ep=1, source_route=None, extended_timeout=False, tsn=112, profile_id=260, cluster_id=18, data=Serialized[b'\x18;\nU\x00!\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-33)
2023-03-24 22:25:20.180 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xD3A6), src_ep=51, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=141, profile_id=260, cluster_id=18, data=Serialized[b'\x18E\nU\x00!\x01\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-34)

```

Maybe you will be able to fix the problem, my head explodes now :)

sophipl, 2023-03-25:
I managed to do it, it took some effort
I don't know why `device_automation_triggers` didn't wanted to match at some point of my development, so I limited it to `COMMAND`

If someone want's to use it, please go ahead

<details>
<summary>aqara_i2aeu1.py</summary>

```python
import copy
from enum import Enum

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zhaquirks import CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    DeviceTemperature,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
    MultistateInput,
    GreenPowerProxy,
    Alarms,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    TRIPLE_PRESS,
    LONG_PRESS,
    ALT_DOUBLE_PRESS,
    DOUBLE_PRESS,
    ALT_SHORT_PRESS,
    LEFT,
    RIGHT,
    COMMAND_TOGGLE,
    COMMAND,
    BUTTON,
    ENDPOINT_ID,
    ARGS,
    CLUSTER_ID,
    ATTR_ID,
    PRESS_TYPE,
    VALUE,
    ZHA_SEND_EVENT,
    COMMAND_HOLD,
    COMMAND_TRIPLE
)
from zhaquirks.xiaomi import (
    LUMI
)

COMMAND_SINGLE = "single"
COMMAND_DOUBLE = "double"
BUTTON_ANY = "any"
LEFT_BUTTON_ENDPOINT=41
RIGHT_BUTTON_ENDPOINT=42
BOTH_BUTTONS_ENDPOINT=51

from zhaquirks.xiaomi.aqara.opple_switch import (OppleSwitchCluster, BOTH_BUTTONS)
[... 214 lines cut ...]
                    MultistateInputCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, LEFT): { COMMAND: "{}_{}".format(LEFT, COMMAND_SINGLE)},
        (DOUBLE_PRESS, LEFT): { COMMAND: "{}_{}".format(LEFT, COMMAND_DOUBLE)},
        (SHORT_PRESS, RIGHT): { COMMAND: "{}_{}".format(RIGHT, COMMAND_SINGLE)},
        (DOUBLE_PRESS, RIGHT): { COMMAND: "{}_{}".format(RIGHT, COMMAND_DOUBLE)},
        (SHORT_PRESS, BOTH_BUTTONS): { COMMAND: "{}_{}".format(BOTH_BUTTONS, COMMAND_SINGLE)},
        (DOUBLE_PRESS, BOTH_BUTTONS): { COMMAND: "{}_{}".format(BOTH_BUTTONS, COMMAND_DOUBLE)},
        (COMMAND_HOLD, BUTTON): { COMMAND: "{}_{}".format(BUTTON_ANY, COMMAND_HOLD)},
    }
```

</details>

sophipl, 2023-03-25:
FYI @TheJulianJES 
If someone thinks it's worth integrating

TheJulianJES, 2023-03-25:
We can leave this open for now (until zha-quirks maybe fixes this).

grzegor, 2023-04-17:
@sophipl thank you! Decoupling worked and the single/double press automations works flawlessly!

cpressland, 2023-04-30:
@sophipl hello! I just received my `lumi.switch.l2aeu1` today. I've loaded your quirk into Home Assistant and can confirm it's being used by ZHA. The only thing I'm not sure about is how to put the device in decoupled mode, any advice you can give me?

sophipl, 2023-05-01:
@cpressland Great that you figured out!
I don't think there is a way to export the value into selection box in HA, so we have to stumble in the menus. :)

Leworoe, 2023-05-25:
I needed to edit this quirk. I don't know why my switch has more `ENDPOINTS` in the `signature`. The `INPUT_CLUSTERS` in the first endpoint differ.
<details>

<summary>Device signature</summary>

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0012",
        "0xfcc0"
      ],
      "output_clusters": []
    },
    "41": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "42": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "51": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0012"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
[... 315 lines cut ...]
                    MultistateInputCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, LEFT): { COMMAND: "{}_{}".format(LEFT, COMMAND_SINGLE)},
        (DOUBLE_PRESS, LEFT): { COMMAND: "{}_{}".format(LEFT, COMMAND_DOUBLE)},
        (SHORT_PRESS, RIGHT): { COMMAND: "{}_{}".format(RIGHT, COMMAND_SINGLE)},
        (DOUBLE_PRESS, RIGHT): { COMMAND: "{}_{}".format(RIGHT, COMMAND_DOUBLE)},
        (SHORT_PRESS, BOTH_BUTTONS): { COMMAND: "{}_{}".format(BOTH_BUTTONS, COMMAND_SINGLE)},
        (DOUBLE_PRESS, BOTH_BUTTONS): { COMMAND: "{}_{}".format(BOTH_BUTTONS, COMMAND_DOUBLE)},
        (COMMAND_HOLD, BUTTON): { COMMAND: "{}_{}".format(BUTTON_ANY, COMMAND_HOLD)},
    }
```

</details>

ardstuebck, 2023-11-11:
Hi, unfortunately, neither of the quirks posted here work for my device. I don't receive any events when listening to "zha_event." My device is shown as two lights, and the right button doesn't change when I press the button physically. Can anyone help me?
<details>
  <summary>My device signature</summary> 

```json
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4447, maximum_buffer_size=127, maximum_incoming_transfer_size=100, server_mask=11264, maximum_outgoing_transfer_size=100, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0002",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0009"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006"
      ],
      "output_clusters": []
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "LUMI",
  "model": "lumi.switch.l2aeu1",
  "class": "zigpy.device.Device"
}
```
</details>

theorlangur, 2023-12-16:
@sophipl , thanks for the quirk, it works for me (l2aeu1)
for anyone attempting to apply it as-is:
it seems that it is a little bit outdated and there is no 'OPPLE_CLUSTER_ID':
```python
from zhaquirks.xiaomi.aqara.opple_remote import (
    OppleCluster,
    OPPLE_CLUSTER_ID,
    STATUS_TYPE_ATTR,
    PRESS_TYPES
)
```
my fix was to remove OPPLE_CLUSTER_ID from the imports and replace all usages in this file with OppleCluster.cluster_id

<details>
  <summary>updated version of aqara_l2aeu1.py</summary>

````python
import copy
from enum import Enum

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zhaquirks import CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    DeviceTemperature,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
    MultistateInput,
    GreenPowerProxy,
    Alarms,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SHORT_PRESS,
    TRIPLE_PRESS,
    LONG_PRESS,
    ALT_DOUBLE_PRESS,
    DOUBLE_PRESS,
    ALT_SHORT_PRESS,
    LEFT,
    RIGHT,
    COMMAND_TOGGLE,
    COMMAND,
    BUTTON,
    ENDPOINT_ID,
    ARGS,
    CLUSTER_ID,
    ATTR_ID,
    PRESS_TYPE,
    VALUE,
    ZHA_SEND_EVENT,
    COMMAND_HOLD,
    COMMAND_TRIPLE
)
from zhaquirks.xiaomi import (
    LUMI
)

[... 220 lines cut ...]
                INPUT_CLUSTERS: [
                    MultistateInputCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 0xA1E0,
                DEVICE_TYPE: 0x0061,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

    device_automation_triggers = {
        (SHORT_PRESS, LEFT): { COMMAND: "{}_{}".format(LEFT, COMMAND_SINGLE)},
        (DOUBLE_PRESS, LEFT): { COMMAND: "{}_{}".format(LEFT, COMMAND_DOUBLE)},
        (SHORT_PRESS, RIGHT): { COMMAND: "{}_{}".format(RIGHT, COMMAND_SINGLE)},
        (DOUBLE_PRESS, RIGHT): { COMMAND: "{}_{}".format(RIGHT, COMMAND_DOUBLE)},
        (SHORT_PRESS, BOTH_BUTTONS): { COMMAND: "{}_{}".format(BOTH_BUTTONS, COMMAND_SINGLE)},
        (DOUBLE_PRESS, BOTH_BUTTONS): { COMMAND: "{}_{}".format(BOTH_BUTTONS, COMMAND_DOUBLE)},
        (COMMAND_HOLD, BUTTON): { COMMAND: "{}_{}".format(BUTTON_ANY, COMMAND_HOLD)},
    } 
````
</details>

dmpcc, 2024-11-14:
The quirk provided by theorlang has enabled the OppleSwitchCluster option for me. I can now decouple the switch from the button. I do see some other options now that are not supported by the Switch (e.g. Power sensor). In addition the device pages does sometimes become unresponsive.