zigpy/zha-device-handlers #2272: [BUG] TS0002 by _TZ3000_fbjdkph9, 2 gang light switch, when either switch to turn off/on by ZHA both do the same action

Diramu, 2023-03-13:
**Describe the bug**
This device has two light entities, when HA turns either on/off, both lights physical switch on/off, then a few seconds later the others entity states updates to match what you did (to the other).  In others words, they are acting like a group.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to device page of this unit via the ZHA integration, both control switches show off.
2. Click on the toggle switch for the first control switch
3. Yes see both lights connected to the device turn on instantly
4. logbook reports: _Ikuu_2sw Light Left turned on triggered by service light.turn_on_
5. about 5 seconds later, the toggle switch for the second control switch turn on, and the logbook reports: _Ikuu_2sw Light Right turned on_

**Expected behavior**
I would have expected just the light controlled by the first control switch to turn on.

**Screenshots**
[image]
<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0xe000",
        "0xe001"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3000_fbjdkph9",
  "model": "TS0002",
  "class": "zigpy.device.Device"
}
```

</details>
[... 503 lines cut ...]
2023-03-13 23:40:25.435 DEBUG (MainThread) [bellows.zigbee.application] Received incomingRouteRecordHandler frame with [0xa712, e0:79:8d:ff:fe:67:bf:5a, 160, -60, []]
2023-03-13 23:40:25.435 DEBUG (MainThread) [bellows.zigbee.application] Processing route record request: (0xa712, e0:79:8d:ff:fe:67:bf:5a, 160, -60, [])
2023-03-13 23:40:25.441 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=230), 160, -60, 0xa712, 255, 255, b'\x18w\n\x00\x00\x10\x00']
2023-03-13 23:40:25.441 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=230), 160, -60, 0xa712, 255, 255, b'\x18w\n\x00\x00\x10\x00']
2023-03-13 23:40:25.441 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xA712), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=230, profile_id=260, cluster_id=6, data=Serialized[b'\x18w\n\x00\x00\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=160, rssi=-60)
2023-03-13 23:40:25.442 DEBUG (MainThread) [zigpy.zcl] [0xA712:1:0x0006] Received ZCL frame: b'\x18w\n\x00\x00\x10\x00'
2023-03-13 23:40:25.442 DEBUG (MainThread) [zigpy.zcl] [0xA712:1:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=119, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-13 23:40:25.442 DEBUG (MainThread) [zigpy.zcl] [0xA712:1:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2023-03-13 23:40:25.442 DEBUG (MainThread) [zigpy.zcl] [0xA712:1:0x0006] Received command 0x0A (TSN 119): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2023-03-13 23:40:25.442 DEBUG (MainThread) [zigpy.zcl] [0xA712:1:0x0006] Attribute report received: on_off=<Bool.false: 0>
2023-03-13 23:40:25.464 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received incomingMessageHandler: [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=2, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=231), 164, -59, 0xa712, 255, 255, b'\x18x\n\x00\x00\x10\x00']
2023-03-13 23:40:25.464 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=2, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=231), 164, -59, 0xa712, 255, 255, b'\x18x\n\x00\x00\x10\x00']
2023-03-13 23:40:25.464 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xA712), src_ep=2, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=231, profile_id=260, cluster_id=6, data=Serialized[b'\x18x\n\x00\x00\x10\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=164, rssi=-59)
2023-03-13 23:40:25.464 DEBUG (MainThread) [zigpy.zcl] [0xA712:2:0x0006] Received ZCL frame: b'\x18x\n\x00\x00\x10\x00'
2023-03-13 23:40:25.464 DEBUG (MainThread) [zigpy.zcl] [0xA712:2:0x0006] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True, *is_reply=True), tsn=120, command_id=10, *direction=<Direction.Client_to_Server: 1>, *is_reply=True)
2023-03-13 23:40:25.465 DEBUG (MainThread) [zigpy.zcl] [0xA712:2:0x0006] Decoded ZCL frame: OnOff:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2023-03-13 23:40:25.465 DEBUG (MainThread) [zigpy.zcl] [0xA712:2:0x0006] Received command 0x0A (TSN 120): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=Bool, value=<Bool.false: 0>))])
2023-03-13 23:40:25.465 DEBUG (MainThread) [zigpy.zcl] [0xA712:2:0x0006] Attribute report received: on_off=<Bool.false: 0>
2023-03-13 23:40:25.465 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=light.ikuu_2sw_light_2, old_state=<state light.ikuu_2sw_light_2=on; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], color_mode=onoff, off_with_transition=False, off_brightness=None, friendly_name=Ikuu_2sw Light Right, supported_features=LightEntityFeature.FLASH @ 2023-03-13T23:31:37.483669+11:00>, new_state=<state light.ikuu_2sw_light_2=off; supported_color_modes=[<ColorMode.ONOFF: 'onoff'>], off_with_transition=False, off_brightness=None, friendly_name=Ikuu_2sw Light Right, supported_features=LightEntityFeature.FLASH @ 2023-03-13T23:40:25.465557+11:00>>
```

</details>

**Additional context**
thats all folks

javicalle, 2023-03-16:
It seems to be another variation for the `TS000X` devices.

My proposal would be:

<details open>
<summary>ts0002.py</summary>

```python
"""tuya TS000X Switches."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODEL,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    TuyaZBE000Cluster,
    TuyaZBElectricalMeasurement,
    TuyaZBExternalSwitchTypeCluster,
    TuyaZBMeteringCluster,
    TuyaZBOnOffAttributeCluster,
)
from zhaquirks.tuya.mcu import EnchantedDevice

class Switch_2G_Var03(EnchantedDevice, CustomDevice):
    """Tuya 2 gang (variation 03))."""

    signature = {
        MODEL: "TS0002",
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=256
            # device_version=1
            # input_clusters=[0, 3, 4, 5, 6, 57344, 57345]
            # output_clusters=[10, 25]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=2 profile=260 device_type=256
            # device_version=1
            # input_clusters=[3, 4, 5, 6, 57344, 57345]
            # output_clusters=[]>
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    TuyaZBE000Cluster.cluster_id,
                    TuyaZBExternalSwitchTypeCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # device_version=0
            # input_clusters=[]
            # output_clusters=[33]>
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaZBOnOffAttributeCluster,
                    TuyaZBE000Cluster,
                    TuyaZBExternalSwitchTypeCluster,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }
```

</details>

javicalle, 2023-03-20:
Is something that anyone must try to know if it will fix the issue or not.

There is a good guide about how configure your local quirks folder:
* https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274

Once configured, create a new file (ie: `ts0002.py`) and put the proposed code inside.

Save changes, restart HA, remove your device from HA, wait a minute or so, and pair the device again.

MattWestb, 2023-03-23:
If the firmware is not supporting it we cant do any thing to make it working.
Best is hidden or deleting the GUI entries for them if you dont like having no working elements.

javicalle, 2023-03-23:
> I found they have no effect. These lights only seems to do LightWhenOn.
> Same with power on state, only seems to do LastSate (vs Off, On)

If don't work when you use it with gang 1 then is not device supported for sure.
Nothing else can be done.

closed 2023-03-24 by TheJulianJES

ausfas, 2025-01-30:
> Just gave it a try. It did fix the problem. HA can now control each switch independently
> 
> I did notice 4 configuration options, not sure if they were there last time. Backlight mode , Power on state x 2
> 
> Backlightmode mode has 3 options
> 
> 1. Off
> 2. LightWhenOn
> 3. LightWhenOff
>    I assume are mention to control the light highlighting the switch.
> 
> I found they have no effect. These lights only seems to do LightWhenOn. Same with power on state, only seems to do LastSate (vs Off, On)
> 
> Is there a step to make these configuration active?

Hi ...did you find a solution to this problem ? I had the same issue..it is very annoying that we can't turn off backlight at night

PR #2296 (2023-03-24): Add Tuya `TS0002` 2 gang switch variant
New `Switch_2G_Var03` variation for the `TS0002` Tuya 2 gang devices.

Fix: #2272