zigpy/zha-device-handlers #2274: [Device Support Request] IKEA E2112 Vindstyrka Air Sensor.

MattWestb, 2023-03-14:
**Is your feature request related to a problem? Please describe.**
Some functions is not ZCL R8 compatible and is not  working.
**Describe the solution you'd like**
Adding support for missing functions in ZHA

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0402",
        "0x0405",
        "0x042a",
        "0xfc57",
        "0xfc7c",
        "0xfc7e"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0x0020",
        "0x0202"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "VINDSTYRKA",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Supervised",
    "version": "2023.3.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "armv7l",
    "timezone": "Europe/Vienna",
    "os_name": "Linux",
    "os_version": "5.15.93-sunxi",
    "supervisor": "2023.03.1",
[... 405 lines cut ...]
<details>
<summary>Additional logs</summary>

```
Nothing HERE !!!
```

</details>

**Additional context**
From CSA-IOT cert:
Zigbee 3.0
Vindstyrka Air Sensor
SKU E2112
Firmware Version 1.0.11
Hardware Version 1
Certificate ID ZIG23024ZB331492-24
Certified Date 02/23/2023

The device is shipped with firmware 1.0.10 but  the new have not being pushed to TF or DG but is very likely being done very soon (I shall trying sniffing it with Dirigea if its possible then the Zigbee module is debug locked so the device is sitting in "caranteen" for not  getting updated alone).

Its looks being very nice ZCL standard cluster but the VOC is not working and the  fan is being used for steering starkvind if being direct pared with it and i think it shall being possible doing the same with ZHA or using its command for other fan devices.
Z2M have making the most work so worth looking what the device can do but i think they have missing some parts and some can being done better in ZHA :-))

The device is working OK for the moment so i trying focus on catching the OTA firmware so we is getting one if users need it and later getting full support for the device.

MattWestb, 2023-03-14:
Then looking i was finding one older version that early devices can being shipped with.
```
CSA-IOT certed Firmware Versions:

1.0.009
06/14/2022

1.0.010
08/08/2022

1.0.11
02/23/2023
```
I only having the 1.0.0.10 and dont knowing if the other is working the same way.

just-oblivious, 2023-03-23:
Cluster `0xfc7e` in endpoint `1` provides a measurement in the range of 0-500 that correlates with the tVOC trend display on the unit itself.

See also: https://community.home-assistant.io/t/ikea-vindstyrka-zigbee-air-quality-sensor/549747/3?u=donny007x

MattWestb, 2023-03-23:
I was adding `0xFC7F` and `0xFC80` cluster for Symfonisk 2 and we need doing  the same for this device and we also need looking if Zigbee is having one standard cluster for tVOC in the  ZCL R8 and also if Matter ZCL is having some update on it but they is not having adding all classes and can coming later (IKEA is very heavy in CSA-IOT / Matter aliens and is using functions that is not in the standard but is coming).

My device is siting in carantene for not getting OTA updates from Dirigera but if some is having one connected to Dirigera and getting update pleas informing my and i trying sniffing it  and making one OTA file that  we can using in ZHA (i only have one chance getting all packages and then i must baying one more device for getting the missing packages). I dont know ind TF is having support for or if getting it but  is no OTA released in the open feed.

I is having little mush in the pipe that must being fixed but i trying helping as much i can with knowledge for getting good support for this device.

Also the fan control is interesting then the device is using it then connecting to Starkvind for steering the fan speed and can being used to controlling other devices but i think it shall being bounded to devices fan cluster and we shall getting it working in ZHA if its possible.
But i dont have any Strakvind and i was testing flashing the OTA firmware on one Zigbee module but it looks not starting OK then its not having all hardware its need for working OK so wee need some with one Starkvind later for testing.

just-oblivious, 2023-03-23:
The ZCL R8 doc specifies a standard cluster ID for PM2.5 (`0x042A`, listed under "Concentration Measurement"), but not one for tVOC. I couldn't find anything in the Matter Clusters on air quality sensing.

The Sensirion SEN54 sensor module inside VINDSTYRKA reports a "VOC index" on a scale of 0-500, which matches the bounds reported by the attribute. So it looks like we're getting the processed value directly from the sensor: https://sensirion.com/media/documents/02232963/6294E043/Info_Note_VOC_Index.pdf

MattWestb, 2023-03-23:
Was looking if Matter have implanting the sensor cluster but i was not finding PM2.5 and tVOC so its not in the pipe for next update of ZCL for Zigbee and Matter.

How have HA implanting it for other sensors ?
Some sensors that is being tracked https://github.com/zigpy/zigpy/discussions/691.
How have Z2M mapping it in HA ?

MattWestb, 2023-04-05:
New version 1.0.11 looks having the same end point and cluster setting:
```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4476, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0402",
        "0x0405",
        "0x042a",
        "0xfc57",
        "0xfc7c",
        "0xfc7e"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019",
        "0x0020",
        "0x0202"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "IKEA of Sweden",
  "model": "VINDSTYRKA",
  "class": "vindstyrka.IkeaVINDSTYRKA"
}
```
I was sniffing the upgrade but i was loosing 2 package so i cant doing one OTA file but i shall fixing it in one other way (sad for IKEA getting one updated device back that was broken of there OTA update  ;-))  ) so users without Dirigear can getting there devices updated.

JohnNeville, 2024-08-10:
Why did this get closed? There is a PR open specifically for this issue
https://github.com/zigpy/zha-device-handlers/pull/2315

jpizquierdo, 2024-10-08:
can we reopen the issue? I have a vindstyrka and I offert myself to help with this work