zigpy/zha-device-handlers #2276: [BUG] Issue pairing and subsequent updating of Aqara FP1 (zhaquirks.xiaomi.aqara.motion_ac01.AqaraLumiMotionAc01)

cvocvo, 2023-03-16:
**Describe the bug**
I am trying to deploy an Aqara FP1 via ZHA on Home Assistant (latest version). I'm able to detect and add the Aqara FP1, however, it fails to add other sensors and it fails to load device information such as Device Temperature. Last Seen timestamp also fails to update. I've tried resetting the device and repairing ~10 times and have power cycled the device and also restarted Home Assistant several times.

**To Reproduce**
What I'm doing is:
- Open Home Assistant
- Go to ZHA and hit Add Device
- Hold the reset button on the back of the Aqara FP1 for 5-10 seconds, it'll blink.
- ZHA detects it as an Aqara lumi.motion.ac01
- It fails to add additional sensors and it fails to update any information from the Aqara FP1 after it's paired with ZHA.
- In the logs I saw these errors:
```
[0x35DB](lumi.motion.ac01): power source: Mains
[0x35DB](lumi.motion.ac01): completed initialization
monitoring_mode is not supported - skipping AqaraMonitoringMode entity creation
approach_distance is not supported - skipping AqaraApproachDistance entity creation
motion_sensitivity is not supported - skipping AqaraMotionSensitivity entity creation
```
- But on subsequent attempts to add it, I didn't see that, because it doesn't think it's a new device:
```
Device is initialized <Device model='lumi.motion.ac01' manuf='aqara' nwk=0x095B ieee=54:ef:44:10:00:6d:ac:e3 is_initialized=True>
Checking quirks for aqara lumi.motion.ac01 (54:ef:44:10:00:6d:ac:e3)
Considering <class 'zhaquirks.xiaomi.aqara.motion_ac01.AqaraLumiMotionAc01'>
Found custom device replacement for 54:ef:44:10:00:6d:ac:e3: <class 'zhaquirks.xiaomi.aqara.motion_ac01.AqaraLumiMotionAc01'>
device - 0x095B:54:ef:44:10:00:6d:ac:e3 entering async_device_initialized - is_new_join: False
device - 0x095B:54:ef:44:10:00:6d:ac:e3 has been reset and re-added or its nwk address changed
skipping discovery for previously discovered device - 0x095B:54:ef:44:10:00:6d:ac:e3
```

**Expected behavior**
Other users with ZHA such as this person have a setup that includes more data and also seems to work.
screenshot via: http://www.youtube.com/watch?v=8Ri2LCvH3-g&t=14m43s
[image]

**Screenshots**
[image]

[image]

Device Reconfiguration also fails:
[image]

<details>

<summary>Device signature</summary>

```yaml
(I can't seem to find the Zigbee Device Signature button, maybe this?)
Device info
lumi.motion.ac01
by aqara
Zigbee info
IEEE: 54:ef:44:10:00:6d:ac:e3
Nwk: 0x2f5e
Device Type: EndDevice
LQI: Unknown
RSSI: Unknown
Last Seen: 2023-03-15T19:52:50
Power Source: Mains
Quirk: zhaquirks.xiaomi.aqara.motion_ac01.AqaraLumiMotionAc01
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
[... 513 lines cut ...]
[0x72F8](S31 Lite zb): Device seen - marking the device available and resetting counter
[0x72F8](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x095B), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=115, profile_id=0, cluster_id=2, data=Serialized[b'\x02\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-52)
[0x095b:zdo] ZDO request ZDOCmd.Node_Desc_req: [0x0000]
[0x095b:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_req([0x0000])
Device 0x095b (54:ef:44:10:00:6d:ac:e3) left the network
[0x095B](lumi.motion.ac01): Update device availability -  device available: True - new availability: False - changed: True
[0x095B](lumi.motion.ac01): Device availability changed and device became unavailable
[0x66E6](S31 Lite zb): Device seen - marking the device available and resetting counter
[0x66E6](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
[0x66E6](S31 Lite zb): Device seen - marking the device available and resetting counter
[0x66E6](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
[0x9602](unk_model): last_seen is None, marking the device unavailable
[0x9602](unk_model): Update device availability -  device available: False - new availability: False - changed: False
[0x5C07](WB01): Device seen - marking the device available and resetting counter
[0x5C07](WB01): Update device availability -  device available: True - new availability: True - changed: False
[0xF3B8](TH01): Device seen - marking the device available and resetting counter
[0xF3B8](TH01): Update device availability -  device available: True - new availability: True - changed: False

```

</details>

**Additional context**
Add any other context about the problem here.

cvocvo, 2023-03-16:
I did a device removal, reboot of HASS OS, and reset of the Aqara FP1; I got the logs during pairing that show more of the errors I think:

<details>

<summary>Logs during pairing</summary>

```
[0xF3B8](TH01): Device seen - marking the device available and resetting counter
[0xF3B8](TH01): Update device availability -  device available: True - new availability: True - changed: False
[0x72F8](S31 Lite zb): Device seen - marking the device available and resetting counter
[0x72F8](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
[0x8C12](WB01): Device seen - marking the device available and resetting counter
[0x8C12](WB01): Update device availability -  device available: True - new availability: True - changed: False
[0x916B](S31 Lite zb): Device seen - marking the device available and resetting counter
[0x916B](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
[0x5C07](WB01): Device seen - marking the device available and resetting counter
[0x5C07](WB01): Update device availability -  device available: True - new availability: True - changed: False
[0x9F5A](TH01): Device seen - marking the device available and resetting counter
[0x9F5A](TH01): Update device availability -  device available: True - new availability: True - changed: False
New device 0x782c (54:ef:44:10:00:6d:ac:e3) joined the network
[0x782c] Scheduling initialization
Tries remaining: 3
[0x782c] Requesting 'Node Descriptor'
Tries remaining: 2
[0x782c] Extending timeout for 0x21 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x782C), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.Broadcast: 15>, address=<BroadcastAddress.ALL_ROUTERS_AND_COORDINATOR: 65532>), dst_ep=0, source_route=None, extended_timeout=False, tsn=132, profile_id=0, cluster_id=19, data=Serialized[b'\x01,x\xe3\xacm\x00\x10D\xefT\x8c'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-50)
Device 0x782c (54:ef:44:10:00:6d:ac:e3) joined the network
[0x782c] Scheduling initialization
[0x782c] Canceling old initialize call
Received frame on uninitialized device <Device model=None manuf=None nwk=0x782C ieee=54:ef:44:10:00:6d:ac:e3 is_initialized=False> from ep 0 to ep 0, cluster 19: b'\x01,x\xe3\xacm\x00\x10D\xefT\x8c'
[0x782c:zdo] ZDO request ZDOCmd.Device_annce: [0x782C, 54:ef:44:10:00:6d:ac:e3, 140]
Tries remaining: 3
[0x782c] Requesting 'Node Descriptor'
Tries remaining: 2
[0x782c] Extending timeout for 0x23 request
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x782C), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=133, profile_id=0, cluster_id=32770, data=Serialized[b'!\x00,x\x02@\x8c4\x12l\x7f\x00\x00,\x7f\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-50)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x782C ieee=54:ef:44:10:00:6d:ac:e3 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'!\x00,x\x02@\x8c4\x12l\x7f\x00\x00,\x7f\x00\x00'
[0x782c:zdo] ZDO request ZDOCmd.Node_Desc_rsp: [<Status.SUCCESS: 0>, 0x782C, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)]
[0x782c:zdo] No handler for ZDO request:ZDOCmd.Node_Desc_rsp([<Status.SUCCESS: 0>, 0x782C, NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered: 140>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=127, server_mask=11264, maximum_outgoing_transfer_size=127, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=False, *is_security_capable=False)])
Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x782C), src_ep=0, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=0, source_route=None, extended_timeout=False, tsn=134, profile_id=0, cluster_id=32770, data=Serialized[b'#\x00,x\x02@\x8c4\x12l\x7f\x00\x00,\x7f\x00\x00'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=255, rssi=-50)
Received frame on uninitialized device <Device model=None manuf=None nwk=0x782C ieee=54:ef:44:10:00:6d:ac:e3 is_initialized=False> from ep 0 to ep 0, cluster 32770: b'#\x00,x\x02@\x8c4\x12l\x7f\x00\x00,\x7f\x00\x00'
[... cut ...]
[0xCADA](S31 Lite zb): Update device availability -  device available: True - new availability: True - changed: False
[0xC315](DS01): Device seen - marking the device available and resetting counter
[0xC315](DS01): Update device availability -  device available: True - new availability: True - changed: False
[0x9602](unk_model): last_seen is None, marking the device unavailable
[0x9602](unk_model): Update device availability -  device available: False - new availability: False - changed: False

```

</details>

TheJulianJES, 2023-03-16:
What coordinator are you using and which firmware version is running on it?

cvocvo, 2023-03-16:
I'm using a Nortek HUSBZB-1 with the v6.7.8 firmware from here: https://github.com/walthowd/husbzb-firmware

dmulcahey, 2023-03-16:
You need to have hashed link keys enabled. How old is your Zigbee network? if it’s a couple years old there is a chance you need to tweak a setting.

cvocvo, 2023-03-16:
Huh yea it’s a year or two old. How do I turn that on or check if it’s configured? Thank you!

dmulcahey, 2023-03-16:
You need to take a bellows backup of the coordinator and then use bellows restore with the -U option to upgrade it to hashed link keys: https://github.com/zigpy/bellows/blob/e700caf2a50454d0060c3fcd5f67f4b49ebfce46/bellows/cli/backup.py#L156

Procedures for backup and restore: https://github.com/zigpy/zigpy/wiki/Coordinator-Backup-and-Migration#instructions-for-ezsp-radios

cvocvo, 2023-03-17:
Ahh perfect that did the trick -- THANK YOU!!!

I upgraded, removed and readded the FP1 and everything is working as expected.

By the way if anyone stumbles across this, you can check via bellows if you have hashed link keys enabled on your coordinator / network.

Running this:
`sudo bellows -d /dev/ttyUSB1 info`

Yielded a bunch of output but, BEFORE:
`[<EmberStatus.SUCCESS: 0>, EmberCurrentSecurityState(bitmask=<EmberCurrentSecurityBitmask.64|32|HAVE_TRUST_CENTER_LINK_KEY|GLOBAL_LINK_KEY:`
And AFTER doing the backup and restore upgrade -- you'll see the HASHED_LINK_KEY part noted in the output:
`[<EmberStatus.SUCCESS: 0>, EmberCurrentSecurityState(bitmask=<EmberCurrentSecurityBitmask.TRUST_CENTER_USES_HASHED_LINK_KEY|64|32|HAVE_TRUST_CENTER_LINK_KEY|GLOBAL_LINK_KEY:`

Backup command:
`sudo bellows -d /dev/ttyUSB1 backup > husbzb1-backup-03162023.txt`

Also, this is the upgrade command I used that didn't work:
```
sudo bellows -d /dev/ttyUSB1 restore -U -B husbzb1-backup-03162023.txt
Restoring NCP
Network is up, not forcing restore
```
But forcing it did work:
```
sudo bellows -d /dev/ttyUSB1 restore -U -B husbzb1-backup-03162023.txt --force
Restoring NCP
```

closed 2023-03-17 by cvocvo