zigpy/zha-device-handlers #2277: math domain error ZHA/SkyConnect/HA

Anto79-ops, 2023-03-16:
**Describe the bug**
Log appeared in my HA Core Logs. Not sure if its affecting my ZHA network

```
Logger: bellows.ezsp
Source: /usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py:354
First occurred: 3:45:03 AM (252 occurrences)
Last logged: 7:24:41 AM

Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 352, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 462, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 507, in _handle_frame
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 875, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 381, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 370, in handle_message
    self.endpoints[src_ep].handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 373, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1480, in handle_cluster_request
    status = getattr(self, handler_name)(*args)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1498, in handle_get_data
    getattr(self, dp_handler)(record)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 1531, in _dp_2_attr_update
    value = dp_map.converter(value)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/ts0601_motion.py", line 194, in <lambda>
    lambda x: 10000 * math.log10(x) + 1,
ValueError: math domain error
```

System:

Home Assistant 2023.3.4
Supervisor 2023.03.1
Operating System 9.5
Frontend 20230309.0 - latest
SkyConnect on FW 7.1.4.0 on ZHA

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<details>

<summary>Device signature</summary>

```yaml
Paste the device signature here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

diagnostics files
[zha-35e4f47bd1aec2897f2fa1af0eeb1329-Zigbee Coordinator-d65cbd0aa0eb07da942d2647eabdcc65.json.txt](https://github.com/zigpy/zha-device-handlers/files/10992164/zha-35e4f47bd1aec2897f2fa1af0eeb1329-Zigbee.Coordinator-d65cbd0aa0eb07da942d2647eabdcc65.json.txt)

<details>

<summary>Diagnostic information</summary>

in my config.yaml

```yaml
# ZHA Configuration Options
zha:
  custom_quirks_path: /config/custom_zha_quirks/
  zigpy_config:
    ota:
      ikea_provider: true       # Auto update Trådfri devices

```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

TheJulianJES, 2023-03-23:
The affected line is this:
https://github.com/zigpy/zha-device-handlers/blob/fa60b9d23a3e94b5a164e6aad1f24f4134544daa/zhaquirks/tuya/ts0601_motion.py#L194
`x` is 0 here.
This was fixed in some custom quirks with either:
```python
converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
```
or
```python
converter=lambda x: 10000 * math.log10(x + 1)
```

@javicalle Was there something that was preferred? Should I PR this?

javicalle, 2023-03-23:
My args and comments are here (and many other places):
* https://github.com/zigpy/zha-device-handlers/issues/1599#issuecomment-1318994425

But it is true that the reverse in HA leads to the first (and most popular) version:
* https://github.com/home-assistant/core/blob/73ed6e039ae77c6d7e0b19d1750fae2efedc8314/homeassistant/components/zha/sensor.py#LL399C9-L399C53

So (without any knowledge of the principles of illumination) the right one would be:
```python
converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
```

Please, fell free to create the PR.

closed 2023-04-07 by TheJulianJES

javicalle, 2023-04-07:
Thanks Julian!