zigpy/zha-device-handlers #2278: [Device Support Request] Tuya/Lora tap TS0601  _TZE200_vucankjx 1 gang in wall dimmer

ShogunMan, 2023-03-16:
**Is your feature request related to a problem? Please describe.**
I have added the TS0601  _TZE200_vucankjx to the ts0601_dimmer.py quirk under:
 class TuyaSingleSwitchDimmerGP(TuyaDimmerSwitch)

The quirk is loaded correctly, and I can turn on and off the light without any issues.
However the payload for dimming seems off. It is at full power at approx 25% and stays quite lit at 1%.
It seems to use TuyaInWallLevelControlNM (Endpoint id: 1, Id: 0x0008, Type: in) for dimming level, but I have no clue of how to fix this.

**Describe the solution you'd like**
Linear dimming for 0-100% (0-255)

<details>

<summary>
</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_vucankjx",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaSingleSwitchDimmerGP"
} ``` marks.
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant Container",
    "version": "2023.3.4",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Oslo",
    "os_name": "Linux",
[... 339 lines cut ...]
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
} ``` marks.
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context or screenshots about the feature request here.

javicalle, 2023-03-16:
Please, enable the debug logs and attach the logs from operating the physical device indicating the operation. Ie: logs dimming from minimun to max, switch off, dimming from 50% to minimum, etc...

ShogunMan, 2023-03-16:
What namespaces should I log?
The dimmer currently has no impulse switch attached. I could attach one for debugging purposes, but are aiming at using it with scene switches only.

ShogunMan, 2023-03-16:
Update:
I think it basically is a shitty dimmer. I tried three different dimmable led bulbs, all had either flicker or did not dim linearly.
Also tested with physical impulse switch with the same result.
A halogen bulb was linear all the way. Conclusion is that it's useless for dimming LED bulbs. Can be used as a on/off switch.

MattWestb, 2023-03-16:
I cant finding it but  i think some tuya dimmer have attributes for setting lode type as Tungsten, Halogen and LED but i dont knowing if this device is having it in the firmware and implanted in the quirk as attribute.

MattWestb, 2023-03-16:
I was finding the "tuya Zigbee" version they is using:

- 0x0008 Level Control Cluster ⬜
  - 0xF000 "tuya Level control" (t.uint16_t) 0-1000 ⬜
  - 0xfc00 Min and Max dimming Level (t.uint16_t) (default 0x01ff) ⬜
    - 0xAABB
AA = The first 8 Bit= Min dim level 0x03XX = 1%, 0xfcXX = 99% and 0xffXX = Not Set
BB = The last 8 Bit = Assumed being the same but for Max dim level.
  - 0xfc02 Bulb type	(uint8) 0x00–0xFF (default 0x00)
    - 0x00 LED
    - 0x01 Incandescent
    - 0x02 Halogen
  - 0xfc03 Scr state (uint8) 0x00–0xFF (default 0x01)
  - 0xfc04 Current percentage	(uint8)  0x00–0xFF (default 0x01)
  - 0xfc05 Min percentage (uint8) 0x00–0xFF	(default 0x01)

But i dont knowing if the same config is implanted in your device.

ShogunMan, 2023-03-16:
I am pretty sure that it is this implementation that is used.
I'll see if I can fiddle with the control cluster from within HA to see if it makes a difference.

ShogunMan, 2023-03-16:
Nah, not getting any way with this. 
@MattWestb Any chance you could adjust the quirk to the implementation you listed?

javicalle, 2023-03-17:
The `TuyaSingleSwitchDimmerGP` already implements the `bulb_type` attribute.
It must be available from the device cluster management, `TuyaInWallLevelControlNM` cluster, `bulb_type (0xEF02)` attribute.

Read the current value and try to set any of the other possible values. After every set, read the value again to see if the value is setted.

javicalle, 2023-03-17:
> What namespaces should I log? 

You can start here:
* https://github.com/zigpy/zha-device-handlers/issues/2278#issuecomment-1472766360

> The dimmer currently has no impulse switch attached. I could attach one for debugging purposes, but are aiming at using it with scene switches only.

I see. Then just enable the debug logs and check the report values from device after every HA command.

ShogunMan, 2023-03-17:
> The `TuyaSingleSwitchDimmerGP` already implements the `bulb_type` attribute. It must be available from the device cluster management, `TuyaInWallLevelControlNM` cluster, `bulb_type (0xEF02)` [...]

Changing the bulb type does not help. Initially it reads 'none'.
Tried all tree types, same results.

ShogunMan, 2023-03-17:
> > What namespaces should I log?
> 
> You can start here:
> 
> * [[Device Support Request] Tuya/Lora tap TS0601  _TZE200_vucankjx 1 gang in wall dimmer #2278 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2278#issuecomment-1472766360)
> 
> > The dimmer currently has no impulse switch attached. I could attach one for debugging purposes, but are aiming at using it with scene switches only.
> 
> I see. Then just enable the debug logs and check the report values from device after every HA command.
[home-assistant_2023-03-17T12-34-48.292Z.log](https://github.com/zigpy/zha-device-handlers/files/11001393/home-assistant_2023-03-17T12-34-48.292Z.log)

I don't know if the attached file makes any sense to you?

ShogunMan, 2023-03-17:
This is the Zigbee2MQTT implementation. I don't know if it is of any use:
`  {
        fingerprint: tuya.fingerprint('TS0601', ['_TZE200_ip2akl4w', '_TZE200_1agwnems', '_TZE200_la2c2uo9', '_TZE200_579lguh2',
            '_TZE200_vucankjx', '_TZE200_4mh6tyyo']),
        model: 'TS0601_dimmer_1',
        vendor: 'TuYa',
        description: '1 gang smart dimmer',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [tuya.exposes.lightBrightnessWithMinMax(), e.power_on_behavior(),
            tuya.exposes.countdown(), tuya.exposes.lightType()],
        meta: {
            tuyaDatapoints: [
                [1, 'state', tuya.valueConverter.onOff, {skip: tuya.skip.stateOnAndBrightnessPresent}],
                [2, 'brightness', tuya.valueConverter.scale0_254to0_1000],
                [3, 'min_brightness', tuya.valueConverter.scale0_254to0_1000],
                [4, 'light_type', tuya.valueConverter.lightType],
                [5, 'max_brightness', tuya.valueConverter.scale0_254to0_1000],
                [6, 'countdown', tuya.valueConverter.countdown],
                [14, 'power_on_behavior', tuya.valueConverter.powerOnBehavior],
            ],
        },
        whiteLabel: [
            {vendor: 'Moes', model: 'MS-105Z'},
            {vendor: 'Lerlink', model: 'X706U'},
            {vendor: 'Moes', model: 'ZS-EUD_1gang'},
        ],
    },`

javicalle, 2023-03-17:
Your logs showing that the quirk conversion is working as expected. And yes, it's the same that the used in Z2M.

Z2M also implements the `min_brightness` and `max_brightness` attributes which aren't implemented here. Not sure which is the right way to implement it, but in theory could help to set the dim limits which would translate in a better dim behavior.

Not in my hand to implement it.

ShogunMan, 2023-03-18:
I set up a new HA for testing with Z2M and it behaves in just the same way. I think the integration is ok. It's just a bad dimmer.
Z2M has options for changing bulb type, this does not fix anything.  I think we can close this thread.

closed 2023-03-19 by ShogunMan