zigpy/zha-device-handlers #2281: [BUG] TS0601 _TZE200_9yapgbuv battery status is not reporting

Craftplorer, 2023-03-18:
**Describe the bug**
The battery status is reported as "Unknown" 

**To Reproduce**
Add the sensor TS0601 _TZE200_9yapgbuv to home assistant. Temperature and Humidity will appear after some time battery status not. Tried for more than 24h without showing it up.

**Expected behavior**
As far as I understand the sensor doesn't support battery percentage reporting. Its only reports the battery state as low, mid or high.
I tested the sensor with the tuya app and the app only shows low, mid and high.

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4417, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0302",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0402",
        "0x0405",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_9yapgbuv",
  "model": "TS0601",
  "class": "zhaquirks.tuya.ts0601_sensor.TuyaTempHumiditySensorVar03"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.4",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
[... 416 lines cut ...]
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster,
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    TuyaPowerConfigurationCluster2AAA,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

</details>

javicalle, 2023-03-19:
In Z2M this device is reporting `batteryPercentageRemaining` and `batteryVoltage`.
Your quirk is adding the DP3. Z2M implements DP3 as `tthBatteryLevel` that report `low`|`middle`|`high` value. These values aren't managed as battery levels in HA.

In the current state I would suggest to manage it as a Tuya manufacturer attribute and add it as an ZHA attribute (a diagnostic one).
The hard part could be how to know which devices must initialize the value (to make it available in HA). Maybe all the `TuyaTempHumiditySensorVar03`?

Craftplorer, 2023-03-24:
Thanks for the feedback. Is there a guideline or a other issue already discussing this? Would it be possible to have low, mid and high just report as 0%, 50% and 100% for the normal battery percentage?

Also if possible can someone with more knowledge implement this quirk properly? My solution isn't really clean.

> The hard part could be how to know which devices must initialize the value (to make it available in HA). Maybe all the `TuyaTempHumiditySensorVar03`?

Maybe just implement it for _TZE200_9yapgbuv for now and see if others got similar problems on other devices later. I also ordered a few more of this sensors from china maybe I get a different version and can give an update.

labaland, 2023-10-18:
same yet

PR #2626 (2023-10-24): Fix battery reporting for `_TZE200_yjjdcqsq`, `_TZE200_9yapgbuv`
Both devices report battery status in three levels: low, middle, and high instead of reporting a battery percentage. I confirmed this with the official TuYa App and TuYa IoT Platform. See Issue #2281. I tested this for both devices: _TZE200_yjjdcqsq and _TZE200_9yapgbuv.

## Proposed change
Since there is no option for battery level in ZHA, this fix converts the levels into equal percentage numbers 25%, 50% and 100%.

## Additional information
Fixes #2281

## Checklist

- [x] The changes are tested and work correctly
- [x] `pre-c [... cut]
closed 2023-10-24 by TheJulianJES