zigpy/zha-device-handlers #2283: [Device Support Request] DDS238-2 devices (TZE200_byzdayie, TZE200_ewxhg6o9 and TZE200_bkkmqmyo) and ZHA

jproveta, 2023-03-19:
Hi, I have several DDS238-2 devices (TZE200_byzdayie, TZE200_ewxhg6o9 and TZE200_bkkmqmyo), all working fine, but with the HA 2023.3 update I no longer have the "Metering_summation_delivered" entity update, if I remove the device and re-add it, that entity (Metering_summation_delivered) does not appear. And the "Power factor" is "Unknown" for ZE200_byzdayie and TZE200_ewxhg6o9, and the TZE200_bkkmqmyo the entity "RMS current" not update and no data.

The file I'm using is
[ts0601_din_power.zip](https://github.com/zigpy/zha-device-handlers/files/11012333/ts0601_din_power.zip)

Thanks for help me.

javicalle, 2023-03-19:
I suppose that you are using a custom quirk...
Check this:
https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417

jproveta, 2023-03-19:
> I suppose that you are using a custom quirk... Check this: [#2250 (comment)](https://github.com/zigpy/zha-device-handlers/issues/2250#issuecomment-1453978417)

Yes, I use a custom quick, but the solution you gave me doesn't work for me, because I don't have "TuyaDPType" and "dp_type=TuyaDPType.XXXXXX" in the file I use.

Thanks.

jproveta, 2023-03-22:
This thing of being dependent on third-party support to have updates when they are needed, it's f... and taking a lot of risk!

Now I have 28 devices for the trash because HA did an update (I don't know what affected the ZHA) and the custom quirk I have for these devices that I got here, didn't work correctly and there's no support!

javicalle, 2023-03-22:
If you want to have support in your incident, I recommend you at least fill in the template with as much information as possible to help the maintainers help you.
As it does not seem to be a known error, I recommend that you include the logs from your devices, especially if they contain any errors or information that can help with the resolution.

dieneuser, 2023-04-10:
In this [post](https://github.com/zigpy/zha-device-handlers/issues/1973) the problem is also already reported.
Since the 2023.3 update I have the following error message in the log:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/__init__.py", line 357, in handle_callback
    handler(*args)
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 463, in ezsp_callback_handler
    self._handle_frame(*args)
  File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 508, in _handle_frame
    self.packet_received(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 913, in packet_received
    self.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 434, in handle_message
    return sender.handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/device.py", line 367, in handle_message
    self.endpoints[src_ep].handle_message(
  File "/usr/local/lib/python3.10/site-packages/zigpy/endpoint.py", line 224, in handle_message
    handler(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zigpy/zcl/__init__.py", line 375, in handle_message
    self.handle_cluster_request(hdr, args, dst_addressing=dst_addressing)
  File "/usr/local/lib/python3.10/site-packages/zhaquirks/tuya/__init__.py", line 494, in handle_cluster_request
    zvalue = ztype(tuya_data)
  File "/usr/local/lib/python3.10/site-packages/zigpy/types/basic.py", line 94, in __new__
    raise ValueError(
ValueError: 196061 is not an unsigned 16 bit integer
```

closed 2023-10-14 by github-actions[bot]