zigpy/zha-device-handlers #2286: [Device Support Request] Niko Double connectable switch,10A (552-721X2)

arvidc, 2023-03-20:
**Is your feature request related to a problem? Please describe.**
The device can be readily connected and the basic functionality (i.e. switching either relay; e.g. to control lights) seems to work as is. However, there are additional functionalities in a manufacturer specific cluster that are not currently supported by ZHA. More specifically, the following functions (per relay) could be added: Operation_mode (control_relay or decoupled), Led_enable (ON or OFF) and Led_state. In particular the first would be highly welcome, as it allows the switch to operate better with smart lights (by not actually switching the relay, but instead sending out an action event that could in turn be used to control the smart light).

**Describe the solution you'd like**
Adding support for missing functions in ZHA.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4703, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0b05",
        "0xfc00",
        "0xfc01"
      ],
      "out_clusters": [
        "0x0019"
      ]
    },
    "2": {
      "profile_id": 260,
      "device_type": "0x010a",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0b05",
        "0xfc00",
        "0xfc01"
      ],
      "out_clusters": []
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0061",
      "in_clusters": [],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "NIKO NV",
  "model": "Double connectable switch,10A",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
[... 571 lines cut ...]
          "id": 97
        },
        "profile_id": 41440,
        "in_clusters": {},
        "out_clusters": {
          "0x0021": {
            "endpoint_attribute": "green_power",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

**Additional context**
The missing functions outlined above have already been implemented in Zigbee2mqtt, so I'm hoping somehow who actually knows what's what (evidently not me) could use that as a solid basis (or even 'just' translate between both?) to bring this into ZHA?
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/niko.js
https://www.zigbee2mqtt.io/devices/552-721X2.html

If you any more info is needed, please let me know.

arvidc, 2023-03-20:
As a side-note: the linked Zigbee-Herdsman Converter covers many more Niko devices. In particular, there is also a single version of this switch (Single connectable switch,10A; model: '552-721X1') that is very closely related. That said, I only own the double, so I could only test that one.

arvidc, 2023-03-23:
To my own surprise, I've actually stumbled my way into writing something that seems to cover the essentials reasonably well...

<details>
<summary>code</summary>

````python

"""Niko switches (only Niko Double connectable switch,10A (552-721X2) at this time)"""
import logging

from zigpy import types as t
from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Identify,
    Groups,
    Scenes,
    OnOff,
    Ota,
    GreenPowerProxy,
)
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.manufacturer_specific import ManufacturerSpecificCluster

from zhaquirks import EventableCluster
from zhaquirks.const import (
    ARGS,
    BUTTON_1,
    BUTTON_2,
    COMMAND_CLICK,
    COMMAND_HOLD,
    COMMAND_RELEASE,
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)

class Niko1(CustomCluster, ManufacturerSpecificCluster):
    """manufacturer specific cluster related to device settings"""

    class ButtonOperationMode(t.enum8):
        # WARN: while one can technically write 0x00 to the operationMode attribute
        #       this seems to brick the device and it will need to be rejoined
        Decoupled = 0x01
        ControlRelay = 0x02
    
    class LedState(t.uint8_t):
        Off = 0
        On = 255

    class LedOperationMode(t.uint24_t):
        Decoupled = 0
        ControlledByRelay = 1
        #Notable behaviour:
        #If LedOperationMode == ControlledByRelay, ButtonOperationMode == ControlRelay and LedState == On, then LED reflects status of corresponding relay
        #If ButtonOperationMode == Decoupled, then neither buttons nor relay affect the LEDs (i.e. LedOperationMode acts as if ==Decoupled)
        #If LedOperationMode == Decoupled and/or ButtonOperationMode == Decoupled, then LED can nevertheless by controlled by writing LedState ('always on' or 'always off')
    
    cluster_id = 0xfc00
    # name = "Niko1"
    # ep_attribute = "Niko_1"
    attributes = {
        # If this would ever be expanded to outlets:
        # WARNING: 0x0000 has different datatypes! 
        # enum8 (switch) vs. bitmap8 (outlet)
        # unknown usage/function on outlet
[... 115 lines cut ...]
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0061, #??
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }

    device_automation_triggers = {
        (COMMAND_CLICK, BUTTON_1): {ARGS: {'value': 64}},
        (COMMAND_HOLD, BUTTON_1): {ARGS: {'value': 32}},
        (COMMAND_RELEASE, BUTTON_1): {ARGS: {'value': 48}},
        (COMMAND_CLICK, BUTTON_2): {ARGS: {'value': 16384}},
        (COMMAND_HOLD, BUTTON_2): {ARGS: {'value': 8192}},
        (COMMAND_RELEASE, BUTTON_2): {ARGS: {'value': 12288}},
    }
````
</details>

If possible (even if not crucial), I'd like to still expose some additional (configuration) entities in HA; such as a select for ButtonOperationMode. However, if I understand e.g. the discussion under #2120 correctly, it seems that would also require changes to e.g. `homeassistant/components/zha/select.py`? Or can someone point me in a different direction?

svenjochems, 2023-09-03:
Hi @arvidc 

Thanks for working on this.
Does your solution also support the external inputs of the device? This was discussed and added to zigbee2mqtt in 04/2023.
https://github.com/Koenkk/zigbee2mqtt/issues/13737#issuecomment-1520002786

arvidc, 2023-09-04:
Hey @svenjochems, thanks for you question.

I had come across your post but haven't added any device_automation_triggers based on the external inputs yet (mostly because I don't have any in my setup at home yet; but that will likely change later this year, so it is something I've been wanting to do).
I did make some further (small) improvements to the script I've been using, so the above isn't the most current version.

One thing that complicates things a bit further is that I've found that if you press both buttons at once, the reported value will be the sum of both. For example, a simple "short dual press" is reported as either 4160 (=64+4096) or 16400 (=16+16384) depending on which button you end up pressing first (and possibly even 16448 if perfectly in sync, but in my small testing I don't believe I ever saw that).
I've been thinking there is probably a more elegant way to handle this than hardcoding all the possibilities: 
1. in principle you can separate both signals (or all 4, with external inputs) with operations like floor/modulo/whatever; and/or
2. representing these as hexadecimals would be a lot more intuitive at least. (*)

I'll see if I can work something out in the course of this week and come back with an updated version (and then see how I could possibly submit it or what more needs to be done).

(*) The above three decimals 4160, 16400 and 16448 could simply map to 0x01040, 0x04010 and 0x04040.
More generally every possible combination should yield a hex of the form 0xDCBA0, with A, B, C and D from 1-4 (corresponding to the four possibilities: 'initial', 'long', 'release', 'short') and A corresponding to the input of the first button, B first external, C second button and D second external.

svenjochems, 2023-10-19:
@arvidc 

I created some test code to shift the bits to get the correct button and extract the action from the normalized value, this seems to give the correct output for all the values:

```
shift_left = 4
shift_left_ext = 8
shift_right = 12
shift_right_ext = 16

def extractAction(value):
  # shift bits and mask last 4
  normalized_left =      value >> shift_left & 0xF
  normalized_left_ext =  value >> shift_left_ext & 0xF
  normalized_right =     value >> shift_right & 0xF
  normalized_right_ext = value >> shift_right_ext & 0xF
  
  if normalized_left > 0:
    print('left ' + mapAction(normalized_left))
  elif normalized_left_ext > 0:
    print('left ext ' + mapAction(normalized_left_ext))
  elif normalized_right > 0:
    print('right ' + mapAction(normalized_right))
  elif normalized_right_ext > 0:
    print('right ext ' + mapAction(normalized_right_ext))
  

def mapAction(normalized_value):
  if normalized_value == 1:
    return 'null'
  elif normalized_value == 4:
    return 'single'
  elif normalized_value == 2:
    return 'hold'
  elif normalized_value == 3:
    return 'release'
  else:
    return 'unknown'

# Test
extractAction(16)
extractAction(64)
extractAction(32)
extractAction(48)
extractAction(4096)
extractAction(16384)
extractAction(8192)
extractAction(12288)
extractAction(256)
extractAction(1024)
extractAction(512)
extractAction(768)
extractAction(65536)
extractAction(262144)
extractAction(131072)
extractAction(196608)

# dual presses
extractAction(4160)
extractAction(16400)
```

svenjochems, 2023-11-07:
Hi @arvidc 

This week I had some time to migrate my zigbee2mqtt setup to zha.
I was able to test your custom quirk successfully.

I've made some small changes to the code and added support for the single switch.
I've also tried to get the conifguration to the frontend, currently without success.

PR: https://github.com/zigpy/zha-device-handlers/pull/2723

closed 2024-05-12 by github-actions[bot]