zigpy/zha-device-handlers #2290: [BUG] TS0501 fan controller reporting null percentage

Brazen00, 2023-03-22:
**Describe the bug**
I have one of these fan controllers https://zigbee.blakadder.com/Mercator_SSWF01G.html

Everything works fine when controlling this through HA. However, when the fan controller is physically activated using the middle "fan on/off" button, the speed percentage is reporting as 'null'. Activating or switching speed using the "fan speed" button correctly sets the percentage in HA.

This was submitted as a HA core issue (https://github.com/home-assistant/core/issues/89413), but reporting here also.

**Expected behavior**
I expect the middle "fan on/off" button to set the speed correctly in HA.

**Screenshots**
[image]

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0100",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0202",
        "0xef00"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3210_lzqq3u4r",
  "model": "TS0501",
  "class": "zhaquirks.tuya.ts0501_fan_switch.TS0501FanSwitch"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.5",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Australia/Brisbane",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.1",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
[... 661 lines cut ...]
            "endpoint_attribute": "ota",
            "attributes": {},
            "unsupported_attributes": {}
          }
        }
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
Add any other context about the problem here.

javicalle, 2023-03-27:
There is no `ts0501_fan_switch.py` or `TS0501FanSwitch` quirk in the code.
Without the code you are using no support can be done.

Brazen00, 2023-03-27:
Isn’t this it? https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0501_fan_switch.py

javicalle, 2023-03-27:
Ouuuch! My fault. I suppose that my local repo isn't updated.

Can you enable the debug logs (can be done from the ZHA integration) and attach the logs from the good and the bad behavior?
Please, also report the sequence of actions to correlate them with the logs.

Brazen00, 2023-07-26:
Ok, sorry for the loooong delay.

I believe this is the log entry sequence for physically pushing the on/off button on the wall fan controller. After this happens, the percentage is being reported as 'null':
```
2023-07-26 14:01:07.562 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=514, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=169), 124, -69, 0xe338, 255, 255, b'\x18{\n\x00\x000\x04']
2023-07-26 14:01:07.562 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE338), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=169, profile_id=260, cluster_id=514, data=Serialized[b'\x18{\n\x00\x000\x04'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=124, rssi=-69)
2023-07-26 14:01:07.563 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Received ZCL frame: b'\x18{\n\x00\x000\x04'
2023-07-26 14:01:07.563 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=123, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-07-26 14:01:07.564 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Decoded ZCL frame: FanCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=enum8, value=<enum8.undefined_0x04: 4>))])
2023-07-26 14:01:07.564 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Received command 0x0A (TSN 123): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=enum8, value=<enum8.undefined_0x04: 4>))])
2023-07-26 14:01:07.564 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Attribute report received: fan_mode=<enum8.undefined_0x04: 4>
2023-07-26 14:01:07.564 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE338:1:0x0202]: Attribute report 'Fan Control'[fan_mode] = FanMode.On
2023-07-26 14:01:07.564 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=fan.fan_controller_office_fan, old_state=<state fan.fan_controller_office_fan=off; preset_modes=['on', 'auto', 'smart'], percentage=0, percentage_step=33.333333333333336, preset_mode=None, friendly_name=Fan Controller Office Fan, supported_features=1 @ 2023-07-26T13:59:50.649212+10:00>, new_state=<state fan.fan_controller_office_fan=on; preset_modes=['on', 'auto', 'smart'], percentage=None, percentage_step=33.333333333333336, preset_mode=on, friendly_name=Fan Controller Office Fan, supported_features=1 @ 2023-07-26T14:01:07.564883+10:00>>

```

and this is the log sequence for subsequently pushing the speed button on the fan. after this, the percentage is reported as 33:
```
2023-07-26 14:01:19.455 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0xE338), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=170, profile_id=260, cluster_id=514, data=Serialized[b'\x18|\n\x00\x000\x01'], tx_options=<TransmitOptions.NONE: 0>, radius=0, non_member_radius=0, lqi=144, rssi=-64)
2023-07-26 14:01:19.455 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Received ZCL frame: b'\x18|\n\x00\x000\x01'
2023-07-26 14:01:19.456 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Decoded ZCL frame header: ZCLHeader(frame_control=FrameControl(frame_type=<FrameType.GLOBAL_COMMAND: 0>, is_manufacturer_specific=0, direction=<Direction.Client_to_Server: 1>, disable_default_response=1, reserved=0, *is_cluster=False, *is_general=True), tsn=124, command_id=10, *direction=<Direction.Client_to_Server: 1>)
2023-07-26 14:01:19.456 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Decoded ZCL frame: FanCluster:Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>))])
2023-07-26 14:01:19.457 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Received command 0x0A (TSN 124): Report_Attributes(attribute_reports=[Attribute(attrid=0x0000, value=TypeValue(type=enum8, value=<enum8.undefined_0x01: 1>))])
2023-07-26 14:01:19.457 DEBUG (MainThread) [zigpy.zcl] [0xE338:1:0x0202] Attribute report received: fan_mode=<enum8.undefined_0x01: 1>
2023-07-26 14:01:19.457 DEBUG (MainThread) [homeassistant.components.zha.core.cluster_handlers] [0xE338:1:0x0202]: Attribute report 'Fan Control'[fan_mode] = FanMode.Low
2023-07-26 14:01:19.457 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=fan.fan_controller_office_fan, old_state=<state fan.fan_controller_office_fan=on; preset_modes=['on', 'auto', 'smart'], percentage=None, percentage_step=33.333333333333336, preset_mode=on, friendly_name=Fan Controller Office Fan, supported_features=1 @ 2023-07-26T14:01:07.564883+10:00>, new_state=<state fan.fan_controller_office_fan=on; preset_modes=['on', 'auto', 'smart'], percentage=33, percentage_step=33.333333333333336, preset_mode=None, friendly_name=Fan Controller Office Fan, suppor [... cut ...]
2023-07-26 14:01:25.534 DEBUG (MainThread) [homeassistant.core] Bus:Handling <Event state_changed[L]: entity_id=fan.fan_controller_office_fan, old_state=<state fan.fan_controller_office_fan=on; preset_modes=['on', 'auto', 'smart'], percentage=33, percentage_step=33.333333333333336, preset_mode=None, friendly_name=Fan Controller Office Fan, supported_features=1 @ 2023-07-26T14:01:07.564883+10:00>, new_state=<state fan.fan_controller_office_fan=off; preset_modes=['on', 'auto', 'smart'], percentage=0, percentage_step=33.333333333333336, preset_mode=None, friendly_name=Fan Controller Office Fan, supported_features=1 @ 2023-07-26T14:01:25.534072+10:00>>
```

Brazen00, 2024-01-22:
This issue is still occurring.