zigpy/zha-device-handlers #2294: [Device Support Request] Generic ZigBee bulbs (Gledopto whitelabel?)

agryson, 2023-03-23:
I purchased a mix of second hand ZigBee bulbs. They all appear to be an unlabelled version of Gledopto bulbs as they have the same pairing process and appear identical in product shots. I have successfully paired them with ZHA and can turn them on and off from the HA Light Card but have no control over color temperature or brightness in the standard cards (controls which are provided but have no effect).

Going into 'manage zigbee device' from the device menu allows me to fire off zigbee commands that change both the brightness and the color temperature (see images at bottom)

So I know the hardware supports it.

Their Manufacturer is listed as "ZigBee/CCT" and the model as "CCT Light"

**Describe the solution you'd like**
As they appear to be generic Gledopto bulbs I see 2 possible solutions:
1. If someone could guide me to how to modify an existing quirk I'll gladly do that
2. If a generic quirk for this "ZigBee/CCT, CCT Light" were availabel that would also solve my issue

I have tried to make a custom quirk but it doesn't seem to 'take' even after removing and repairing one of the bubs.

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.Router: 1>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|RxOnWhenIdle|MainsPowered|FullFunctionDevice: 142>, manufacturer_code=4137, maximum_buffer_size=127, maximum_incoming_transfer_size=90, server_mask=11264, maximum_outgoing_transfer_size=90, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=False, *is_full_function_device=True, *is_mains_powered=True, *is_receiver_on_when_idle=True, *is_router=True, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x010c",
      "in_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0x100f",
        "0xffff"
      ],
      "out_clusters": [
        "0x0019",
        "0x100f"
      ]
    },
    "242": {
      "profile_id": 41440,
      "device_type": "0x0066",
      "in_clusters": [
        "0x0021"
      ],
      "out_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "ZigBee/CCT",
  "model": "CCT Light",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
# I couldn't download diagnostics, it just opens an empty tab, will try again later and post it here if I'm successful
```

</details>

<details>
<summary>Additional logs</summary>

```
Paste any additional debug logs here.
Don't remove the extra line breaks outside the ``` marks.
```

</details>

**Additional context**
using this interface lets me change the temperature and 
[image]
[image]

TheJulianJES, 2023-04-11:
> brightness in the standard cards (controls which are provided but have no effect).

Just to confirm, when the light is already on and you're just changing the brightness through the UI, the brightness of the actual bulb does not change at all **and** the UI also reverts to the old brightness?

Or does the bulb change brightness and the UI just doesn't reflect that?

TheJulianJES, 2023-04-11:
Please install this as a [custom quirk](https://github.com/zigpy/zha-device-handlers/discussions/693#discussioncomment-857274) and let me know if it works.

<details>
<summary>Custom Quirk (click to expand)</summary>

```python
"""Quirk for ZigBee/CCT CCT Light."""

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.lighting import (
    Color,
)
from zigpy.zcl.clusters.lightlink import (
    LightLink,
)
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.kof.kof_mr101z import NoReplyMixin

class LevelControlNoReply(NoReplyMixin, CustomCluster, LevelControl):
    """LevelControl cluster that does not require default responses."""

    void_input_commands = {cmd.id for cmd in LevelControl.commands_by_name.values()}

class ZigbeeCctCctLight(CustomDevice):
    """Zigbee/Cct CCT Light custom device implementation."""

    signature = {
        MODELS_INFO: [("ZigBee/CCT", "CCT Light")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    0x100F,
                    0xFFFF,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    0x100F,
                ],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0066,
                INPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COLOR_TEMPERATURE_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControlNoReply,
                    Color.cluster_id,
                    LightLink.cluster_id,
                    0x100F,
                    0xFFFF,
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                    0x100F,
                ],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 0x0066,
                INPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
                OUTPUT_CLUSTERS: [
                    GreenPowerProxy.cluster_id,
                ],
            },
        },
    }
```

</details>

agryson, 2023-04-13:
Hello again,
In a bittersweet development, I plugged the bulbs back in to confirm if the behaviour was "move back to x%" or not and found that the bulbs now properly allow level and color control...
I did upgrade from 2023.3 (version at time of report) to 2023.4 but the bulbs do not report as using a quirk so am not sure if that's the change.
So, while I'm of course happy that they seem to now "just work", I am a little embarrassed at you having spent effort apparently needlessly @TheJulianJES .
In any case, thank you very much for your help and since I've now confirmed that all 4 bulbs work, can close this issue.

closed 2023-04-13 by agryson