zigpy/zha-device-handlers #2297: [Device Support Request] XHS2-UE Contact Sensor by Sercomm Corp.

sjbarkey, 2023-03-24:
**Is your feature request related to a problem? Please describe.**
The HXS2-SE device does report in contact open/close status, as well as the temperature reading, but the batter level always reports as "unknown", even after multiple days of use.

This appears to be another variant/version of the same device fixed in https://github.com/zigpy/zha-device-handlers/issues/1550.

**Describe the solution you'd like**
I would like the XHS2-SE ZigBee device to accurately report it's battery status so I can monitor the levels and replace batteries as necessary

<details>

<summary>Device signature</summary>

```yaml
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4401, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=0, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0402",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0020",
        "0x0402",
        "0x0500",
        "0x0b05"
      ],
      "out_clusters": [
        "0x0003",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Sercomm Corp.",
  "model": "XHS2-SE",
  "class": "zigpy.device.Device"
}
```

</details>

<details>

<summary>Diagnostic information</summary>

```yaml
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2023.3.6",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.10.10",
    "docker": true,
    "arch": "aarch64",
    "timezone": "America/Los_Angeles",
    "os_name": "Linux",
    "os_version": "5.15.90",
    "supervisor": "2023.03.2",
    "host_os": "Home Assistant OS 9.5",
    "docker_version": "20.10.22",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "o365": {
      "version": "v4.2.8",
      "requirements": [
[... 384 lines cut ...]
      }
    }
  }
}
```

</details>

<details>
<summary>Additional logs</summary>

```
N/A
```

</details>

**Additional context**
I have other physically identical devices that have a different model number, URC4460BC0-X-R, that had this same fix implemented in https://github.com/zigpy/zha-device-handlers/issues/1550

Working Device:
[image]

Problem Device:
[image]

javicalle, 2023-03-26:
Probably you can get it working with that quirk:
 ```python
class ContactSensor(CustomDevice):
    """Custom device representing iMagic by Greatstar motion sensors."""

    signature = {
        #  <SimpleDescriptor endpoint=1 profile=260 device_type=0x0402
        #  input_clusters=["0x0000", "0x0001", "0x0003","0x0020","0x0402", "0x0500", "0x0b05","0xfc01","0xfc02"]
        #  output_clusters=["0x0003","0x0019"]>
        MODELS_INFO: [(SERCOMM, "XHS2-SE")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    PowerConfiguration.cluster_id,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    IasZone.cluster_id,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    SercommPowerConfiguration,
                    Identify.cluster_id,
                    PollControl.cluster_id,
                    TemperatureMeasurement.cluster_id,
                    IasZone.cluster_id,
                    Diagnostic.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
            }
        }
    }
```

The easy way to check is:
* enable the local quirk configuration
* copy in your folder the current version of `szwtd02n.py`:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/sercomm/szwtd02n.py
* add in the imports zone (at top of the file):
```python
from zigpy.zcl.clusters.measurement import TemperatureMeasurement
```
* add the suggested quirk at the bottom of file
* save changes, restart HA and repair the device

Battery reports may take a while (couple of hours) to come. Be patient.

sjbarkey, 2023-03-29:
I was able to successfully create a local quirk, with some **very slight** differences...

```
"""Device handler for Sercomm XHS2-SE contact sensor."""

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Identify, Ota, PollControl
from zigpy.zcl.clusters.homeautomation import Diagnostic
from zigpy.zcl.clusters.security import IasZone
from zigpy.zcl.clusters.measurement import TemperatureMeasurement

from zhaquirks import PowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.sercomm import SERCOMM

class SercommPowerConfiguration(PowerConfigurationCluster):
    """Sercomm power configuration cluster for contact sensor."""

    MAX_VOLTS = 3.2
    MIN_VOLTS = 2.1

class XHS2SE(CustomDevice):
   """Custom device representing XHS2-SE by Sercomm Corp."""

   signature = {
       #  <SimpleDescriptor endpoint=1 profile=260 device_type=0x0402
       #  input_clusters=["0x0000", "0x0001", "0x0003","0x0020","0x0402", "0x0500", "0x0b05","0xfc01","0xfc02"]
       #  output_clusters=["0x0003","0x0019"]>
       MODELS_INFO: [(SERCOMM, "XHS2-SE")],
       ENDPOINTS: {
           1: {
               PROFILE_ID: zha.PROFILE_ID,
               DEVICE_TYPE: zha.DeviceType.IAS_ZONE,
               INPUT_CLUSTERS: [
                   Basic.cluster_id,
                   SercommPowerConfiguration.cluster_id,
                   Identify.cluster_id,
                   PollControl.cluster_id,
                   TemperatureMeasurement.cluster_id,
                   IasZone.cluster_id,
                   Diagnostic.cluster_id,
               ],
               OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
           }
       },
   }

   replacement = {
       ENDPOINTS: {
           1: {
               INPUT_CLUSTERS: [
                   Basic.cluster_id,
                   SercommPowerConfiguration,
                   Identify.cluster_id,
                   PollControl.cluster_id,
                   TemperatureMeasurement.cluster_id,
                   IasZone.cluster_id,
                   Diagnostic.cluster_id,
               ],
               OUTPUT_CLUSTERS: [Identify.cluster_id, Ota.cluster_id],
           }
       }
   }
```

javicalle, 2023-03-30:
Glad to know that you can make it work!
Do you mind to create the PR to add your device to mainstream?
You can do it just editing the file from the web browser and following the instructions:
* https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/sercomm/szwtd02n.py

Take in acount that we want to add a new quirk (and not to replace/modify the existing one).
Your code (and mine) have a space less in tabulations than the existing code. I'm not sure but code validator can complain about it. In any case it would be better to make it homogeneous in the whole file.

Thanks in advanced.

sjbarkey, 2023-04-05:
> Glad to know that you can make it work! Do you mind to create the PR to add your device to mainstream? You can do it just editing the file from the web browser and following the instructions:
>
> * [...]

I've actually never done a PR, but happy to give it a shot when I have the chance.

sjbarkey, 2023-10-08:
Appears my PR failed for some reason.  Will work on resubmitting.

sjbarkey, 2023-10-08:
This issue is resolved and the PR has been merged at this point.

closed 2023-10-08 by sjbarkey